espressif/esp-iot-solution #101: The set coordinates exceed the screen size

ogghst, 2021-04-08:
## Environment

- Development Kit:      [ESP32-DevKitC]
- Kit version (DevKitC): [v4]
- Module or chip used:  [ESP32-WROOM-32D]
- IDF version (run ``git describe --tags`` to find it): 
    idf4.3_py3.9_2021-01-07
- Build System:         [CMake]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System:     [Windows]
- Power Supply:         [USB]

## Problem Description

trying the lvgl thermostat and other apps, everything runs OK wiring the display to the DEVKITC V4 board as per description, and removing the touch calibration line. 

but when connecting the XPT2046 pins (using a dedicated CS and the same MOSI/MISO/CLK pins), scaling down the touch clock to 2500000, i receive correct coordinates from the touch screen (something around 2000 and 3000 both X and Y) but immediately after 100-200 errors equals to this:

C:/Development/esp/esp-iot-solution-master/components/display/screen/controller_driver/ili9341/ili9341.c:248 (lcd_ili9341_set_window):The set coordinates exceed the screen size

### Code to reproduce this issue

```
void app_main()
{
    iot_board_init();
    spi_bus_handle_t spi2_bus = iot_board_get_handle(BOARD_SPI2_ID);

    scr_driver_t lcd_drv;
    touch_panel_driver_t touch_drv;
    scr_interface_spi_config_t spi_lcd_cfg = {
        .spi_bus = spi2_bus,
        .pin_num_cs = BOARD_LCD_SPI_CS_PIN,
        .pin_num_dc = BOARD_LCD_SPI_DC_PIN,
        .clk_freq = BOARD_LCD_SPI_CLOCK_FREQ,
        .swap_data = true,
    };

    scr_interface_driver_t *iface_drv;
    scr_interface_create(SCREEN_IFACE_SPI, &spi_lcd_cfg, &iface_drv);

    scr_controller_config_t lcd_cfg = {
        .interface_drv = iface_drv,
        .pin_num_rst = 18,
        .pin_num_bckl = 23,
        .rst_active_level = 0,
        .bckl_active_level = 1,
        .offset_hor = 0,
        .offset_ver = 0,
        .width = 240,
        .height = 320,
        .rotate = SCR_DIR_TBLR,
    };
    scr_find_driver(SCREEN_CONTROLLER_ILI9341, &lcd_drv);
    lcd_drv.init(&lcd_cfg);

    touch_panel_config_t touch_cfg = {
        .interface_spi = {
            .spi_bus = spi2_bus,
            .pin_num_cs = BOARD_TOUCH_SPI_CS_PIN,
            .clk_freq = 2500000,
        },
        .interface_type = TOUCH_PANEL_IFACE_SPI,
        .pin_num_int = -1,
        .direction = TOUCH_DIR_TBLR,
        .width = 240,
        .height = 320,
    };
    touch_panel_find_driver(TOUCH_PANEL_CONTROLLER_XPT2046, &touch_drv);
    touch_drv.init(&touch_cfg);
    touch_drv.calibration_run(&lcd_drv, true);
    /* Initialize LittlevGL GUI */
    lvgl_init(&lcd_drv, &touch_drv);

    // thermostat initialize
    littlevgl_thermostat();

    ESP_LOGI(TAG, "[APP] IDF version: %s", esp_get_idf_version());
    ESP_LOGI(TAG, "[APP] Free memory: %d bytes", esp_get_free_heap_size());
}

```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs

```
I (31) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (38) boot.esp32: SPI Speed      : 40MHz
I (43) boot.esp32: SPI Mode       : DIO
I (48) boot.esp32: SPI Flash Size : 4MB
I (52) boot: Enabling RNG early entropy source...
I (58) boot: Partition Table:
I (61) boot: ## Label            Usage          Type ST Offset   Length
I (68) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (76) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (83) boot:  2 factory          factory app      00 00 00010000 00100000
I (91) boot: End of partition table
I (95) boot_comm: chip revision: 1, min. application chip revision: 0
I (102) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=8ea84h (584324) map
I (331) esp_image: segment 1: paddr=0009eaac vaddr=3ffb0000 size=0156ch (  5484) load
I (334) esp_image: segment 2: paddr=000a0020 vaddr=400d0020 size=449f4h (281076) map
I (443) esp_image: segment 3: paddr=000e4a1c vaddr=3ffb156c size=01788h (  6024) load
I (446) esp_image: segment 4: paddr=000e61ac vaddr=40080000 size=0edd0h ( 60880) load
I (475) esp_image: segment 5: paddr=000f4f84 vaddr=50000000 size=00010h (    16) load
I (483) boot: Loaded app from partition at offset 0x10000
I (483) boot: Disabling RNG early entropy source...
I (495) cpu_start: Pro cpu up.
I (495) cpu_start: Starting app cpu, entry point is 0x40081430
I (0) cpu_start: App cpu up.
I (509) cpu_start: Pro cpu start user code
I (509) cpu_start: cpu freq: 160000000
I (510) cpu_start: Application information:
I (514) cpu_start: Project name:     lvgl_thermostat
I (520) cpu_start: App version:      1
I (524) cpu_start: Compile time:     Apr  7 2021 22:40:38
I (530) cpu_start: ELF file SHA256:  653b92bcbc2c0bc0...
I (536) cpu_start: ESP-IDF:          v4.4-dev-744-g1cb31e509
I (543) heap_init: Initializing. RAM available for dynamic allocation:
I (550) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (556) heap_init: At 3FFB3B40 len 0002C4C0 (177 KiB): DRAM
I (562) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (568) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (575) heap_init: At 4008EDD0 len 00011230 (68 KiB): IRAM
I (582) spi_flash: detected chip: generic
I (586) spi_flash: flash io: dio
I (591) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (607) spi_bus: SPI2 bus created
I (610) Board: Board Info: ESP32-LCDKit
I (614) Board: Board Init Done ...
I (619) spi_bus: SPI2 bus device added, CS=5 Mode=0 Speed=20000000
I (1127) lcd ili9341: MADCTL=28
I (1127) spi_bus: SPI2 bus device added, CS=32 Mode=0 Speed=2000000
I (1127) XPT2046: Touch panel size width: 240, height: 320
I (1133) XPT2046: Initial successful | GPIO INT:-1 | GPIO CS:32 | dir:4
I (1409) lcd ili9341: MADCTL=28
I (2388) lcd ili9341: MADCTL=8
I (12618) Touch calibration: [0] X:2931 Y:3386
E (12648) lcd ili9341: C:/Development/esp/esp-iot-solution-master/components/display/screen/controller_driver/ili9341/ili9341.c:248 (lcd_ili9341_set_window):The set coordinates exceed the screen size
E (12655) lcd ili9341: C:/Development/esp/esp-iot-solution-master/components/display/screen/controller_driver/ili9341/ili9341.c:248 (lcd_ili9341_set_window):The set coordinates exceed the screen size
E (12674) lcd ili9341: C:/Development/esp/esp-iot-solution-master/components/display/screen/controller_driver/ili9341/ili9341.c:248 (lcd_ili9341_set_window):The set coordinates exceed the screen size
E (12692) lcd ili9341: C:/Development/esp/esp-iot-solution-master/components/display/screen/controller_driver/ili9341/ili9341.c:248 (lcd_ili9341_set_window):The set coordinates exceed the screen size
E (12711) lcd ili9341: C:/Development/esp/esp-iot-solution-master/components/display/screen/controller_driver/ili9341/ili9341.c:248 (lcd_ili9341_set_window):The set coordinates exceed the screen size
E (12729) lcd ili9341: C:/Development/esp/esp-iot-solution-master/components/display/screen/controller_driver/ili9341/ili9341.c:248 (lcd_ili9341_set_window):The set coordinates exceed the screen size
E (12747) lcd ili9341: C:/Development/esp/esp-iot-solution-master/components/display/screen/controller_driver/ili9341/ili9341.c:248 (lcd_ili9341_set_window):The set coordinates exceed the screen size
E (12766) lcd ili9341: C:/Development/esp/esp-iot-solution-master/components/display/screen/controller_driver/ili9341/ili9341.c:248 (lcd_ili9341_set_window):The set coordinates exceed the screen size
E (12784) lcd ili9341: C:/Development/esp/esp-iot-solution-master/components/display/screen/controller_driver/ili9341/ili9341.c:248 (lcd_ili9341_set_window):The set coordinates exceed the screen size
E (12803) lcd ili9341: C:/Development/esp/esp-iot-solution-master/components/display/screen/controller_driver/ili9341/ili9341.c:248 (lcd_ili9341_set_window):The set coordinates exceed the screen size
```

## Other items if possible

[sdkconfig.txt](https://github.com/espressif/esp-iot-solution/files/6276260/sdkconfig.txt)

alic-maker, 2021-04-08:
@ogghst It looks like the LCD function was called somewhere with an incorrect coordinate parameter. If we find the wrong parameter, we may find the cause of the problem. 

I don't know how you modify the code, so please provide the complete code.

ogghst, 2021-08-14:
my fault, fixed

closed 2021-08-14 by ogghst

kylix34, 2022-03-02:
@ogghst I have the same roblem with the default example code. Would you be so kind to tell me how did you solve the problem?