espressif/esp-iot-solution #102: Support for TK043F1569 (16 bit parallel RGB mode)

ammaree, 2021-04-16:
## Problem Description

I recently purchased 2x displays with adapters to use with the ESP32 DevKit V4 (WROVER-IE) and the Espressif LCDKit v1.1
Based on advice from EspressIf (TK043F1508 ) and from users (link below) I ordered what I thought was the correct module.
https://github.com/espressif/esp-iot-solution/issues/89#issue-674226075

I purchased the display modules from the following link:
https://buy.tbfocus.com/item.php?id=560872117142#3512775208514

Upon receiving the modules I find that the pins are (slightly different) but more important that the controller has changed from NT35510 or RM68120 to LG4572

### Expected Behavior
Any chance to get a driver adapted to support the LG4572 when used together with the DevKit and LCDKit mentioned above?

Photo of modules I purchased:
[image]

Datasheet of controller:
[Datasheet.hk_lg4572b_8093633.pdf](https://github.com/espressif/esp-iot-solution/files/6325789/Datasheet.hk_lg4572b_8093633.pdf)

ammaree, 2021-04-21:
@projectgus @mahavirj 

Not being and LCD/LED/display technology expert I am unable to evaluate whether this display can work at all, or not. Can anybody possibly advise on the following questions, it will help us make a decision whether to abort the project and restart with different controllers.

Q1. Is it technically feasible to write a display driver to use this controller with the ESP32 DevKit V4 (WROVER-IE) and the Espressif LCDKit v1.1 ?

Q2. If possible, can you recommend the optimal starting point (existing driver) to use as foundation for developing an LG4572B driver.

Q3. If not, what makes the specific LG4572B controller incompatible with the  ESP32 DevKit V4 (WROVER-IE) and the Espressif LCDKit v1.1 ?

JimNickerson, 2021-04-21:
Have you seen this  https://github.com/espressif/esp-dev-kits/blob/master/esp32-s2-hmi-devkit-1/README_en.md

JimNickerson, 2021-04-21:
I wonder if the Espressif LCD Kit refered to waas this https://github.com/espressif/esp-dev-kits/blob/master/esp32-lcdkit/docs/ESP32_LCDKit_guide_en.md

ammaree, 2021-04-21:
@JimNickerson 
Yes, I saw the link to the HMI devout, but not sure how it is relevant to my situation?

ammaree, 2021-04-21:
@JimNickerson 
Re the link to the LCDKit, yes that is the version (v1.1) that I have bought. But my problem related to the supposed to be compatible (but now incompatible?) LG4572 based module I was supplied?

JimNickerson, 2021-04-21:
I see no mention of the "LG4572 " controller in any of the Espressif examples.

JimNickerson, 2021-04-21:
That controller chip is difficult to find documentation for, I did find this http://www.datasheet.hk/view_download.php?id=2044705&file=0523%5Clg4572b_8093633.pdf

ammaree, 2021-04-21:
Correct, I already have the data sheet, it is attached at the bottom of my first post here, below the picture.

JimNickerson, 2021-04-21:
so you have, with that and the example for the other controller used in the Espressif example you can begin to write some code

suda-morris, 2021-04-22:
Maybe the NT35510 or RM68120 are around to EOL so they changed to other LCD controller chips I guess. The adapter board you bought might not be pin-to-pin to Espressif LCDKit v1.1, better to connect your LCD to  ESP32 DevKit with jumper wires. 
For most of the LCD controllers, you only have to pay attention to the initialize code provided by OEM (not from Espressif), and replace NT35510 with that one.

alen1359, 2021-05-17:
I have the same issue, did anybody figured it out?

ammaree, 2021-05-20:
@alen1359 no news yet, deadly silence.

MH-GoL, 2022-01-13:
@ammaree @alen1359
I've gathered some information on https://github.com/MH-GoL/ESP32-S3-LCD.

TK043F1508:
First Problem is that it look like the TK043F1508 was produced in two versions, one with the NT35510 and one with the RM68120. The pinout of the two Displays should be the same (IMHO) but they have different Intialcodes (you can find them in the link above).
Please check the Adapterboard for the TK043F1508 for right IM configuration for 80-series 16-bit (its the same for NT35510 and RM68120): IM0=**1**, IM1=**0**, IM2=**0**, IM3=**0** (or NC?)

TK043F1569:
Compare the pinout of TK043F1508 and TK043F1569 (picture in Link above) shows that the IM Configuration on TK043F1569 is not useable.
Adapterboard configuration for TK043F1569: IM0=NC, IM1=NC, IM2=NC, IM3=NC.
The Datasheet of the TK043F1569 show RGB Interface for Input Data.
But the Datasheet also show: "This module can be easily
accessed by micro-processor-unit (MPU) via parallel 80-system interface and RGB interface"
I have some sample code for the TK499 Development Board here were i extracted the LG4572B_INITIAL_CODE from.
I try to flash the board with the binary file to run the TK043F1569 on it.
If this is working then we have working sample code for "translating" to the esp-idf.

ammaree, 2022-01-15:
@MH-GoL 
Any progress with your test?

MH-GoL, 2022-01-15:
@ammaree Yes, i had a little fight to compile the samplecode in uVision but now its running.
The TK043F1569 is hardwired to SPI Mode (5.5 in the LG4572B [Datasheet](https://github.com/MH-GoL/ESP32-S3-LCD/blob/main/docs/LG4572B_DS_V1.0.4.pdf)) and in this mode it can only run MIPI DPI/DSI.
I'm not 100% sure but i think only the ESP32-S3 can run MIPI DPI-2 (VSYNC, HSYNC, DE, PCLK, DB[15:0]) via esp_lcd_rgb_panel (Am i right? @suda-morris). For me it is good because my target is the ESP32-S3 but for you that are bad news (sorry) your chance is to move to the TK043F1508 or find a display were the IM[3:0] of the LG4572B are accessible.
If you can confirm this then i think you can close this issue.

ammaree, 2022-02-01:
@MH-GoL @suda-morris 

My key requirements are to use the TK043F1569 displays as purchased, as well as the LCDKIT v1.1. This will enable me to have a stable & reliable dev platform without a bunch of wires.
 
If changing from the current ESP32-DevKitCV[I/IE] to an ESP32-S3 based DevKit, provided it will work with the LCDKIT v1.1, I will be happy to change the devkit.

Will a devkit change make the TK043F1569 and LCDKIT v1.1 combination work, if so where can I find a compatible devkit?

MH-GoL, 2022-02-01:
@ammaree It is not possible to run the the TK043F1569 only using the pin header, you need to connect a cable to the connector for DE, PCLK, HSYNC and VSYNC. This is due to the fact that this board is also used for other displays. The only way to change this is to create a custom Board. I have the plan to create a Board with a ESP32-S3-WROOM1 on it were the LCD and CT are directly connected to but right now there is no time for this project.
The LCDKIT v1.1 is created for the ESP32-DevKitC this has a form factor that is not compatible (IMHO) with the ESP32-S3-DevKitC-1. Right now there is no easy plug and play for using the TK043F1569.

ammaree, 2022-02-16:
Thanks for the feedback, sadly the money is wasted on 2 high spec but unusable displays.
Anybody can/want them, welcome to have if you willing to pay the shipping

closed 2022-02-16 by ammaree