espressif/esp-iot-solution #104: "Touch Sensor ‎Application Note" Link broken

Zeni241, 2021-05-15:
I am on this page(] https://docs.espressif.com/projects/esp-idf/en/v4.2/esp32/api-‎reference/peripherals/touch_pad.html ]). All the links on this page (“ESP32 Technical Reference Manual ‎‎(PDF)”, “Touch Sensor Application Note” and “Guide for ESP32-Sense Development Kit)” are broken. ‎Please correct these ASAP because these documents are very important, I am specially interested in  “**Touch Sensor ‎Application Note**”.‎

TDA-2030, 2021-05-15:
@Zeni241 The documents of the ESP32-Sense board have been moved to the repository [esp-dev-kits](https://github.com/espressif/esp-dev-kits/tree/master/esp32-sense-kit/docs). All documents about the touch sensor have deleted on master branch of esp-iot-solution, and you can read them on release/v1.0 branch.

For example, the URL of **"Touch Sensor ‎Application Note"** in ESP-IDF Programming Guide is 
```
https://github.com/espressif/esp-iot-solution/blob/master/documents/touch_pad_solution/touch_sensor_design_en.md
```

You can read it by modify the URL to
```
https://github.com/espressif/esp-iot-solution/blob/release/v1.0/documents/touch_pad_solution/touch_sensor_design_en.md
```

I hope this piece of suggestion will be useful to you.

Zeni241, 2021-05-15:
OK. Thank you. 
A suggestion. Can you redirect to these pages from old URL instead of 404. It may help many.

TDA-2030, 2021-05-25:
@Zeni241 I have brought this problem to the attention of my colleagues

closed 2023-07-10 by leeebo