espressif/esp-iot-solution #106: iot_tp_set_serial_trigger() trigger_thres_sec 延时 BUG

larryli, 2021-06-24:
准确的说应该是 [touch_pad_set_filter_period()](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/touch_pad.html?highlight=touch_pad_set_filter_period#_CPPv427touch_pad_set_filter_period8uint32_t) 的 BUG。

当 `new CTouchPad()`/`iot_tp_create()` 超过两个时，`touch_pad_set_filter_period(TOUCHPAD_FILTER_TOUCH_PERIOD)` 实际 callback 间隔并不是 `TOUCHPAD_FILTER_TOUCH_PERIOD` 默认值 `10`，而是 `new CTouchPad()`/`iot_tp_create()`  数量 `NUM / 2 * 10`。所以就会导致 ESP32-Sense Kit 初始化 `6` 个弹簧按键，`iot_tp_set_serial_trigger(1)` 时激发间隔是 `3s`。

相关环境：
- esp-idf 版本：v4.0.2
- esp-iot-solution release/v1.1 分支
- [touch_pad_evb](https://github.com/espressif/esp-iot-solution/tree/release/v1.1/examples/touch_pad_evb) 示例
- ESP32-Sense Kit 开发板 V3.2

修改 `evb_touch_spring.cpp`：
```c++
static void spring_trigger_cb(void *arg)
{
    CTouchPad* tp_dev = (CTouchPad*) arg;
    touch_pad_t tp_num = tp_dev->tp_num();
    ESP_LOGI(TAG, "trigger callback of touch pad num %d", tp_num);
}

void evb_touch_spring_init()
{
    // ...
    tp_dev[5]->set_serial_trigger(1, 500, spring_trigger_cb, tp_dev[5]);
    // ...
}
```

Log 输出：
```
I (745) evb_main: components init...
I (755) evb_adc: ADC read voltage: 142 mV
I (755) evb_adc: Daughter Board is Spring Button
I (3195) touch_spring: push callback of touch pad num 2
I (3195) touch_spring: spring tp evt[2]
I (6195) touch_spring: trigger callback of touch pad num 2
I (6705) touch_spring: trigger callback of touch pad num 2
I (7195) touch_spring: trigger callback of touch pad num 2
I (7695) touch_spring: trigger callback of touch pad num 2
I (8205) touch_spring: trigger callback of touch pad num 2
```

**原本设置的延时 1 秒变成 3 秒。**

如果注释掉前 `4` 个按键初始化：
```c++
void evb_touch_spring_init()
{
    //single touch
    // tp_dev[0] = new CTouchPad(TOUCH_BUTTON_0, SPRING_BUTTON_MAX_CHANGE_RATE_0);
    // tp_dev[1] = new CTouchPad(TOUCH_BUTTON_1, SPRING_BUTTON_MAX_CHANGE_RATE_1);
    // tp_dev[2] = new CTouchPad(TOUCH_BUTTON_2, SPRING_BUTTON_MAX_CHANGE_RATE_2);
    // tp_dev[3] = new CTouchPad(TOUCH_BUTTON_3, SPRING_BUTTON_MAX_CHANGE_RATE_3);
    tp_dev[4] = new CTouchPad(TOUCH_BUTTON_4, SPRING_BUTTON_MAX_CHANGE_RATE_4);
    tp_dev[5] = new CTouchPad(TOUCH_BUTTON_5, SPRING_BUTTON_MAX_CHANGE_RATE_5);

    // tp_dev[0]->add_cb(TOUCHPAD_CB_PUSH, spring_push_cb, tp_dev[0]);
    // tp_dev[1]->add_cb(TOUCHPAD_CB_PUSH, spring_push_cb, tp_dev[1]);
    // tp_dev[2]->add_cb(TOUCHPAD_CB_PUSH, spring_push_cb, tp_dev[2]);
    // tp_dev[3]->add_cb(TOUCHPAD_CB_PUSH, spring_push_cb, tp_dev[3]);
    tp_dev[4]->add_cb(TOUCHPAD_CB_PUSH, spring_push_cb, tp_dev[4]);
    tp_dev[5]->add_cb(TOUCHPAD_CB_PUSH, spring_push_cb, tp_dev[5]);

    tp_dev[5]->set_serial_trigger(1, 500, spring_trigger_cb, tp_dev[5]);
    // ...
}
```

Log 输出：
```
I (745) evb_main: components init...
I (755) evb_adc: ADC read voltage: 142 mV
I (755) evb_adc: Daughter Board is Spring Button
I (2775) touch_spring: push callback of touch pad num 2
I (2775) touch_spring: spring tp evt[2]
I (3775) touch_spring: trigger callback of touch pad num 2
I (4275) touch_spring: trigger callback of touch pad num 2
I (4775) touch_spring: trigger callback of touch pad num 2
I (5275) touch_spring: trigger callback of touch pad num 2
I (5775) touch_spring: trigger callback of touch pad num 2
```

正常。

[patch.txt](https://github.com/espressif/esp-iot-solution/files/6708083/patch.txt)

leeebo, 2021-07-28:
@larryli 由于该代码库 touch 读取原始数据的 API 是阻塞的，会导致采集较多通道时，延时比较长。目前这部分代码可能无法及时维护，目前建议是使用 ESP32-S2 做 Touch 功能

larryli, 2021-11-09:
@leeebo ESP32-S3 年底可以正式出货不？ESP-IDF v4.4 明年上半年能发布么？

C3 没有触摸，S2 没蓝牙。就只有 S3 可以替换。

leeebo, 2021-11-09:
@larryli 商务问题请联系 https://www.espressif.com/zh-hans/contact-us/sales-questions