espressif/esp-iot-solution #108: Contribution: faster    painter_draw_char()

jonshouse1, 2021-07-14:
I don't really know how to use github well.

I've been using the basic painter routines to make a simple GUI, but i've found the font code too slow.

I've written a faster font routine for "components/display/touch_panel/calibration/basic_painter.c" could somebody add it to the project and test it a little more.
Thanks. 

```
void painter_draw_char(int x, int y, char ascii_char, const font_t *font, uint16_t color)
{
   PAINTER_CHECK(ascii_char >= ' ', "ACSII code invalid");
   PAINTER_CHECK(NULL != font, "Font pointer invalid");
   int i, j;
   uint16_t char_size = font->Height * (font->Width / 8 + (font->Width % 8 ? 1 : 0));
   unsigned int char_offset = (ascii_char - ' ') * char_size;
   const unsigned char *ptr = &font->table[char_offset];
   uint16_t* buf = malloc(font->Width*font->Height*sizeof(uint16_t));
   int ox=0;
   int oy=0;

   for (i=0;i<font->Width*font->Height;i++)
      buf[i]=g_back_color;

   for (j = 0; j < char_size; j++) {
      uint8_t temp = ptr[j];
      for (i = 0; i < 8; i++) {
         if (temp & 0x80)
              buf[ox+(font->Width*oy)]=g_point_color;
         else buf[ox+(font->Width*oy)]=g_back_color;

         temp <<= 1;
         ox++;
         if (ox == font->Width)
         {
            ox=0;
            oy++;
            break;
         }
     }
   }
   g_lcd.draw_bitmap(x, y, font->Width, font->Height, buf);                      // Draw NxN char
   free(buf);
}
```

alic-maker, 2021-07-15:
Thanks for your contribution. The `components/display/touch_panel/calibration/basic_painter.c` is not for users, just to show some calibration information for the touch panel, so the speed is not critical. Allocating memory for each character maybe causes too much fragmentation. You can integrate this code into your own project if you need it.

jonshouse1, 2021-07-15:
It really does make a difference, the calibration becomes noticeably faster to use if nothing else.

What  about something like this then ?  
The largest font currently is font24 with 17x24 pixels per character.

```
void painter_draw_char(int x, int y, char ascii_char, const font_t *font, uint16_t color)
{
        PAINTER_CHECK(ascii_char >= ' ', "ACSII code invalid");
        PAINTER_CHECK(NULL != font, "Font pointer invalid");
        int i, j;
        uint16_t char_size = font->Height * (font->Width / 8 + (font->Width % 8 ? 1 : 0));
        unsigned int char_offset = (ascii_char - ' ') * char_size;
        const unsigned char *ptr = &font->table[char_offset];
        static uint16_t buf[18*25*sizeof(uint16_t)];
        int ox=0;
        int oy=0;

        for (i=0;i<font->Width*font->Height;i++)
                buf[i]=painter_get_back_color();
        for (j = 0; j < char_size; j++)
        {
                uint8_t temp = ptr[j];
                for (i = 0; i < 8; i++)
                {
                        if (temp & 0x80)
                                buf[ox+(font->Width*oy)]=painter_get_point_color();

                        temp <<= 1;
                        ox++;
                        if (ox == font->Width)
                        {
                                ox=0;
                                oy++;
                                break;
                        }
                }
        }
        g_lcd.draw_bitmap(x, y, font->Width, font->Height, (uint16_t*)&buf);          // Draw NxN char
}
```

alic-maker, 2021-07-19:
I will consider incorporating your code.

jonshouse1, 2021-07-22:
Great new, thanks.

jonshouse1, 2021-07-28:
If you do use it i've noticed

```
static uint16_t buf[18*25*sizeof(uint16_t)];
```

Should be:
```
uint16_t buf[18*25*sizeof(uint16_t)];
```

To make it safe if people use it externally from multiple tasks.

alic-maker, 2021-07-29:
I'm still testing with other improvements

commit 68bfe795 (2021-08-23): touch_panel: revise calibration prompt; closes #108, closes #109
closed 2021-08-23 by zhanzhaocheng