espressif/esp-iot-solution #110: Ethernet works,  SPI display works, but using the two together causes display corruption

jonshouse1, 2021-07-26:
[https://www.olimex.com/Products/IoT/ESP32/ESP32-POE/open-source-hardware
](url)

I'm using an Olimex POE board with wired network together with the esp-iot-solution using the ILI9341 driver.

LCD works fine on its own.  Networking works fine on its own.

But this (VNC client code i'm writing) produces an 80% ok screen with a number of  corrupted lines

```
    if (rec.width*2 > sizeof(pixels))
        ESP_LOGE(TAG,"Proable buffer overrun!");
    len = recv(vncc_sock, (char*)&pixels, rec.width*2, 0);  // One lines worth of pixel data
    jag_draw_icon(rec.xpos, rec.ypos+l, rec.width, 1, (char*)&pixels);
```

But If I fill the pixel buffer with a color and comment out the recv() then the display is fine:

```
    if (rec.width*2 > sizeof(pixels))
        ESP_LOGE(TAG,"Proable buffer overrun!");
    //len = recv(vncc_sock, (char*)&pixels, rec.width*2, 0);  // One lines worth of pixel data
    jag_draw_icon(rec.xpos, rec.ypos+l, rec.width, 1, (char*)&pixels);
```

The jag_draw_icon() function is called for each line of 240 pixels, this in turn calls the LCD drivers draw_bitmap()
If I put a delay as the first line of the jag_draw_icon() function then it seems to mostly fix it. Is this an interaction between drivers or something i've overlooked?

```
// online line at a time, stack buffer, probably thread safe
void jag_draw_icon_s2(uint16_t x, uint16_t y, uint16_t w, uint16_t h, const char *image)
{
        uint16_t        buf[JAG_MAXPIXELS_PERLINE];                             // stack allocated buffer, thread safe 
        uint16_t        l   = 0;
        uint16_t        bytesperline=0;

        bytesperline = w*sizeof(uint16_t);
        for (l=0;l<h;l++)                                                       // for every line of image
        {
                memcpy(&buf, image+(bytesperline*l), bytesperline);             // copy one line of pixels from flash to RAM
                jag_draw_bitmap(x, y+l, w, 1, (uint16_t*)&buf);                 // draw line of pixels
        }
}

void jag_draw_icon(uint16_t x, uint16_t y, uint16_t w, uint16_t h, const char *image)
{
      ets_delay_us(250);
      ag_draw_icon_s2(x, y, w, h, image);  
}

```

Any help appreciated, thanks.

alic-maker, 2021-07-28:
There seems to be no problem. I'm not sure, but I still want you to try.
modify
```
uint16_t        buf[JAG_MAXPIXELS_PERLINE]; 
```
to 
```
static uint16_t        buf[JAG_MAXPIXELS_PERLINE]; 
```
Maybe the function jag_draw_bitmap is not blocking execution

jonshouse1, 2021-07-28:
Not sure I understand what you are doing with 'static', that would create a single buffer that would corrupt if the function was called reentrantly. I will try it anyway as only the vnc process is driving the display and only from a single task. 

Ok, tried it.....

I commented out the delay and declared buf as static.  The display corrupts, I periodically get fragments of pixels the expected size but in the wrong locations, looks like the set_window part of the display transaction has not always been properly decoded.

I put the delay back in and that fixes it,  the delay is key to fixing the issue hence my suspicion of a driver problem, or maybe some limitation of the hardware I do not understand. It is difficult to make a test case for it as the issue only happens when both Ethernet and the SPI display are active at the same time. 

I have my VNC client code working (mostly) now, I will try and upload the project to github later and put a link here.

alic-maker, 2021-07-28:
This is really a strange problem, which may also have something to do with the hardware

jonshouse1, 2021-07-28:
Not sure how DMA on the ESP works but I would be willing to bet 250us has something to do with a length of some transaction somewhere, maybe the driver maintainers could have a look at it.  Does the esp-iot-solution SPI driver and the Ethernet driver in the idf play nicely together?

jonshouse1, 2021-07-28:
https://github.com/jonshouse1/ESPVNCC/

Code is here now

alic-maker, 2021-08-02:
@jonshouse1 
I connected LAN8720 and LCD with wires for testing. When VNC is not connected, the display is normal. 
**Screenshot**
[image]

I tried to connect to VNC, but I received following error message:

```
I (22892) lcdtouchvnc: Trying 192.167.0.100:5901
I (22892) lcdtouchvnc: Got connection
vncc_state=1
I (22892) lcdtouchvnc: Successfully connected RFB 003.008

vncc_state=2
security types, got 3 bytes
02 13 02 
vncc_state=4
vncc_state=0
I (22902) lcdtouchvnc: Trying 192.167.0.100:5901
I (22912) lcdtouchvnc: Got connection
vncc_state=1
I (22912) lcdtouchvnc: Successfully connected RFB 003.008

vncc_state=2
security types, got 3 bytes
02 13 02 
vncc_state=4
vncc_state=0
I (22922) lcdtouchvnc: Trying 192.167.0.100:5901
I (22932) lcdtouchvnc: Got connection
vncc_state=1
I (22932) lcdtouchvnc: Successfully connected RFB 003.008

vncc_state=2
security types, got 3 bytes
02 13 02 
vncc_state=4
vncc_state=0
I (22952) lcdtouchvnc: Trying 192.167.0.100:5901
I (22952) lcdtouchvnc: Got connection
vncc_state=1
I (22962) lcdtouchvnc: Successfully connected RFB 003.008

vncc_state=2
security types, got 3 bytes
02 13 02 
vncc_state=4
vncc_state=0
I (22972) lcdtouchvnc: Trying 192.167.0.100:5901
I (22982) lcdtouchvnc: Got connection
vncc_state=1
I (22982) lcdtouchvnc: Successfully connected RFB 003.008

vncc_state=2
security types, got 3 bytes
02 13 02 
vncc_state=4
vncc_state=0
I (23002) lcdtouchvnc: Trying 192.167.0.100:5901
I (23002) lcdtouchvnc: Got connection
vncc_state=1
E (23012) lcdtouchvnc: expected 12 bytes, got 46
I (23012) lcdtouchvnc: Successfully connected RFB 003.003

vncc_state=2
security types, got -1 bytes

vncc_state=4
vncc_state=0
I (23032) lcdtouchvnc: Trying 192.167.0.100:5901
I (23032) lcdtouchvnc: Got connection
vncc_state=1
E (23042) lcdtouchvnc: expected 12 bytes, got 46
I (23042) lcdtouchvnc: Successfully connected RFB 003.003

vncc_state=2
security types, got -1 bytes

vncc_state=4
vncc_state=0
I (23052) lcdtouchvnc: Trying 192.167.0.100:5901
I (23062) lcdtouchvnc: Got connection
vncc_state=1
E (23062) lcdtouchvnc: expected 12 bytes, got 46
I (23072) lcdtouchvnc: Successfully connected RFB 003.003

vncc_state=2
security types, got -1 bytes

vncc_state=4
vncc_state=0
I (23082) lcdtouchvnc: Trying 192.167.0.100:5901
I (23092) lcdtouchvnc: Got connection
```
I haven't found any display problems from beginning to end. When did the display abnormality you said happen? Can you show some pictures when the display is damaged?

jonshouse1, 2021-08-02:
Interesting, sorry you are having problems with my code, i've only tested on one bit of ESP hardware and one VNC server.

Did you start the VNC server like this ? "Xvnc :1 -geometry 240x320 -depth 16"  My code will fail if the VNC server requires authentication, most of wrappers that start it will turn authentication on.

It could be a timing isssue. Try adding a delay before the recv() in "case VNCC_EXPECTING_NUM_SECURITY_TYPES:"
My output looks like this:
```
vncc_state=0
I (5562) lcdtouchvnc: Trying 192.168.1.111:5901
I (5572) lcdtouchvnc: Got connection
vncc_state=1
I (5572) lcdtouchvnc: Successfully connected RFB 003.008

vncc_state=2
security types, got 3 bytes
02 01 10 
vncc_state=4
security result (should be 00 00 00 00) = 00 00 00 00 
vncc_state=5
read server init, got 24 bytes, expecting 24
00 F0 01 40 10 10 00 01 00 1F 00 3F 00 1F 0B 05 00 00 00 00 00 00 00 1C 
read name, got 28 bytes
width=240
height=320
bpp=16
depth=16
namelen=28
```
If I comment out the ets_delay_us() I get this result, with it I see no corruption

The text output typically does not corrupt as mostly ethernet is not active when the text is rendered  (or something, still not sure exactly what is going on to cause the corruption).
I've been looking to see if I have any sort of memory corruption with the code, not found any yet but I was working on the assumption that the lock issue may be stack corruption if it is not a driver issue.

Thanks for looking at this

[image]

jonshouse1, 2021-08-02:
Just noticed your code gets to "state 4" (VNCC_EXPECTING_SERVER_INIT) and fails silently. Looks like the length of the server init is not the expected length.

Try changing line 602 to
```
   len=readbytes(vncc_sock, (char*)&vncc_rxbuf, sizeof(struct vnc_ServerInit));
```
https://github.com/jonshouse1/ESPVNCC/blob/main/main/lcd_vncc.c

alic-maker, 2021-08-02:
I found a screen like yours, and things became more strange when I connected it. Even if I commented out all the codes of ethernet and added delay in `jag_draw_bitmap()`, the display still looks incorrect. It seems that the problem of display has nothing to do with ethernet.

And when I commented out `led_pwm_init()` and just output high level to LED pin of LCD module, display looks good. I think you can also try to do this and test the display part of the program separately to narrow the scope of the problem as much as possible.

## before commented out `led_pwm_init()`
[image]
## after
[image]

jonshouse1, 2021-08-02:
I modified lcd_init()

```
// Inil lcd display and touch screen
void lcd_init()
{
        int p                           = TRUE;                         // default to portrait
        spi_bus_handle_t                bus_handle;
        scr_interface_driver_t          *iface_drv;

        //led_pwm_init();
        //led_pwm_set(255);                                             // turn backlight on
ESP_LOGI(TAG,"without PWM");
        gpio_set_direction(GPIO_LCDBL, GPIO_MODE_OUTPUT);
        gpio_set_level(GPIO_LCDBL, 1);  //on

        spi_config_t            bus_conf =
        {
                .miso_io_num    = GPIO_MISO,
                .mosi_io_num    = GPIO_MOSI,
                .sclk_io_num    = GPIO_SCLK,
                //.max_transfer_sz= 32,
        };
        //bus_handle = spi_bus_create(SPI3_HOST, &bus_conf);            // or SPI2_HOST ?
        bus_handle = spi_bus_create(SPI2_HOST, &bus_conf);              // or SPI2_HOST ?
       ............. and so on...........
```

Checked the output (make sure I was running the modified code)
```
I (389) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (398) lcdtouchvnc: without PWM
```

Still corrupts

[image]

jonshouse1, 2021-08-02:
I've modified my code so that it can use either Wifi or Ethernet (will sync it with github after I tidy it up)

Without the added delay Ethernet corrupts the display, Wifi does not.   PWM is still commented out.

In summary:

No PWM  +  Ethernet + Display  = corrupt display
No PWM  +  Wifi  + Display = works fine.

I've modified lcd_vncc.c to use readbytes() instead of recv() for the VNCC_EXPECTING_SERVER_INIT: I seem remember that the module you are using for ethernet always returns data in small chunks, I suggest you do the same modification and see if the VNC client then works, I will try and add some better feedback in the output text.

```
                        case VNCC_EXPECTING_SERVER_INIT:
                                vTaskDelay(1000 / portTICK_PERIOD_MS);
                                len = readbytes(vncc_sock, (char*)&vncc_rxbuf, sizeof(struct vnc_ServerInit));
```

Can you confirm what IC you are using on the Ethernet module, I can try and duplicate what you have.

jonshouse1, 2021-08-02:
Please try the latest from github.  I've re-written the VNC negotiation slightly, with luck it should now work for you now, let me know how it goes?

If it still does not connect try reducing cs=512 to cs=64 line 102, but that should not be needed.
https://github.com/jonshouse1/ESPVNCC/blob/main/main/lcd_vncc.c

To replicate my display corruption make sure line 19 is commented out  (#define USE_WIFI)
https://github.com/jonshouse1/ESPVNCC/blob/main/main/lcdtouchvnc.c

Comment out line 38  (#define DRIVER_BUG)
https://github.com/jonshouse1/ESPVNCC/blob/main/main/jag.c

That combination will compile for Ethernet with no ets_delay_us().

alic-maker, 2021-08-03:
> Can you confirm what IC you are using on the Ethernet module, I can try and duplicate what you have.

The IC I use is the lan8720 mentioned earlier. the [module](https://www.ebay.com/itm/164409966246?hash=item264798fea6:g:rpYAAOSwsNFgm4c5) manufactory is Waveshare

closed 2023-07-10 by leeebo