espressif/esp-iot-solution #111: about USB CDC 4G Module 示例程序

lightsalt2011, 2021-07-28:
we follow  https://github.com/espressif/esp-iot-solution/tree/usb/add_usb_solutions/examples/usb/usb_cdc_4g_module project, build pass, 

we connect CAT.1 device(EC600S-CN), but the AT fails 

error log as below:
> I (987) USB_HCDC: Pipe Default Created
I (987) USB_HCDC: 1. Set Device Addr = 1
I (989) USB_HCDC: Set Device Addr Done
I (993) USB_HCDC: 2. Sending set_config = 1
I (998) USB_HCDC: Config is set
I (1002) USB_HCDC: 3. Sending set_line state itf= 0 dtr=1 rts=0
I (1008) USB_HCDC: Line state is set
I (1013) USB_HCDC: Creating bulk out pipe
I (1018) USB_HCDC: Creating bulk in pipe
W (1018) USB_HCDC: line 508 Pipe: bulk_out HCD_PIPE_EVENT_ERROR_XFER
E (1176) esp-modem-dte: esp_modem_dte_send_cmd(192): process command timeout
 response timeoutm_dce: esp_modem_dce_generic_command(31): Command:AT
I (1181) modem_board: Current timeouts: 1 and errors: 0
E (6187) esp-modem-dte: esp_modem_dte_send_cmd(192): process command timeout
W (6187) esp_modem_dce: esp_modem_dce_generic_command(31): Command:+++ response timeout
E (6692) esp-modem-dte: esp_modem_dte_send_cmd(192): process command timeout
 response timeoutm_dce: esp_modem_dce_generic_command(31): Command:AT
E (7197) esp-modem-dte: esp_modem_dte_send_cmd(192): process command timeout
 response timeoutm_dce: esp_modem_dce_generic_command(31): Command:AT
I (7202) modem_board: Current timeouts: 2 and errors: 0
I (7208) modem_board: Restart to connect modem........
I (7214) modem_board: modem_board_reset!
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x4002583c

the AT command timeout, how can I fix it? 

and how can I get the 4G 模组 USB CDC 输入端点 （IN）和 输出端点（OUT） 地址 ?

lightsalt2011, 2021-07-28:
another, can you provide the  purchase link of "中移 ML302" , we can verify the 4G PPPoE function firstly, and then we can use EC600S-CN for production, thanks

lightsalt2011, 2021-07-28:
> static esp_err_t esp_modem_dte_send_cmd(esp_modem_dte_t *dte, const char *command, uint32_t timeout)
{
    esp_err_t ret = ESP_FAIL;
    esp_modem_dce_t *dce = dte->dce;
    ESP_MODEM_ERR_CHECK(dce, "DTE has not yet bind with DCE", err);
    ESP_MODEM_ERR_CHECK(command, "command is NULL", err);
    esp_modem_dte_internal_t *esp_dte = __containerof(dte, esp_modem_dte_internal_t, parent);
    /* Calculate timeout clock tick */
    /* Reset runtime information */
    dce->state = ESP_MODEM_STATE_PROCESSING;
    /* Send command via UART */
    usbh_cdc_write_bytes((const uint8_t*)command, strlen(command));
    /* Check timeout */
    EventBits_t bits = xEventGroupWaitBits(esp_dte->process_group, (ESP_MODEM_COMMAND_BIT|ESP_MODEM_STOP_BIT), pdTRUE, pdFALSE, pdMS_TO_TICKS(timeout));
    ESP_MODEM_ERR_CHECK(bits&ESP_MODEM_COMMAND_BIT, "process command timeout", err);
    ret = ESP_OK;
err:
    dce->handle_line = NULL;
    return ret;
}

AT command "process command timeout", due to the usb usb enumeration fail?

leeebo, 2021-07-28:
@lightsalt2011 
**error message:** `HCD_PIPE_EVENT_ERROR_XFER` seems you are using a wrong endpoint address.
**About endpoint address:** Please check the module's specifications to find which endpoint is used for PPP Dial, then check the address of the endpoint using `lsusb -v` . (You can also attach the module's specifications, etc. We will help)
**For ML302:** You can search then purchase any ML302 develop board from taobao.com, please be careful the different sub-names of ML302, like ML302-DNLM support FDD and TDD, but ML302-CNLM only support TDD.

lightsalt2011, 2021-07-28:
> Bus 001 Device 006: ID 2c7c:6002  
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass          239 Miscellaneous Device
  bDeviceSubClass         2 ?
  bDeviceProtocol         1 Interface Association
  bMaxPacketSize0        64
  idVendor           0x2c7c 
  idProduct          0x6002 
  bcdDevice            3.18
  iManufacturer           1 
  iProduct                2 
  iSerial                 3 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength          209
    bNumInterfaces          5
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower              500mA
    Interface Association:
      bLength                 8
      bDescriptorType        11
      bFirstInterface         0
      bInterfaceCount         2
      bFunctionClass          2 Communications
      bFunctionSubClass       6 Ethernet Networking
      bFunctionProtocol       0 
      iFunction               5 
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         2 Communications
      bInterfaceSubClass      6 Ethernet Networking
      bInterfaceProtocol      0 
      iInterface              5 
      CDC Header:
        bcdCDC               1.10
      CDC Ethernet:
        iMacAddress                     14 (??)
        bmEthernetStatistics    0x00000000
        wMaxSegmentSize               1544
        wNumberMCFilters            0x0001
        bNumberPowerFilters              0
      CDC Union:
        bMasterInterface        0
        bSlaveInterface         1 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x87  EP 7 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval              16
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
[... 134 lines cut ...]
        bInterval              16
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x0a  EP 10 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0

lsusb -v output is above message,  EP IN: 0x83, EP OUT:0x0c  ? 

still error like use default EP address

leeebo, 2021-07-28:
@lightsalt2011 Can you attach the product's specification?

lightsalt2011, 2021-07-28:
how about the spec,   
https://github.com/ZhiliangMa/EC600S-CN-Application-Guidance/blob/master/Quectel_EC200x%26EC600S%26EG912Y%E7%B3%BB%E5%88%97_PPP_%E5%BA%94%E7%94%A8%E6%8C%87%E5%AF%BC_V1.0.pdf

achao1104, 2021-07-29:
I have the same question, is the CDC in and out same as USB PID and VID? BTW, I am using SIM7600

leeebo, 2021-07-29:
@lightsalt2011 

>lsusb -v output is above message, EP IN: 0x83, EP OUT:0x0c ?
>still error like use default EP address

Yes, based on the above descriptor message, the address seems like EP IN: 0x83, EP OUT:0x0c. But I found the EP MPS of that is 512, while in the demo code we default MPS=64 (USB1.1 Fullspeed mode only support MPS 64 in bulk transfer, NOT Sure if EC600S-CN only support USB2.0), you can contact 4g module vendor to check that.

P.S. You can add a USB Fullspeed isolator before the module's USB connector (force the module works in USB1.1), then check the descriptor again using `lsusb -v`. (Some usb device could switch descriptors in different USB speed mode, and could have different endpoint addresses).

lightsalt2011, 2021-07-29:
thanks for ur reply. 

> while in the demo code we default MPS=64 (USB1.1 Fullspeed mode only support MPS 64 in bulk transfer

as I known, CDC protocol be supported after USB2.0,  so "USB1.1 Fullspeed mode" means USB2.0 full speed(12Mbps)  USB2.0 high speed(480Mbps) ? 

>NOT Sure if EC600S-CN only support USB2.0

I will check with EC600S-CN vendor, but I thinks it both support

>You can add a USB Full speed isolator before the module's USB connector (force the module works in USB1.1)

it is really a pity that we donot have a USB Full speed isolator device

by the way, "中移 ML302" have been booking, we will verify this function with the devices which you have verified, and then we will debug EC600S-CN for production

pls have a check if this CAT.1 you have verified or NOT, thansk
https://item.taobao.com/item.htm?spm=a230r.1.14.33.70c92151IzyptY&id=613756460633&ns=1&abbucket=15#detail

leeebo, 2021-07-29:
1. USB1.1 Fullspeed mode and USB2.0 Full speed are similar, but USB1.1 Fullspeed mode and USB2.0 High speed have much more difference, The USB defines the allowable maximum bulk data payload sizes to be only 8, 16,32, or 64 bytes for full-speed endpoints and 512 bytes for high-speed endpoints. EC600S-CN CDC Data interface only has 512-bytes endpoint, which should be high-speed only.
2. I think the ML302 module you book should work with the demo code

lightsalt2011, 2021-07-29:
thanks for ur reply, we will test when ML302 module I get. 

another,
1. when the usb_cdc_4g_module project will support USB2.0 High speed of CDC ?
2. is there any plan that support  USB CDC RNDIS protocol for esp32s2/s3 ?

leeebo, 2021-07-30:
> * when the usb_cdc_4g_module project will support USB2.0 High speed of CDC ?
> * is there any plan that support  USB CDC RNDIS protocol for esp32s2/s3 ?

1. USB2.0 High speed needs different USB hardware controller, which esp32s2/s3 don't have. 
2. No USB Host RNDIS plan now, it depends on user requirements in near future. But third-party project like [tinyusb](https://github.com/hathach/tinyusb) have support USB Device RNDIS in esp32s2/s3.

leeebo, 2021-07-30:
> I have the same question, is the CDC in and out the same as USB PID and VID? BTW, I am using SIM7600

No, they are different concepts in USB specifications. IN and OUT endpoint addresses used to distinguish different functions(one device may have multi-functions and multi-endpoints) . USB PID and VID used to distinguish different devices.

lightsalt2011, 2021-08-02:
> 1. USB1.1 Fullspeed mode and USB2.0 Full speed are similar, but USB1.1 Fullspeed mode and USB2.0 High speed have much more difference, The USB defines the allowable maximum bulk data payload sizes [...]

lightsalt2011, 2021-08-02:
use this new module https://item.taobao.com/item.htm?spm=a230r.1.14.33.70c92151IzyptY&id=613756460633&ns=1&abbucket=15#detail, we get log as below, it seems like that usb enum pass ? but I cannot visit The Internet when I connect the AP named "esp_4G_router", I am sure when I use AT cmd "AT+MPING="www.baidu.com",,1" can access the BAIDU.

>W (6059) USB_HCDC: line 609 Pipe: in HCD_PIPE_EVENT_INVALID
E (6187) esp-modem-dte: esp_modem_dte_send_cmd(192): process command timeout
W (6187) esp_modem_dce: esp_modem_dce_generic_command(31): Command:+++ response timeout
E (6692) esp-modem-dte: esp_modem_dte_send_cmd(192): process command timeout
 response timeoutm_dce: esp_modem_dce_generic_command(31): Command:AT
E (7197) esp-modem-dte: esp_modem_dte_send_cmd(192): process command timeout
 response timeoutm_dce: esp_modem_dce_generic_command(31): Command:AT
I (7202) modem_board: Current timeouts: 2 and errors: 0
I (7208) modem_board: Restart to connect modem........
I (7214) modem_board: modem_board_reset!
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x4002583c
0x4002583c: esp_restart_noos at /Users/lvfeng/trunk/esp-refs/esp-idf-repo/esp-idf/components/esp_system/port/soc/esp32s2/system_internal.c:112 (discriminator 1)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe6100,len:0x17ac
load:0x4004c000,len:0xcf8
load:0x40050000,len:0x2c5c
entry 0x4004c238
I (24) boot: ESP-IDF v4.4-dev-2184-g166c30e7b-dirty 2nd stage bootloader
I (24) boot: compile time 16:40:17
I (24) boot: chip revision: 0
I (28) qio_mode: Enabling default flash chip QIO
I (33) boot.esp32s2: SPI Speed      : 80MHz
I (38) boot.esp32s2: SPI Mode       : QIO
I (43) boot.esp32s2: SPI Flash Size : 4MB
I (48) boot: Enabling RNG early entropy source...
I (53) boot: Partition Table:
I (57) boot: ## Label            Usage          Type ST Offset   Length
I (64) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (71) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (79) boot:  2 factory          factory app      00 00 00010000 00100000
I (86) boot: End of partition table
I (91) esp_image: segment 0: paddr=00010020 vaddr=3f000020 size=19468h (103528) map
I (117) esp_image: segment 1: paddr=00029490 vaddr=3ffcbe90 size=031ech ( 12780) load
I (120) esp_image: segment 2: paddr=0002c684 vaddr=40024000 size=03994h ( 14740) load
I (126) esp_image: segment 3: paddr=00030020 vaddr=40080020 size=79b5ch (498524) map
I (217) esp_image: segment 4: paddr=000a9b84 vaddr=40027994 size=144f0h ( 83184) load
I (235) esp_image: segment 5: paddr=000be07c vaddr=50000000 size=00010h (    16) load
I (247) boot: Loaded app from partition at offset 0x10000
I (247) boot: Disabling RNG early entropy source...
I (258) cache: Instruction cache        : size 16KB, 4Ways, cache line size 16Byte
I (259) cpu_start: Pro cpu up.
I (261) cache: flash_count=1, size=16, spiram_count=0, size=0,together=0
I (268) cache: Flash wrap enabled, size = 16.
I (285) cpu_start: Pro cpu start user code
I (285) cpu_start: cpu freq: 240000000
I (285) cpu_start: Application information:
I (290) cpu_start: Project name:     cdc_4g
I (294) cpu_start: App version:      e7f7cda
I (299) cpu_start: Compile time:     Aug  2 2021 20:24:19
I (305) cpu_start: ELF file SHA256:  bfea0ab7ff726f6a...
I (311) cpu_start: ESP-IDF:          v4.4-dev-2184-g166c30e7b-dirty
I (318) heap_init: Initializing. RAM available for dynamic allocation:
I (326) heap_init: At 3FF9E000 len 00002000 (8 KiB): RTCRAM
I (332) heap_init: At 3FFD3960 len 000286A0 (161 KiB): DRAM
I (338) heap_init: At 3FFFC000 len 00003A10 (14 KiB): DRAM
I (345) spi_flash: detected chip: generic
I (349) spi_flash: flash io: qio
I (356) cpu_start: Starting scheduler on PRO CPU.
I (614) main: ====================================
I (614) main:      ESP 4G Cat.1 Wi-Fi Router
I (614) main: ====================================
I (619) gpio: GPIO[15]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (628) gpio: GPIO[17]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (637) gpio: GPIO[16]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
[... 77 lines cut ...]
I (5614) USB_HCDC: USBH CDC Ringbuffer Dump:
I (5619) USB_HCDC: usb ringbuffer size, out = 20480, in = 25600
I (5626) USB_HCDC: usb ringbuffer High water mark, out = 46, in = 46
I (8633) main: Task dump
I (8634) main: Load     Stack left      Name    PRI
I (8634) main: 3.49     1072    main    1
I (8635) main: 94.99    1256    IDLE    0
I (8639) main: 0.03     1512    bulk-out        6
I (8644) main: 0.04     2688    tiT     18
I (8648) main: 0.01     1744    Tmr Svc         1
I (8652) main: 1.22     1356    usb_event       4
I (8657) main: 0.00     1100    sys_evt         20
I (8662) main: 0.01     1520    bulk-in         5
I (8666) main: 0.05     3540    esp_timer       22
I (8671) main: 0.16     4628    wifi    23
I (8675) main: 0.00     1544    port    9
I (8680) main: 0.00     1520    dflt    8
I (8684) main: Free heap=22160 bigst=8192, internal=22040 bigst=8192
I (8691) main: ..............
I (8695) USB_HCDC: USBH CDC Transfer Buffer Dump:
I (8700) USB_HCDC: usb transfer Buffer size, out = 1536, in = 1536
I (8707) USB_HCDC: usb transfer Max packet size, out = 46, in = 46
I (8714) USB_HCDC: USBH CDC Ringbuffer Dump:
I (8719) USB_HCDC: usb ringbuffer size, out = 20480, in = 25600
I (8726) USB_HCDC: usb ringbuffer High water mark, out = 46, in = 46

leeebo, 2021-08-02:
@lightsalt2011 Which device did you use, Android or IOS or PC with Windows?

lightsalt2011, 2021-08-03:
> @lightsalt2011
> 
> 1. Please add your SIM Card Information
> 2. Which device did you use for the internet, Android or IOS or PC with Windows?
> 3. Did you get any error message in your browser?
> 4. You can try to modify your APN (in `menuconfig -> 4G Modem Configuration -> Set Modem APN`) from default `internet` to that provided by internet operator, such as `cmnet` when using China Mobile.)

**1. Please add your SIM Card Information**
CHINA TELECOM and CHINA MOBILE IoT SIM

**2. Which device did you use for the internet, Android or IOS or PC with Windows?**
Android and PC with Windows both cannot access Internet

**3. Did you get any error message in your browser?**
NAME_NOT_RESOLVED, -105;  DNS_PROBE_STARTED  DNS_PROBE_FINISHED_NO_INTERNET

**4. You can try to modify your APN (in `menuconfig -> 4G Modem Configuration -> Set Modem APN`) from default `internet` to that provided by internet operator, such as `cmnet` when using China Mobile.)**
same issue

 >I am sure when I use AT cmd "AT+MPING="www.baidu.com",,1" can access the BAIDU.

I refs this link [https://m.c114.com.cn/w118-1127343.html](url) use AT CMD with windows pc
```AT+VERCTRL?
AT+CEREG?
AT+CGACT?
AT+CGDCONT?
AT+CGDCONT=1,"IP","CMNET"
AT+CGACT=1,1
AT+CGDCONT?
AT+MPING="www.baidu.com",,2```

is there any suggestion to fix this issue ?

lightsalt2011, 2021-08-03:
more logs as below,  "ppp ip: 10.52.94.238, mask: 255.255.255.255, gw: 192.168.0.1" seems CAT.1 has already I ip,  is there some methods that can break-down PPPoE with wifi AP function ? 

I (290) cpu_start: Project name:     cdc_4g
I (294) cpu_start: App version:      e7f7cda-dirty
I (300) cpu_start: Compile time:     Aug  3 2021 10:37:11
I (306) cpu_start: ELF file SHA256:  05722da0be9d5f9d...
I (312) cpu_start: ESP-IDF:          v4.4-dev-2184-g166c30e7b-dirty
I (319) heap_init: Initializing. RAM available for dynamic allocation:
I (326) heap_init: At 3FF9E000 len 00002000 (8 KiB): RTCRAM
I (332) heap_init: At 3FFD3960 len 000286A0 (161 KiB): DRAM
I (339) heap_init: At 3FFFC000 len 00003A10 (14 KiB): DRAM
I (345) spi_flash: detected chip: generic
I (350) spi_flash: flash io: qio
I (357) cpu_start: Starting scheduler on PRO CPU.
I (615) main: ====================================
I (615) main:      ESP 4G Cat.1 Wi-Fi Router
I (615) main: ====================================
I (620) gpio: GPIO[15]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (629) gpio: GPIO[17]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (638) gpio: GPIO[16]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (648) USB_HCDC: Waitting 4G_moudle AT response
I (653) USB_HCDC: usb driver install suceed
I (658) gpio: GPIO[12]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (667) gpio: GPIO[13]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (928) USB_HCDC: Resetting Port

I (988) USB_HCDC: Port speed = 1

I (988) USB_HCDC: Pipe Default Created
I (988) USB_HCDC: 1. Set Device Addr = 1
I (990) USB_HCDC: Set Device Addr Done
I (994) USB_HCDC: 2. Sending set_config = 1
I (999) USB_HCDC: Config is set
I (1003) USB_HCDC: 3. Sending set_line state itf= 0 dtr=1 rts=0
I (1009) USB_HCDC: Line state is set
I (1014) USB_HCDC: Creating bulk out pipe

I (1019) USB_HCDC: Creating bulk in pipe

**I (4377) modem_board: Modem event! 0
I (4377) modem_board: Modem PPP Started
I (4386) esp-netif_lwip-ppp: Connected
I (4386) esp-netif_lwip-ppp: Name Server1: 183.230.126.225
I (4386) esp-netif_lwip-ppp: Name Server2: 0.0.0.0
I (4393) modem_board: IP event! 6
I (4397) modem_board: Modem Connected to PPP Server
I (4402) modem_board: ppp ip: 10.52.94.238, mask: 255.255.255.255, gw: 192.168.0.1
I (4410) modem_board: Main DNS: 183.230.126.225
I (4416) modem_board: Backup DNS: 0.0.0.0
I (4423) wifi:wifi driver task: 3ffee9a0, prio:23, stack:6656, core=0**

I (4427) system_api: Base MAC address is not set
I (4432) system_api: read default base MAC address from EFUSE
I (4440) wifi:wifi firmware version: 4c9ede1
I (4442) wifi:wifi certification version: v7.0
I (4446) wifi:config NVS flash: enabled
I (4450) wifi:config nano formating: disabled
I (4454) wifi:Init data frame dynamic rx buffer num: 64
I (4459) wifi:Init management frame dynamic rx buffer num: 64
I (4464) wifi:Init management short buffer num: 32
I (4469) wifi:Init dynamic tx buffer num: 64
I (4473) wifi:Init static rx buffer size: 1600
I (4477) wifi:Init static rx buffer num: 16
I (4481) wifi:Init dynamic rx buffer num: 64
I (4485) wifi_init: rx ba win: 32
I (4489) wifi_init: tcpip mbox: 64
I (4493) wifi_init: udp mbox: 64
I (4497) wifi_init: tcp mbox: 64
I (4501) wifi_init: tcp tx win: 65534
I (4505) wifi_init: tcp rx win: 65534
I (4510) wifi_init: tcp mss: 1440
I (4514) wifi_init: WiFi IRAM OP enabled
I (4518) wifi_init: WiFi RX IRAM OP enabled
I (4523) wifi_init: LWIP IRAM OP enabled
I (4528) phy_init: phy_version 1800,e7ef680,Apr 13 2021,11:45:08
W (4534) phy_init: failed to load RF calibration data (0x1102), falling back to full calibration
I (4736) wifi:mode : softAP (7c:df:a1:39:4e:83)
I (4736) wifi:Total power save buffer number: 32
I (4737) wifi:Init max length of beacon: 752/752
I (4738) wifi:Init max length of beacon: 752/752
I (5270) wifi:Total power save buffer number: 32
I (5272) modem_wifi: softap ssid: esp_4g_router_3 password: 12345678
I (5272) modem_wifi: NAT is enabled
I (5276) main: Task dump

I (5278) main: Load     Stack left      Name    PRI

I (5283) main: 0.33     1984    main    1

I (5287) main: 10.94    1256    IDLE    0

I (5291) main: 0.02     1512    bulk-out        6

I (5296) main: 0.03     2688    tiT     18

I (5300) main: 0.00     1744    Tmr Svc         1

I (5305) main: 0.13     1356    usb_event       4

I (5309) main: 0.07     1520    dflt    8

I (5314) main: 0.01     1520    bulk-in         5

I (5318) main: 86.12    3540    esp_timer       22

I (5323) main: 0.08     1100    sys_evt         20

I (5328) main: 2.26     4628    wifi    23

I (5332) main: 0.00     1544    port    9

I (5336) main: Free heap=22292 bigst=8192, internal=22172 bigst=8192
I (5343) main: ..............

I (5347) USB_HCDC: USBH CDC Transfer Buffer Dump:
I (5352) USB_HCDC: usb transfer Buffer size, out = 1536, in = 1536
I (5359) USB_HCDC: usb transfer Max packet size, out = 46, in = 46

I (5366) USB_HCDC: USBH CDC Ringbuffer Dump:
I (5371) USB_HCDC: usb ringbuffer size, out = 20480, in = 25600
I (5378) USB_HCDC: usb ringbuffer High water mark, out = 46, in = 46

leeebo, 2021-08-03:
> **3. Did you get any error message in your browser?**
> NAME_NOT_RESOLVED, -105; DNS_PROBE_STARTED DNS_PROBE_FINISHED_NO_INTERNET

It should be the DNS problem, the DNS address got from the DHCP server seems inaccessible. Temporary solution, you can fix your device's DNS address to `114.114.114.114` or `8.8.8.8`

lightsalt2011, 2021-08-03:
do you means change esp_netif_get_dns_info func ? I see function modem_wifi_set_dhcps there is no caller

```
esp_err_t esp_netif_get_dns_info(esp_netif_t *esp_netif, esp_netif_dns_type_t type, esp_netif_dns_info_t *dns)
{
    #define EXAMPLE_STATIC_DNS_ADDR "114.114.114.114"
    #define EXAMPLE_STATIC_DNS_ADDR_2 "8.8.8.8"

    if (_IS_NETIF_ANY_POINT2POINT_TYPE(esp_netif)) {
        const ip_addr_t *dns_ip = dns_getserver(type);
        if (dns_ip == IP_ADDR_ANY) {
            return ESP_ERR_ESP_NETIF_DNS_NOT_CONFIGURED;
        }
        ESP_LOGE(TAG, "********* lvfeng Main DNS: " IPSTR, IP2STR(&dns_ip->u_addr.ip4));
        ip4_addr_t tmp_addr;
        tmp_addr.addr = ipaddr_addr(EXAMPLE_STATIC_DNS_ADDR_2); //for temp setting

#if CONFIG_LWIP_IPV6
        // memcpy(&dns->ip.u_addr.ip4, &dns_ip->u_addr.ip4, sizeof(ip4_addr_t));
        memcpy(&dns->ip.u_addr.ip4, &tmp_addr, sizeof(ip4_addr_t));
#else
        // memcpy(&dns->ip.u_addr.ip4, &dns_ip->addr, sizeof(ip4_addr_t));
        memcpy(&dns->ip.u_addr.ip4, &tmp_addr, sizeof(ip4_addr_t));

#endif
        return ESP_OK;
    }

    esp_netif_dns_param_t dns_param = {
        .dns_type = type,
        .dns_info = dns
    };
    return esp_netif_lwip_ipc_call(esp_netif_get_dns_info_api, esp_netif, (void *)&dns_param);
}
```
```
esp_err_t modem_wifi_set_dhcps(esp_netif_t *netif, uint32_t addr)
{
    esp_netif_dns_info_t dns;
    dns.ip.u_addr.ip4.addr = addr;
    dns.ip.type = IPADDR_TYPE_V4;
    dhcps_offer_t dhcps_dns_value = OFFER_DNS;
    esp_netif_dhcps_option(netif, ESP_NETIF_OP_SET, ESP_NETIF_DOMAIN_NAME_SERVER, &dhcps_dns_value, sizeof(dhcps_dns_value));
    ESP_ERROR_CHECK(esp_netif_set_dns_info(netif, ESP_NETIF_DNS_MAIN, &dns));
    return ESP_OK;
}
```

log as below 
```

I (988) USB_HCDC: Pipe Default Created
I (988) USB_HCDC: 1. Set Device Addr = 1
I (990) USB_HCDC: Set Device Addr Done
I (994) USB_HCDC: 2. Sending set_config = 1
I (999) USB_HCDC: Config is set
I (1003) USB_HCDC: 3. Sending set_line state itf= 0 dtr=1 rts=0
I (1009) USB_HCDC: Line state is set
I (1014) USB_HCDC: Creating bulk out pipe

I (1019) USB_HCDC: Creating bulk in pipe

I (4393) modem_board: Modem event! 0
I (4393) modem_board: Modem PPP Started
I (4402) esp-netif_lwip-ppp: Connected
I (4402) esp-netif_lwip-ppp: Name Server1: 183.230.126.225
I (4402) esp-netif_lwip-ppp: Name Server2: 0.0.0.0
I (4409) modem_board: IP event! 6
I (4413) modem_board: Modem Connected to PPP Server
I (4418) modem_board: ppp ip: 10.188.155.44, mask: 255.255.255.255, gw: 192.168.0.1
E (4426) esp_netif_lwip: ********* lvfeng Main DNS: 183.230.126.225
I (4433) modem_board: Main DNS: 8.8.8.8
E (4438) esp_netif_lwip: ********* lvfeng Main DNS: 0.0.0.0
I (4444) modem_board: Backup DNS: 8.8.8.8
I (4451) wifi:wifi driver task: 3ffee9a0, prio:23, stack:6656, core=0
I (4455) system_api: Base MAC address is not set
I (4460) system_api: read default base MAC address from EFUSE
I (4468) wifi:wifi firmware version: 4c9ede1
I (4471) wifi:wifi certification version: v7.0
I (4475) wifi:config NVS flash: enabled
I (4478) wifi:config nano formating: disabled
I (4483) wifi:Init data frame dynamic rx buffer num: 64
I (4488) wifi:Init management frame dynamic rx buffer num: 64
I (4493) wifi:Init management short buffer num: 32
I (4497) wifi:Init dynamic tx buffer num: 64
I (4502) wifi:Init static rx buffer size: 1600
I (4506) wifi:Init static rx buffer num: 16
I (4510) wifi:Init dynamic rx buffer num: 64
I (4514) wifi_init: rx ba win: 32
I (4518) wifi_init: tcpip mbox: 64
I (4522) wifi_init: udp mbox: 64
I (4526) wifi_init: tcp mbox: 64
I (4529) wifi_init: tcp tx win: 65534
I (4534) wifi_init: tcp rx win: 65534
I (4538) wifi_init: tcp mss: 1440
I (4542) wifi_init: WiFi IRAM OP enabled
I (4547) wifi_init: WiFi RX IRAM OP enabled
I (4552) wifi_init: LWIP IRAM OP enabled
I (4557) phy_init: phy_version 1800,e7ef680,Apr 13 2021,11:45:08
W (4563) phy_init: failed to load RF calibration data (0x1102), falling back to full calibration
I (4738) wifi:mode : softAP (7c:df:a1:39:4e:83)
I (4739) wifi:Total power save buffer number: 32
I (4739) wifi:Init max length of beacon: 752/752
I (4741) wifi:Init max length of beacon: 752/752
I (5270) wifi:Total power save buffer number: 32
I (5272) modem_wifi: softap ssid: esp_4g_router_3 password: 12345678
I (5272) modem_wifi: NAT is enabled
I (5276) main: Task dump
```

still cannot access Internet

lightsalt2011, 2021-08-03:
without hardcode for DNS, sometimes esp32s2 can access Internet normally, but it is not stable. 

- not sure for access Internet 100% when device power on 
- when I connect AP to browser the video like bilibili, sometimes error comes out, like below
```
 (4551) wifi_init: LWIP IRAM OP enabled
I (4556) phy_init: phy_version 1800,e7ef680,Apr 13 2021,11:45:08
W (4562) phy_init: failed to load RF calibration data (0x1102), falling back to full calibration
I (4737) wifi:mode : softAP (7c:df:a1:39:4e:83)
I (4738) wifi:Total power save buffer number: 32
I (4738) wifi:Init max length of beacon: 752/752
I (4740) wifi:Init max length of beacon: 752/752
I (5247) wifi:Total power save buffer number: 32
I (5249) modem_wifi: softap ssid: esp_4g_router_4 password: 12345678
I (5249) modem_wifi: NAT is enabled
I (20808) wifi:new:<13,2>, old:<13,2>, ap:<13,2>, sta:<255,255>, prof:13
I (20808) wifi:station: a4:9b:4f:a5:06:0a join, AID=1, bgn, 40D
I (20840) modem_wifi: station a4:9b:4f:a5:06:0a join, AID=1
I (20983) esp_netif_lwip: DHCP server assigned IP to a station, IP is: 192.168.4.2
I (20983) modem_board: IP event! 2
W (21455) wifi:<ba-add>idx:4 (ifx:1, a4:9b:4f:a5:06:0a), tid:0, ssn:13, winSize:64
E (35978) esp-netif_lwip-ppp: pppos_input_tcpip failed with -1
E (35982) esp-netif_lwip-ppp: pppos_input_tcpip failed with -1
E (35993) esp-netif_lwip-ppp: pppos_input_tcpip failed with -1
E (35996) esp-netif_lwip-ppp: pppos_input_tcpip failed with -1
E (36002) esp-netif_lwip-ppp: pppos_input_tcpip failed with -1
E (36005) esp-netif_lwip-ppp: pppos_input_tcpip failed with -1
E (36009) esp-netif_lwip-ppp: pppos_input_tcpip failed with -1
E (36015) esp-netif_lwip-ppp: pppos_input_tcpip failed with -1
E (36022) esp-netif_lwip-ppp: pppos_input_tcpip failed with -1
E (36028) esp-netif_lwip-ppp: pppos_input_tcpip failed with -1
E (36035) esp-netif_lwip-ppp: pppos_input_tcpip failed with -1
E (36041) esp-netif_lwip-ppp: pppos_input_tcpip failed with -1
E (36048) esp-netif_lwip-ppp: pppos_input_tcpip failed with -1
E (36054) esp-netif_lwip-ppp: pppos_input_tcpip failed with -1
E (36061) esp-netif_lwip-ppp: pppos_input_tcpip failed with -1
E (36067) esp-netif_lwip-ppp: pppos_input_tcpip failed with -1
E (36074) esp-netif_lwip-ppp: pppos_input_tcpip failed with -1
E (36080) esp-netif_lwip-ppp: pppos_input_tcpip failed with -1
E (36087) esp-netif_lwip-ppp: pppos_input_tcpip failed with -1
E (36093) esp-netif_lwip-ppp: pppos_input_tcpip failed with -1
E (36100) esp-netif_lwip-ppp: pppos_input_tcpip failed with -1
E (36106) esp-netif_lwip-ppp: pppos_input_tcpip failed with -1

```

lightsalt2011, 2021-08-03:
another question
1. how can I make DATA bandwidth(upload and download) increased 
2. how can I test the DATA bandwidth performance

leeebo, 2021-08-03:
1. Please update to the latest commit, optimization suggestions added in [readme](https://github.com/espressif/esp-iot-solution/tree/usb/add_usb_solutions/examples/usb/usb_cdc_4g_module#usb-cdc-4g-module-%E7%A4%BA%E4%BE%8B%E7%A8%8B%E5%BA%8F%E8%AF%B4%E6%98%8E)
2. You can also find how to solve the `esp-netif_lwip-ppp: pppos_input_tcpip failed with -1` problem in readme

lightsalt2011, 2021-08-03:
thanks for ur reply, BTW, how can I test the DATA bandwidth performance

lightsalt2011, 2021-08-03:
we test with Mac OS / Windows 10 / IOS 14.6 / Android OS
just Android OS can access Internet smoothly, 
Mac OS / Windows 10: cannot 
IOS 14.6: very low speed

is this issue about PPP protocol or ESP wifi functionality ? 

if we use this eps32 which connected with CAT.1 to develop wifi mesh router , is it OK ?

leeebo, 2021-08-04:
> Mac OS / Windows 10: cannot

@lightsalt2011 Hi, that should be the DNS problem, have fixed it, please try with the new commit 93bb98d. Thanks for your report.

leeebo, 2021-08-04:
> thanks for ur reply, BTW, how can I test the DATA bandwidth performance

You can use any `internet speed measuring tool` for testing, please try different tools to avoid tool problems.

lightsalt2011, 2021-08-04:
> > > Mac OS / Windows 10: cannot
> > 
> > 
> > @lightsalt2011 Hi, that should be the DNS problem, have fixed it, please try with the new commit. Thanks for your report.
> 
> @lightsalt2011 DNS server address can be manually config now, and fix some bug about DNS. We have test IOS, Android Windows and Linux, they all works as normal, you can try with the latest commit.
> 
> > if we use this eps32 which connected with CAT.1 to develop wifi mesh router , is it OK ?
> 
> @lightsalt2011 Yes you can, esp32 should work with no difference to other wifi devices.

@leeebo  Great works!  we will have a test

lightsalt2011, 2021-08-04:
@leeebo  how about the power consumption of CAT.1  ? if CAT.1 4G module always connect with ESP32S2 by usb. 
if no packet communication , CAT.1 will suspend or not ? does  the protocol of USB CDC PPP can guarantee the suspend or resume ?

leeebo, 2021-08-04:
> @leeebo how about the power consumption of CAT.1 ? if CAT.1 4G module always connect with ESP32S2 by usb.
> if no packet communication , CAT.1 will suspend or not ? does the protocol of USB CDC PPP can guarantee the suspend or resume ?

Please open a new topic, separating different issues and make them clear can help more developers

leeebo, 2021-08-18:
@lightsalt2011 Hi, we have tested the EC600 module, which works as normal now, you can pull the latest and try.

You can choose the module in menuconfig now Menuconfig → Component config → ESP-MODEM → Choose Modem Board, no need to manully config the endpoint.

[image]

commit dd9eb1ef (2021-08-17): 4g_module: support air724ug ec600n

fix #111, fix #116
closed 2023-07-10 by leeebo