espressif/esp-iot-solution #112: any stable or released branch of IDF can be used for USB CDC 4G Module?

achao1104, 2021-07-29:
Hello,

Instead of version of below:
git pull
git checkout 166c30e7b2ed1d
git submodule update --init --recursive

Can I use any stable version of IDF?

Thanks.

leeebo, 2021-07-29:
USB Host will be supported in future IDF 4.4 (released in November maybe), most functions are under developing (USB Host stack, device classes, demo applications, etc.) until now we have no release branch, but some demo code will be open for user test.

leeebo, 2023-07-10:
The stable version of `iot_usbh_modem` has been merged in [master](https://github.com/espressif/esp-iot-solution/tree/master/components/usb/iot_usbh_modem) and has been released as a [component in esp-registry](https://components.espressif.com/components/espressif/iot_usbh_modem), please give a try

closed 2023-07-10 by leeebo