espressif/esp-iot-solution #113: SPI bus_lock issue using display drivers

jonshouse1, 2021-07-30:
I've got  my VNC client running fast now.

Only issue is that I randomly and infrequently get this

````
E (8143603) bus_lock: spi_bus_lock_acquire_end(694): Cannot release a lock that hasn't been acquired.
````
Seems to be during calls to draw_bitmap()

Project is here:
https://github.com/jonshouse1/ESPVNCC

I am using two tasks, one writing to the display one reading the touch pad position, I assume the esp-iot-solution display and touch drivers are thread safe ?

leeebo, 2021-07-31:
@jonshouse1 It is recommended to assign a dedicated SPI peripheral for LCD display, because of the high throughput and uninterruptible,

jonshouse1, 2021-07-31:
> It is recommended to assign a dedicated SPI peripheral for LCD display, because of the high throughput and uninterruptible,

1) I Can't without re-laying out the board, it is not my board

2)  Are you telling me I wasted my time writing this as someone choose to ship a driver they know to be unstable without possibility of a fix ?

3) What does "because of the high throughput and uninterruptible" actually mean ?   The driver is using DMA, the DMA transaction would generate interrupts but as far as I know it does not disable them, it would also leave the machine free during the output itself, hence would be "interruptible", at least as far as I know?

alic-maker, 2021-08-02:
@jonshouse1 It is possible for the display screen and the touch panel to share an SPI peripheral, which is exactly what our example does. And there seems to be nothing wrong with your code. I'm trying to reproduce the problem with board ESP32-Ethernet-Kit, but it is not identical to your hardware, so my test is in some trouble.

jonshouse1, 2021-11-17:
I've updated my code in an effort to try and work around this, but I am still getting issues with the iot display driver.  

I get infrequent crashes like this one, core may be 0 or 1 but the failure message is always consistent.

```
assertion "handle == get_acquiring_dev(host)" failed: file "IDF/components/driver/spi_master.c", line 949, function: spi_device_polling_end

abort() was called at PC 0x40100ba3 on core 0
```

steveglennon, 2022-10-21:
There is a fix for this issue at commit https://github.com/espressif/esp-idf/commit/3c8f6dd1cfebaee3860fbc66ff985528ba56b447

I was having the same problem and confirmed that this fixed it.

jonshouse1, 2022-11-02:
> There is a fix for this issue at commit [espressif/esp-idf@3c8f6dd](https://github.com/espressif/esp-idf/commit/3c8f6dd1cfebaee3860fbc66ff985528ba56b447)
> 
> I was having the same problem and confirmed that this fixed it.

Great news, thanks for letting me know.

closed 2023-07-10 by leeebo