espressif/esp-iot-solution #114: fatal error: screen_driver.h: No such file or directory

modi12jin, 2021-07-31:
# How to fix this error

[image]

```
Processing esp32-s2-saola-1 (platform: espressif32; board: esp32-s2-saola-1; framework: espidf)
----------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32-s2-saola-1.html
PLATFORM: Espressif 32 (3.3.1) > Espressif ESP32-S2-Saola-1
HARDWARE: ESP32S2 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-espidf 3.40300.0 (4.3.0) 
 - tool-cmake 3.16.4 
 - tool-esptoolpy 1.30100.210531 (3.1.0) 
 - tool-ninja 1.7.1 
 - toolchain-esp32s2ulp 1.22851.191205 (2.28.51) 
 - toolchain-riscv-esp 1.80400.0 (8.4.0) 
 - toolchain-xtensa32s2 1.80400.210211 (8.4.0)
Reading CMake configuration...
Generating assembly for certificate bundle...
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 0 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Compiling .pio/build/esp32-s2-saola-1/src/main.o
Generating LD script .pio/build/esp32-s2-saola-1/esp32s2_out.ld
Generating partitions .pio/build/esp32-s2-saola-1/partitions.bin
Compiling .pio/build/esp32-s2-saola-1/app_trace/app_trace.o
Compiling .pio/build/esp32-s2-saola-1/app_trace/app_trace_util.o
src/main.c:20:10: fatal error: screen_driver.h: No such file or directory

***********************************************************************
* Looking for screen_driver.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:screen_driver.h"
* Web  > https://platformio.org/lib/search?query=header:screen_driver.h
*
***********************************************************************

 #include "screen_driver.h"
          ^~~~~~~~~~~~~~~~~
compilation terminated.
Compiling .pio/build/esp32-s2-saola-1/app_trace/host_file_io.o
*** [.pio/build/esp32-s2-saola-1/src/main.o] Error 1
```

leeebo, 2021-07-31:
@modi12jin I am not familiar with platformio environment, but I think it should be a simple dependency problem. Please keep in mind folders like `display` is not a component but a component's dir, which contents three components `digital_tube` `screen` and `touch_panel`. So, if you want to add `screen` in your project's component dir, please move it to the project's component dir, like:

```
project dir
  -components
    -bus
    -screen
    -others
```

modi12jin, 2021-07-31:
[image]

thank you very much

closed 2021-07-31 by modi12jin