espressif/esp-iot-solution #115: LCD Portrait/Landscape

jonshouse1, 2021-07-31:
This is giving me a headache.   How do I switch between portrait and landscape with these drivers.   Does the calibration only work in portrait ?

Here is my init function, I want to pass either (320, 240)   or (240,320) and have the display come up in landscape or portrait. Can anyone advise what I am doing wrong.

I can get the calibration screen in landscape but the bottom left circle seems to be outside the display area, touching the screen  gives this.
```
E (373311) lcd ili9341: .... /ili9341/ili9341.c:199 (lcd_ili9341_set_window):The set coordinates exceed the screen size
```

```
// Inil lcd display and touch screen
void lcd_init(int w, int h)
{
        int p                           = TRUE;                         // default to portrait
        spi_bus_handle_t                bus_handle;
        scr_interface_driver_t          *iface_drv;

        if (w >h)                                                       // screen wider than tall ?
                p=FALSE;                                                // then landscape

        led_pwm_init();
        led_pwm_set(255);                                               // turn backlight on
        spi_config_t            bus_conf =
        {
                .miso_io_num    = GPIO_MISO,
                .mosi_io_num    = GPIO_MOSI,
                .sclk_io_num    = GPIO_SCLK,
                //.max_transfer_sz= 32,
        };
        //bus_handle = spi_bus_create(SPI3_HOST, &bus_conf);            // or SPI2_HOST ?
        bus_handle = spi_bus_create(SPI2_HOST, &bus_conf);              // or SPI2_HOST ?

        scr_interface_spi_config_t spi_lcd_cfg =
        {
                .spi_bus    = bus_handle,
                .pin_num_cs = GPIO_LCDCS,
                .pin_num_dc = GPIO_DATACMD,
                .clk_freq   = SPI_SPEED_LCD_HZ,
                .swap_data  = true,
        };
        scr_interface_create(SCREEN_IFACE_SPI, &spi_lcd_cfg, &iface_drv);

        // possible rotations SCR_DIR_TBLR  SCR_DIR_LRBT   SCR_DIR_BTRL
        scr_controller_config_t lcd_cfg =
        {
                .interface_drv          = iface_drv,
                .pin_num_rst            = GPIO_LCDRESET,
                .pin_num_bckl           = -1,
                .rst_active_level       = 0,
                .bckl_active_level      = 1,
                .offset_hor             = 0,
                .offset_ver             = 0,
                .width                  = w,
                .height                 = h,
                .rotate                 = SCR_DIR_LRBT,
        };

        if (p != TRUE)                                                  // landscape ?
        {
                //lcd_cfg.width  = h;
                //lcd_cfg.height = w;
                lcd_cfg.rotate = SCR_DIR_TBLR;
        }

        scr_find_driver(SCREEN_CONTROLLER_ILI9341, &lcd_drv);
        ESP_LOGI(TAG, "JA doing lcd_drv init");
        lcd_drv.init(&lcd_cfg);

        // Same SPI MISO MOSI CLK pins as LCD
        touch_panel_config_t touch_cfg =
        {
                .interface_spi =
                {
                        //.spi_bus = spi2_bus,
                        .spi_bus = bus_handle,
                        .pin_num_cs = GPIO_TCS,
                        .clk_freq = SPI_SPEED_TOUCH_HZ,
                },
                .interface_type = TOUCH_PANEL_IFACE_SPI,
                .pin_num_int    = GPIO_TIRQ,
                .width          = w,
                .height         = h,
                .direction      = TOUCH_DIR_LRBT,
        };
        if (p != TRUE)
        {
                //touch_cfg.width  = h;
                //touch_cfg.height = w;
                touch_cfg.direction = SCR_DIR_TBLR;
        }

        touch_panel_find_driver(TOUCH_PANEL_CONTROLLER_XPT2046, &touch_drv);
        ESP_LOGI(TAG, "doing touch_drv init");

        touch_drv.init(&touch_cfg);
        touch_drv.calibration_run(&lcd_drv, false);             // true=force, false read from flash if possible
        gpio_intr_disable(36);                                  // common issue

        ESP_LOGI(TAG, "[APP] IDF version: %s", esp_get_idf_version());
        ESP_LOGI(TAG, "[APP] Free memory: %d bytes", esp_get_free_heap_size());
}

alic-maker, 2021-08-02:
The parameter width and height of scr_controller_config_t is only dependent on hardware. The correct value is the available area size when the rotation is set to SCR_DIR_LRTB(original direction of the screen). If the screen resolution you use is 240x320, you only need to fix it as width = 240 and height = 320. Resistive touch panel are generally set to the same width and height.

If you want to switch landscape to portrait, you can call the `lcd_drv.set_direction()` with proper parameter to rotate the display. 
After rotation, the display size of the screen may change. You can know the length and width of the displayable area of the current screen through `lcd_drv.get_info()`.

jonshouse1, 2021-08-02:
Thanks for clarification.

Just to get the order clear in my head, I always initialise portrait as that is the physical display layout, then rotate to landscape, at which point I may have less pixels than I expected?

``` 
  lcd_cfg width=240 height=320
  lcd_drv.init(&lcd_cfg);
  touch_cfg width=240 height=320
  touch_drv.init()
  touch_drv.calibration_run()
  lcd_drv.set_direction()
  touch_drv.set_direction() 
  lcd_drv.get_info()
```

Will the calibration run rotated ?
```
  lcd_cfg width=240 height=320
  lcd_drv.init(&lcd_cfg);
  touch_cfg width=240 height=320
  touch_drv.init()
  lcd_drv.set_direction()
  touch_drv.set_direction() 
  touch_drv.calibration_run()
  lcd_drv.get_info()
```

If  the display always needs to be initialised to one physical orientation then the documentations is unhelpful, or simply wrong.  It bangs on at length about orientation then goes into an example init but fails to mention that the display can only be initialised effectively in its hardware raster orientation, it does not even mention set_direction() other than in the API documentation.

alic-maker, 2021-08-02:
If you initialize portrait then rotate to landscape, just the display area from 240x320 change to 320x240. The total number of pixels does not change. If you rotate the resistive touch panel, the touch coordinates may become inaccurate unless you recalibrate.

jonshouse1, 2021-08-02:
>just the display area from 240x320 change to 320x240
Yes, I thought that, but the documentation was saying the offsets can be different in portrait vs landscape, I guess it was late night reading as a display rotated should have exactly the same number of pixels as a display not rotated, guess I need more sleep...

alic-maker, 2021-08-03:
If you use offset, that is, the resolution of the screen module you use is less than that of the screen control chip. At this time, the situation becomes a little more complicated, because different rotation directions will change the offset from the coordinate origin, but the total size of the display area is still the same. Comparing these two [pictures](https://docs.espressif.com/projects/espressif-esp-iot-solution/en/latest/display/screen.html#offset-of-the-display-area) may help you understand. BTW,You don't need to consider this situation here.

jonshouse1, 2021-08-03:
Thanks for the clarification.
I am not sure if this is the "correct" way to handle it, but it seems to work ok for the situations I am likely to use.

```
scr_driver_t                            lcd_drv;
touch_panel_driver_t                    touch_drv;

void app_main(void)
{
        bzero(&lcd_drv, sizeof(scr_driver_t));
        lcd_init(240, 320);                                                                     // initialise the SPI LCD driver

        // Change LCD rotation
        // possible rotations   portrait:   SCR_DIR_LRTB,  SCR_DIR_LRBT,  SCR_DIR_RLTB,  SCR_DIR_RLBT
        //                      landscape:  SCR_DIR_TBLR,  SCR_DIR_BTLR,  SCR_DIR_TBRL,  SCR_DIR_BTRL
        lcd_ts_rotate(SCR_DIR_TBLR);                                                            // uncomment for default potrait LRBT
```

```
// Change LCD and touch screen rotation
// 'r' is screen direction, the touch panel will be set to match
void lcd_ts_rotate(scr_dir_t r)
{
        switch (r)
        {
                case SCR_DIR_LRTB:      touch_drv.set_direction(TOUCH_DIR_LRTB);        break;
                case SCR_DIR_LRBT:      touch_drv.set_direction(TOUCH_DIR_LRBT);        break;
                case SCR_DIR_RLTB:      touch_drv.set_direction(TOUCH_DIR_RLTB);        break;
                case SCR_DIR_RLBT:      touch_drv.set_direction(TOUCH_DIR_RLBT);        break;
                case SCR_DIR_TBLR:      touch_drv.set_direction(TOUCH_DIR_TBLR);        break;
                case SCR_DIR_BTLR:      touch_drv.set_direction(TOUCH_DIR_BTLR);        break;
                case SCR_DIR_TBRL:      touch_drv.set_direction(TOUCH_DIR_TBRL);        break;
                case SCR_DIR_BTRL:      touch_drv.set_direction(TOUCH_DIR_BTRL);        break;
                default:        break;
        }
        lcd_drv.set_direction(r);
}
```

closed 2021-08-06 by jonshouse1