espressif/esp-iot-solution #116: about USB CDC 4G Module 示例程序 :  HCD_PIPE_EVENT_ERROR_STALL

qq0901001, 2021-08-04:
I (24) boot: ESP-IDF v4.4-dev-2184-g166c30e7b-dirty 2nd stage bootloader
I (24) boot: compile time 16:02:41
I (24) boot: chip revision: 0
I (28) qio_mode: Enabling default flash chip QIO
I (33) boot.esp32s2: SPI Speed      : 80MHz
I (38) boot.esp32s2: SPI Mode       : QIO
I (43) boot.esp32s2: SPI Flash Size : 4MB
I (48) boot: Enabling RNG early entropy source...
I (53) boot: Partition Table:
I (57) boot: ## Label            Usage          Type ST Offset   Length
I (64) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (71) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (79) boot:  2 factory          factory app      00 00 00010000 00100000
I (86) boot: End of partition table
I (91) esp_image: segment 0: paddr=00010020 vaddr=3f000020 size=1912ch (102700) map
I (117) esp_image: segment 1: paddr=00029154 vaddr=3ffcbc30 size=031ech ( 12780) load
I (119) esp_image: segment 2: paddr=0002c348 vaddr=40024000 size=03cd0h ( 15568) load
I (126) esp_image: segment 3: paddr=00030020 vaddr=40080020 size=79844h (497732) map
I (216) esp_image: segment 4: paddr=000a986c vaddr=40027cd0 size=13f5ch ( 81756) load
I (235) esp_image: segment 5: paddr=000bd7d0 vaddr=50000000 size=00010h (    16) load
I (246) boot: Loaded app from partition at offset 0x10000
I (246) boot: Disabling RNG early entropy source...
I (258) cache: Instruction cache 	: size 16KB, 4Ways, cache line size 16Byte
I (258) cpu_start: Pro cpu up.
I (260) cache: flash_count=1, size=16, spiram_count=0, size=0,together=0
I (268) cache: Flash wrap enabled, size = 16.
I (284) cpu_start: Pro cpu start user code
I (284) cpu_start: cpu freq: 240000000
I (284) cpu_start: Application information:
I (289) cpu_start: Project name:     cdc_4g
I (294) cpu_start: App version:      e7f7cda-dirty
I (299) cpu_start: Compile time:     Aug  3 2021 19:11:48
I (305) cpu_start: ELF file SHA256:  b0b44e1ee0f79092...
I (311) cpu_start: ESP-IDF:          v4.4-dev-2184-g166c30e7b-dirty
I (318) heap_init: Initializing. RAM available for dynamic allocation:
I (326) heap_init: At 3FF9E000 len 00002000 (8 KiB): RTCRAM
I (332) heap_init: At 3FFD35E8 len 00028A18 (162 KiB): DRAM
I (338) heap_init: At 3FFFC000 len 00003A10 (14 KiB): DRAM
I (344) spi_flash: detected chip: generic
I (349) spi_flash: flash io: qio
I (356) cpu_start: Starting scheduler on PRO CPU.
I (614) main: ====================================
I (614) main:      ESP 4G Cat.1 Wi-Fi Router
I (614) main: ====================================
I (619) gpio: GPIO[15]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (628) gpio: GPIO[17]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (637) gpio: GPIO[16]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (647) USB_HCDC: Waitting 4G_moudle AT response
I (652) USB_HCDC: usb driver install suceed
I (657) gpio: GPIO[12]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (666) gpio: GPIO[13]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (927) USB_HCDC: Resetting Port

I (987) USB_HCDC: Port speed = 1

I (987) USB_HCDC: Pipe Default Created
I (987) USB_HCDC: 1. Set Device Addr = 1
I (989) USB_HCDC: Set Device Addr Done
I (993) USB_HCDC: 2. Sending set_config = 1
I (998) USB_HCDC: Config is set
I (1002) USB_HCDC: 3. Sending set_line state itf= 0 dtr=1 rts=0
W (1008) USB_HCDC: line 400 Pipe: default HCD_PIPE_EVENT_ERROR_STALL
E (1176) esp-modem-dte: esp_modem_dte_send_cmd(192): process command timeout
 response timeoutm_dce: esp_modem_dce_generic_command(31): Command:AT
I (1181) modem_board: Current timeouts: 1 and errors: 0
E (6187) esp-modem-dte: esp_modem_dte_send_cmd(192): process command timeout
W (6187) esp_modem_dce: esp_modem_dce_generic_command(31): Command:+++ response timeout
E (6692) esp-modem-dte: esp_modem_dte_send_cmd(192): process command timeout
 response timeoutm_dce: esp_modem_dce_generic_command(31): Command:AT
E (7197) esp-modem-dte: esp_modem_dte_send_cmd(192): process command timeout
[... 216 lines cut ...]
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              7 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x86  EP 6 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x05  EP 5 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0

=====================================================================

qq0901001, 2021-08-04:
**my usb_desc_ep_t  config as below**

static usb_desc_ep_t bulk_in_ep_desc = {
    .bLength = sizeof(usb_desc_ep_t),
    .bDescriptorType = 0x05,
    .bEndpointAddress = 0x82,       //EP 2 IN
    .bmAttributes = 0x02,
    .wMaxPacketSize = 0x0200,           //MPS of 64 bytes
    .bInterval = 0x00,
};

static usb_desc_ep_t bulk_out_ep_desc = {
    .bLength = sizeof(usb_desc_ep_t),
    .bDescriptorType = 0x05,
    .bEndpointAddress = 0x01,       //EP 1 OUT
    .bmAttributes = 0x02,
    .wMaxPacketSize = 0x0200,           //MPS of 64 bytes
    .bInterval = 0,
};

qq0901001, 2021-08-04:
https://doc.openluat.com/wiki/6?wiki_page_id=2145

**AT+SETUSB?

mode: 2
Vid: 0x1782
Pid: 0x4e00

OK**

qq0901001, 2021-08-04:
i have changed testing cat1 module to ECM mode

leeebo, 2021-08-04:
> i have changed testing cat1 module to ECM mode

1. ECM mode is not supported, the example using PPP mode.
2. Please check with 4g module vendor if that support USB1.1 Fullspeed mode, from the USB descriptor above the device seems can not be supported, you can refer related topic here https://github.com/espressif/esp-iot-solution/issues/111#issuecomment-888862377

qq0901001, 2021-08-04:
@leeebo thanks for your reply!

i tested cat1 modules, both of them are 480M and CDC Data interface only has 512-bytes endpoint

waiting for ml302 development kit , :)

qq0901001, 2021-08-05:
run [USB CDC 4G Module demo ](https://github.com/espressif/esp-iot-solution/tree/usb/add_usb_solutions/examples/usb/usb_cdc_4g_module) with ml302 devKit , it works, but through 'lsusb' to get info from ml302, it seems, nothing was different with other cat.1 modules (tested: BG96, AIR724UG)

> 1. CDC Data interface also only has 512-bytes endpoint
> 2. USB speed mode is high speed mode (480M)

```
➜  usb_cdc_4g_module git:(usb/add_usb_solutions) ✗ lsusb -t
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/10p, 480M
    |__ Port 7: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M
        |__ Port 1: Dev 12, If 0, Class=Vendor Specific Class, Driver=, 480M
        |__ Port 1: Dev 12, If 1, Class=Vendor Specific Class, Driver=, 480M
        |__ Port 1: Dev 12, If 2, Class=Vendor Specific Class, Driver=, 480M
        |__ Port 1: Dev 12, If 3, Class=Vendor Specific Class, Driver=, 480M
        |__ Port 1: Dev 12, If 4, Class=Vendor Specific Class, Driver=, 480M
        |__ Port 1: Dev 12, If 5, Class=Vendor Specific Class, Driver=, 480M
        |__ Port 1: Dev 12, If 6, Class=Vendor Specific Class, Driver=, 480M
        |__ Port 1: Dev 12, If 7, Class=Vendor Specific Class, Driver=, 480M
```

```
➜  usb_cdc_4g_module git:(usb/add_usb_solutions) ✗ lsusb -v
Bus 003 Device 012: ID 1782:4d10 Spreadtrum Communications Inc. 
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass          239 Miscellaneous Device
  bDeviceSubClass         2 
  bDeviceProtocol         1 Interface Association
  bMaxPacketSize0        64
  idVendor           0x1782 Spreadtrum Communications Inc.
  idProduct          0x4d10 
  bcdDevice            0.00
  iManufacturer           1 
  iProduct                2 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x00c1
    bNumInterfaces          8
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xc0
      Self Powered
    MaxPower              400mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              3 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
[... 195 lines cut ...]
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              3 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x88  EP 8 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x08  EP 8 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0

```

leeebo, 2021-08-05:
@qq0901001 ML302 works as normal in both high-speed and full-speed mode, different modes have different device descriptors (high-speed mode data endpoint MPS=512; full-speed mode MPS=64), but no matter in which mode it's CDC data endpoint address is the same (IN 0x81, OUT 0x01)

For AIR724UG you testing, the problem may:
1. Full-speed mode does not support
2. Full-speed mode supported, but the endpoint address is different from high-speed (You can add a USB Fullspeed isolator to make it works in full-speed mode, then output the descriptors in Fullspeed mode. )

qq0901001, 2021-08-05:
@leeebo got it ,thank you very much for you acvice!

closed 2021-08-05 by qq0901001

leeebo, 2021-08-18:
@qq0901001 Hi we have test with the Air724 module, which works as normal now, you can pull the latest and try.
You can choose the module in menuconfig now  `Menuconfig → Component config → ESP-MODEM → Choose Modem Board`, no need to manully config the endpoint.

[image]

commit dd9eb1ef (2021-08-17): 4g_module: support air724ug ec600n

fix #111, fix #116