espressif/esp-iot-solution #118: esp-netif_lwip-ppp: pppos_input_tcpip failed with -1

qq0901001, 2021-08-05:
## Problem Description
[usb_cdc_4g_module](https://github.com/espressif/esp-iot-solution/tree/usb/add_usb_solutions/examples/usb/usb_cdc_4g_module) Error log: ***pppos_input_tcpip failed with -1***

## Detailed log

***1. When the code starts to run, the initialization is correct, and the wifi hot spot can be established normally***
```
I (612) main: ====================================
I (612) main:      ESP 4G Cat.1 Wi-Fi Router
I (612) main: ====================================
I (617) gpio: GPIO[15]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (626) gpio: GPIO[17]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (635) gpio: GPIO[16]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (645) USB_HCDC: Waitting 4G_moudle AT response
I (650) USB_HCDC: usb driver install suceed
I (655) gpio: GPIO[12]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (664) gpio: GPIO[13]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (925) USB_HCDC: Resetting Port

I (985) USB_HCDC: Port speed = 1

I (985) USB_HCDC: Pipe Default Created
I (985) USB_HCDC: 1. Set Device Addr = 1
I (987) USB_HCDC: Set Device Addr Done
I (991) USB_HCDC: 2. Sending set_config = 1
I (996) USB_HCDC: Config is set
I (1000) USB_HCDC: 3. Sending set_line state itf= 0 dtr=1 rts=0
I (1006) USB_HCDC: Line state is set
I (1011) USB_HCDC: Creating bulk out pipe

I (1016) USB_HCDC: Creating bulk in pipe

I (4955) modem_board: Modem event! 0
I (4955) modem_board: Modem PPP Started
I (4964) esp-netif_lwip-ppp: Connected
I (4964) esp-netif_lwip-ppp: Name Server1: 211.136.17.107
I (4964) esp-netif_lwip-ppp: Name Server2: 0.0.0.0
I (4971) modem_board: IP event! 6
I (4975) modem_board: Modem Connected to PPP Server
I (4980) modem_board: ppp ip: 10.50.205.15, mask: 255.255.255.255, gw: 192.168.0.1
I (4988) modem_board: Main DNS: 211.136.17.107
I (4993) modem_board: Backup DNS: 0.0.0.0
I (5001) wifi:wifi driver task: 3ffeaa08, prio:23, stack:6656, core=0
I (5004) system_api: Base MAC address is not set
I (5010) system_api: read default base MAC address from EFUSE
I (5017) wifi:wifi firmware version: 4c9ede1
I (5020) wifi:wifi certification version: v7.0
I (5024) wifi:config NVS flash: enabled
I (5028) wifi:config nano formating: disabled
I (5032) wifi:Init data frame dynamic rx buffer num: 64
I (5037) wifi:Init management frame dynamic rx buffer num: 64
I (5042) wifi:Init management short buffer num: 32
I (5047) wifi:Init dynamic tx buffer num: 64
I (5051) wifi:Init static rx buffer size: 1600
I (5055) wifi:Init static rx buffer num: 16
I (5059) wifi:Init dynamic rx buffer num: 64
I (5063) wifi_init: rx ba win: 32
I (5067) wifi_init: tcpip mbox: 64
I (5071) wifi_init: udp mbox: 64
I (5075) wifi_init: tcp mbox: 64
I (5079) wifi_init: tcp tx win: 65534
I (5083) wifi_init: tcp rx win: 65534
I (5087) wifi_init: tcp mss: 1440
I (5091) wifi_init: WiFi IRAM OP enabled
I (5096) wifi_init: WiFi RX IRAM OP enabled
I (5101) wifi_init: LWIP IRAM OP enabled
I (5106) phy_init: phy_version 1800,e7ef680,Apr 13 2021,11:45:08
W (5112) phy_init: failed to load RF calibration data (0x1102), falling back to full calibration
I (5265) wifi:mode : softAP (7c:df:a1:34:b4:77)
I (5266) wifi:Total power save buffer number: 32
I (5266) wifi:Init max length of beacon: 752/752
I (5267) wifi:Init max length of beacon: 752/752
I (5775) wifi:Total power save buffer number: 32
***I (5777) modem_wifi: softap ssid: esp_4g_router password: 12345678***
I (5777) modem_wifi: NAT is enabled
```
***2. Open the browser or use bilibili to use the network, it will report the following error***
```
E (146235) esp-netif_lwip-ppp: pppos_input_tcpip failed with -1
E (146242) esp-netif_lwip-ppp: pppos_input_tcpip failed with -1
E (146249) esp-netif_lwip-ppp: pppos_input_tcpip failed with -1
E (146255) esp-netif_lwip-ppp: pppos_input_tcpip failed with -1
E (146262) esp-netif_lwip-ppp: pppos_input_tcpip failed with -1
E (146268) esp-netif_lwip-ppp: pppos_input_tcpip failed with -1
E (146275) esp-netif_lwip-ppp: pppos_input_tcpip failed with -1
E (146281) esp-netif_lwip-ppp: pppos_input_tcpip failed with -1
E (146288) esp-netif_lwip-ppp: pppos_input_tcpip failed with -1
E (146295) esp-netif_lwip-ppp: pppos_input_tcpip failed with -1
E (146301) esp-netif_lwip-ppp: pppos_input_tcpip failed with -1
E (146335) esp-netif_lwip-ppp: pppos_input_tcpip failed with -1
E (146364) esp-netif_lwip-ppp: pppos_input_tcpip failed with -1
E (146381) esp-netif_lwip-ppp: pppos_input_tcpip failed with -1
E (146430) esp-netif_lwip-ppp: pppos_input_tcpip failed with -1
E (146479) esp-netif_lwip-ppp: pppos_input_tcpip failed with -1
E (146525) esp-netif_lwip-ppp: pppos_input_tcpip failed with -1
E (146575) esp-netif_lwip-ppp: pppos_input_tcpip failed with -1
E (146592) esp-netif_lwip-ppp: pppos_input_tcpip failed with -1
E (146621) esp-netif_lwip-ppp: pppos_input_tcpip failed with -1
E (146670) esp-netif_lwip-ppp: pppos_input_tcpip failed with -1
```
***3. Although the operation will report an error, the transparent transmission of the 'wifi' network of 'esp32s2' is still normal***

leeebo, 2021-08-05:
It happens if the USB PPP interface try to input data to LWIP stack, but run out of RAM (network may works as normal, because the process will try until succeed). You can try to [enable PSRAM to alleviate the problem](https://github.com/espressif/esp-iot-solution/tree/usb/add_usb_solutions/examples/usb/usb_cdc_4g_module#%E4%BD%BF%E7%94%A8%E8%AF%B4%E6%98%8E).

AD-Codex, 2023-09-21:
Is adding external RAM chip for esp the only option?

leeebo, 2023-10-09:
@AD-Codex This is an equilibrium problem between consumers and producers. if data is generated faster than it is consumed, If it is a short period of large traffic impact, a larger storage area is needed to temporarily store the data. 

Slow down the producers is another solution (eg. slow down the USB request), but the logic is more complicated

closed 2024-12-23 by leeebo