espressif/esp-iot-solution #121: USB CDC 4G Module start failed with sim7600jc

achao1104, 2021-08-06:
Hello,

esp32-s2-wrover and sim7600jc, it reports the following failure. thanks.

I (40878) main: ====================================
I (40878) main:      ESP 4G Cat.1 Wi-Fi Router
I (40878) main: ====================================
I (40883) USB_HCDC: Waitting 4G_moudle AT response
I (40888) USB_HCDC: usb driver install suceed
I (41163) USB_HCDC: Resetting Port

I (41223) USB_HCDC: Port speed = 1

I (41223) USB_HCDC: Pipe Default Created
I (41223) USB_HCDC: 1. Set Device Addr = 1
I (41226) USB_HCDC: Set Device Addr Done
I (41230) USB_HCDC: 2. Sending set_config = 1
I (41236) USB_HCDC: Config is set
I (41239) USB_HCDC: 3. Sending set_line state itf= 0 dtr=1 rts=0
W (41246) USB_HCDC: line 400 Pipe: default HCD_PIPE_EVENT_ERROR_STALL
E (41394) esp-modem-dte: esp_modem_dte_send_cmd(192): process command timeout
 response timeoutem_dce: esp_modem_dce_generic_command(31): Command:AT
I (41399) modem_board: Current timeouts: 1 and errors: 0
E (46405) esp-modem-dte: esp_modem_dte_send_cmd(192): process command timeout
W (46405) esp_modem_dce: esp_modem_dce_generic_command(31): Command:+++ response timeout
E (46910) esp-modem-dte: esp_modem_dte_send_cmd(192): process command timeout
 response timeoutem_dce: esp_modem_dce_generic_command(31): Command:AT
E (47415) esp-modem-dte: esp_modem_dte_send_cmd(192): process command timeout
 response timeoutem_dce: esp_modem_dce_generic_command(31): Command:AT
I (47420) modem_board: Current timeouts: 2 and errors: 0
I (47426) modem_board: Restart to connect modem........
I (47432) modem_board: modem_board_reset!
ESP-ROM:esp32s2-rc4-20191025

leeebo, 2021-08-07:
@achao1104 Hi, seems esp32s2 has got the AT echo from your 4G module, but no response when try to change mode, then esp32s2 try to reset your 4G Module board from reset pin.. 

Please attach the 4G Module's RESET pin to esp32s2 reset control IO ([gpio13 by default](https://github.com/espressif/esp-iot-solution/blob/usb/add_usb_solutions/examples/usb/usb_cdc_4g_module/main/modem_board.c#L49)), then try again.

achao1104, 2021-08-07:
I have restarted for a couple of times, it still failed, do you mean it is not a usb cdc endpoint in and out address issue?

leeebo, 2021-08-07:
As the AT command can be responded to esp32s2, it could not be the bulk endpoint problem.

Sorry, I missed the warning message `W (41246) USB_HCDC: line 400 Pipe: default HCD_PIPE_EVENT_ERROR_STALL`,  esp32s2 encounter failure when try to set 4G module CDC line state, not sure if your module PPP mode initialization requires any other special steps. Please attach your module's user manual If you are convenient.

leeebo, 2021-08-07:
Same with #116

leeebo, 2021-08-07:
#116

achao1104, 2021-08-07:
Is it possible to solve? the 4g module is sim7600jc, user manual is here https://www.simcom.com/product/SIM7600X-H.html
By the way, from your codes, it looks you are integrating 7600 because I see a file name is 7600.

achao1104, 2021-08-09:
hello, can you help to confirm if sim7600 can be used?

achao1104, 2021-08-11:
mrq@ubuntu:~$ lsusb -v

Bus 001 Device 002: ID 1e0e:9001 Qualcomm / Option 
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1e0e Qualcomm / Option
  idProduct          0x9001 
  bcdDevice            3.18
  iManufacturer           1 
  iProduct                2 
  iSerial                 3 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength          258
    bNumInterfaces          6
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass    255 Vendor Specific Subclass
      bInterfaceProtocol    255 Vendor Specific Protocol
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           3
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
[... 634 lines cut ...]
    iConfiguration          0 
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower                0mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         9 Hub
      bInterfaceSubClass      0 Unused
      bInterfaceProtocol      0 Full speed (or root) hub
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0002  1x 2 bytes
        bInterval             255

achao1104, 2021-08-11:
hello, by lsusb -v, it looks I get a lot of endpoints, which one should I use?

leeebo, 2021-08-18:
> As the AT command can be responded to esp32s2, it could not be the bulk endpoint problem.
>
> Sorry, I missed the warning message `W (41246) USB_HCDC: line 400 Pipe: default [...]

@achao1104 We have fixed the issue `W (41246) USB_HCDC: line 400 Pipe: default HCD_PIPE_EVENT_ERROR_STALL`, which makes you initialize failed. You can try again with the new commits.

leeebo, 2021-08-18:
> hello, by lsusb -v, it looks I get a lot of endpoints, which one should I use?

You can ask the module's vendor for the specified endpoint address, or try the list one by one:

```
IN 0x81, OUT 0x01
IN 0x82, OUT 0x02
IN 0x84, OUT 0x03
IN 0x86, OUT 0x04
IN 0x88, OUT 0x05
IN 0x8a, OUT 0x06
```
Any good news, please let me know :smiley:

achao1104, 2021-09-01:
sorry, but where is the USB CDC 4G Module now? the link below is not acessable.
https://github.com/espressif/esp-iot-solution/tree/usb/add_usb_solutions/examples/usb/usb_cdc_4g_module

achao1104, 2021-09-01:
get it.

achao1104, 2021-12-03:
Hello,

I have tried all the list of endpoints as you said, the log is as below. Could you tell me if my modem can be used from the log?

IN 0x81, OUT 0x01
IN 0x82, OUT 0x02
IN 0x84, OUT 0x03
IN 0x86, OUT 0x04
IN 0x88, OUT 0x05
IN 0x8a, OUT 0x06

0x84 0x03, 0x86 0x04,
 I (34967) gpio: GPIO[16]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (35027) USB_HCDC: usb driver install succeed
I (35027) USB_HCDC: Waitting Device Connection
I (35057) USB_HCDC: USB Port=1 init succeed
I (35057) USB_HCDC: Waitting USB Connection
I (35057) USB_HCDC: Port power: ON
I (35309) USB_HCDC: line 263 HCD_PORT_EVENT_CONNECTION
I (35309) USB_HCDC: Resetting Port
I (35369) USB_HCDC: Port reset succeed
I (35369) USB_HCDC: Getting Port Speed
I (35369) USB_HCDC: Port speed = 1
I (35370) USB_HCDC: USB Speed: full-speed
I (35375) USB_HCDC: Set Device Addr = 1
I (35380) USB_HCDC: Set Device Addr Done
I (35384) USB_HCDC: Set Device Configuration = 1
I (35390) USB_HCDC: Set Device Configuration Done
I (35395) USB_HCDC: Creating bulk in pipe
I (35400) USB_HCDC: Creating bulk out pipe
I (35405) USB_HCDC: Device Connected
I (35409) gpio: GPIO[12]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (35418) gpio: GPIO[13]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
E (35430) esp-modem-dte: esp_dte_handle_line(88): handle line failed
E (35437) esp-modem-dte: esp_dte_handle_line(88): handle line failed
W (35446) esp_modem_dce: esp_modem_dce_generic_command(35): Command:AT&W
...failed
I (35450) modem_board: Current timeouts: 0 and errors: 1
W (35458) esp_modem_dce: esp_modem_dce_generic_command(35): Command:AT+CPIN?
...failed
W (35467) esp_modem_dce: esp_modem_dce_generic_command(35): Command:AT+CPIN=1234
...failed
W (36477) esp_modem_dce: esp_modem_dce_generic_command(35): Command:AT+CPIN?
...failed
E (36477) modem_board: modem_board_start_up(172): store_profile failed

0x81 0x01, 0x82 0x02,0x88 0x05, 0x8a 0x06

I (34361) main: ====================================
I (34366) gpio: GPIO[15]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (34376) gpio: GPIO[17]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (34385) gpio: GPIO[16]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (34445) USB_HCDC: usb driver install succeed
I (34445) USB_HCDC: Waitting Device Connection
I (34475) USB_HCDC: USB Port=1 init succeed
I (34475) USB_HCDC: Waitting USB Connection
I (34475) USB_HCDC: Port power: ON
I (34727) USB_HCDC: line 263 HCD_PORT_EVENT_CONNECTION
I (34727) USB_HCDC: Resetting Port
I (34787) USB_HCDC: Port reset succeed
I (34787) USB_HCDC: Getting Port Speed
I (34787) USB_HCDC: Port speed = 1
I (34788) USB_HCDC: USB Speed: full-speed
I (34793) USB_HCDC: Set Device Addr = 1
I (34799) USB_HCDC: Set Device Addr Done
I (34802) USB_HCDC: Set Device Configuration = 1
I (34809) USB_HCDC: Set Device Configuration Done
I (34813) USB_HCDC: Creating bulk in pipe
I (34818) USB_HCDC: Creating bulk out pipe
I (34823) USB_HCDC: Device Connected
I (34827) gpio: GPIO[12]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (34836) gpio: GPIO[13]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
E (35346) esp-modem-dte: esp_modem_dte_send_cmd(196): process command timeout
 response timeoutem_dce: esp_modem_dce_generic_command(31): Command:AT
I (35351) modem_board: Current timeouts: 1 and errors: 0
E (40357) esp-modem-dte: esp_modem_dte_send_cmd(196): process command timeout
W (40357) esp_modem_dce: esp_modem_dce_generic_command(31): Command:+++ response timeout
E (40862) esp-modem-dte: esp_modem_dte_send_cmd(196): process command timeout
 response timeoutem_dce: esp_modem_dce_generic_command(31): Command:AT
E (41367) esp-modem-dte: esp_modem_dte_send_cmd(196): process command timeout
 response timeoutem_dce: esp_modem_dce_generic_command(31): Command:AT
I (41372) modem_board: Current timeouts: 2 and errors: 0
I (41378) modem_board: Restart to connect modem........
I (41384) modem_board: modem_board_reset!

achao1104, 2021-12-03:
it looks work now, but the message looks like the pppos sample, is it correctly use usb now? my modem is cat4 module.

I (34363) main: ====================================
I (34363) main:      ESP 4G Cat.1 Wi-Fi Router
I (34363) main: ====================================
I (34368) gpio: GPIO[15]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (34378) gpio: GPIO[17]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (34387) gpio: GPIO[16]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (34447) USB_HCDC: usb driver install succeed
I (34447) USB_HCDC: Waitting Device Connection
I (34477) USB_HCDC: USB Port=1 init succeed
I (34477) USB_HCDC: Waitting USB Connection
I (34477) USB_HCDC: Port power: ON
I (34729) USB_HCDC: line 263 HCD_PORT_EVENT_CONNECTION
I (34729) USB_HCDC: Resetting Port
I (34789) USB_HCDC: Port reset succeed
I (34789) USB_HCDC: Getting Port Speed
I (34789) USB_HCDC: Port speed = 1
I (34790) USB_HCDC: USB Speed: full-speed
I (34795) USB_HCDC: Set Device Addr = 1
I (34800) USB_HCDC: Set Device Addr Done
I (34804) USB_HCDC: Set Device Configuration = 1
I (34810) USB_HCDC: Set Device Configuration Done
I (34815) USB_HCDC: Creating bulk in pipe
I (34820) USB_HCDC: Creating bulk out pipe
I (34825) USB_HCDC: Device Connected
I (34829) gpio: GPIO[12]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (34838) gpio: GPIO[13]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (34848) esp-modem-dte: send len 3,AT
I (34854) esp-modem-dte: recv len 6,
OK

I (34857) esp-modem-dte: send len 5,ATE0
I (34864) esp-modem-dte: recv len 6,
OK

I (34867) esp-modem-dte: send len 11,AT+IFC=0,0
I (34874) esp-modem-dte: recv len 6,
OK

I (34877) esp-modem-dte: send len 5,AT&W
I (34884) esp-modem-dte: recv len 6,
OK

I (36887) esp-modem-dte: send len 29,AT+CGDCONT=1,"IP","internet"
I (36907) esp-modem-dte: recv len 6,
OK

I (36907) esp-modem-dte: send len 12,ATD*99***1#
I (36917) esp-modem-dte: recv len 18,
CONNECT 115200

I (36918) modem_board: Modem PPP Started
W (36918) modem_board: PPP netif event = 259
W (36924) modem_board: PPP netif event = 262
W (36934) modem_board: PPP netif event = 263
W (36934) modem_board: PPP netif event = 265
I (36955) esp-netif_lwip-ppp: Connected
I (36955) esp-netif_lwip-ppp: Name Server1: 211.136.17.107
I (36955) esp-netif_lwip-ppp: Name Server2: 211.136.20.203
I (36961) modem_board: IP event! 6
I (36965) modem_board: Modem Connected to PPP Server
I (36971) modem_board: ppp ip: 100.121.15.42, mask: 255.255.255.255, gw: 10.64.64.64
I (36979) modem_board: Main DNS: 211.136.17.107
I (36984) modem_board: Backup DNS: 211.136.20.203
W (36990) modem_board: PPP netif event = 266
I (36995) main: ap dns addr(auto): 211.136.17.107
I (37001) wifi:wifi driver task: 3ffe78d4, prio:23, stack:6656, core=0
I (37006) system_api: Base MAC address is not set
I (37011) system_api: read default base MAC address from EFUSE
I (37020) wifi:wifi firmware version: f84e709
I (37022) wifi:wifi certification version: v7.0
I (37026) wifi:config NVS flash: enabled
I (37030) wifi:config nano formating: disabled
I (37034) wifi:Init data frame dynamic rx buffer num: 32
I (37039) wifi:Init management frame dynamic rx buffer num: 32
I (37045) wifi:Init management short buffer num: 32
I (37049) wifi:Init dynamic tx buffer num: 32
I (37054) wifi:Init static rx buffer size: 1600
I (37058) wifi:Init static rx buffer num: 10
I (37062) wifi:Init dynamic rx buffer num: 32
I (37066) wifi_init: tcpip mbox: 32
I (37070) wifi_init: udp mbox: 6
I (37074) wifi_init: tcp mbox: 6
I (37078) wifi_init: tcp tx win: 5744
I (37082) wifi_init: tcp rx win: 5744
I (37086) wifi_init: tcp mss: 1440
I (37090) wifi_init: WiFi IRAM OP enabled
I (37095) wifi_init: WiFi RX IRAM OP enabled
I (37100) wifi_init: LWIP IRAM OP enabled
I (37105) phy_init: phy_version 1800,e7ef680,Apr 13 2021,11:45:08
I (37199) wifi:mode : softAP (7c:df:a1:09:4a:7b)
I (37201) wifi:Total power save buffer number: 16
I (37201) wifi:Init max length of beacon: 752/752
I (37202) wifi:Init max length of beacon: 752/752
I (37210) wifi:Total power save buffer number: 16
I (37212) modem_wifi: softap ssid: esp_4g_router password: 12345678
I (37218) modem_wifi: NAT is enabled

achao1104, 2021-12-03:
but occasionally, it reports the following error:

W (37751) USB_HCDC: line 271 HCD_PORT_EVENT_ERROR

I (37640) wifi_init: WiFi RX IRAM OP enabled
I (37645) wifi_init: LWIP IRAM OP enabled
I (37650) phy_init: phy_version 1800,e7ef680,Apr 13 2021,11:45:08
I (37722) phy: phy_filt_delay fail!!!, k=11
I (37744) wifi:mode : softAP (7c:df:a1:09:4a:7b)
I (37745) wifi:Total power save buffer number: 8
I (37746) wifi:Init max length of beacon: 752/752
I (37746) wifi:Init max length of beacon: 752/752
W (37751) USB_HCDC: line 271 HCD_PORT_EVENT_ERROR
I (37756) USB_HCDC: hcd port state = 7
E (37760) modem_board: Force reset system
I (37765) gpio: GPIO[13]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (37777) wifi:Total power save buffer number: 8
I (37780) modem_wifi: softap ssid: esp_4g_router password: 12345678
I (37786) modem_wifi: NAT is enabled
W (37974) wifi:age timer already deinit
I (37974) wifi:flush txq

leeebo, 2021-12-03:
@achao1104  `HCD_PORT_EVENT_ERROR` error mostly from hardware connection or power supply.

leofabri, 2022-01-25:
@achao1104 can I ask which `CONFIG_MODEM_USB_OUT_EP_ADDR` & `CONFIG_MODEM_USB_IN_EP_ADDR` you used to get the AT commands to work? like this https://github.com/espressif/esp-iot-solution/issues/121#issuecomment-985229629

achao1104, 2022-01-26:
thanks for your reply, it is 0x86,0x04
I see you only test it with cat1 module, however, it looks work almost time with cat4 module(mine is sim7600jc-h), and just report error occasionally.

commit 9fc08ebf (2021-08-17): usbh_cdc: control line state can be configured

fix #121
closed 2023-07-10 by leeebo