espressif/esp-iot-solution #122: How to set the color inversion?

modi12jin, 2021-08-12:
[image]

```
/* component includes */
#include <stdio.h>

/* freertos includes */
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/timers.h"
#include "freertos/semphr.h"
#include "freertos/queue.h"
#include "freertos/event_groups.h"
#include "esp_freertos_hooks.h"
#include "board.h"
#include "lvgl_gui.h"
#include "esp_log.h"

#include "lv_examples/src/lv_demo_printer/lv_demo_printer.h"
#include "lv_examples/src/lv_demo_widgets/lv_demo_widgets.h"
#include "lv_examples/src/lv_ex_get_started/lv_ex_get_started.h"
#include "lv_examples/src/lv_demo_benchmark/lv_demo_benchmark.h"
#include "lv_examples/src/lv_demo_keypad_encoder/lv_demo_keypad_encoder.h"
#include "lv_examples/src/lv_demo_stress/lv_demo_stress.h"
#include "lv_examples/src/lv_ex_style/lv_ex_style.h"

static const char *TAG = "example_lvgl";
static scr_info_t g_lcd_info;

void app_main()
{
    esp_err_t ret = ESP_OK;
    scr_driver_t lcd_drv;
    touch_panel_driver_t touch_drv;
    i2s_lcd_config_t i2s_lcd_cfg = {
        .data_width  = BOARD_LCD_I2S_BITWIDTH,
        .pin_data_num = {
            BOARD_LCD_I2S_D0_PIN,
            BOARD_LCD_I2S_D1_PIN,
            BOARD_LCD_I2S_D2_PIN,
            BOARD_LCD_I2S_D3_PIN,
            BOARD_LCD_I2S_D4_PIN,
            BOARD_LCD_I2S_D5_PIN,
            BOARD_LCD_I2S_D6_PIN,
            BOARD_LCD_I2S_D7_PIN,
        },
        .pin_num_cs = BOARD_LCD_I2S_CS_PIN,
        .pin_num_wr = BOARD_LCD_I2S_WR_PIN,
        .pin_num_rs = BOARD_LCD_I2S_RS_PIN,

        .clk_freq = 20000000,
        .i2s_port = I2S_NUM_0,
        .buffer_size = 32000,
        .swap_data = true,
    };

    scr_interface_driver_t *iface_drv;
    scr_interface_create(SCREEN_IFACE_8080, &i2s_lcd_cfg, &iface_drv);

    ret = scr_find_driver(SCREEN_CONTROLLER_ILI9488, &lcd_drv);
    if (ESP_OK != ret) {
        return;
        ESP_LOGE(TAG, "screen find failed");
    }

    scr_controller_config_t lcd_cfg = {
        .interface_drv = iface_drv,
        .pin_num_rst = BOARD_LCD_I2S_RESET_PIN,
        .pin_num_bckl = BOARD_LCD_I2S_BL_PIN,
        .rst_active_level = 0,
        .bckl_active_level = 1,
        .offset_hor = 0,
        .offset_ver = 0,
        .width = 320,
        .height = 480,
        .rotate = SCR_SWAP_XY | SCR_MIRROR_Y, /** equal to SCR_DIR_BTLR */
    };
    lcd_drv.init(&lcd_cfg);

        if (ESP_OK != ret) {
        return;
        ESP_LOGE(TAG, "screen initialize failed");
    }

    lcd_drv.get_info(&g_lcd_info);
    ESP_LOGI(TAG, "Screen name:%s | width:%d | height:%d", g_lcd_info.name, g_lcd_info.width, g_lcd_info.height);

    touch_panel_config_t touch_cfg = {
        .interface_i2c = {
            .i2c_bus = BOARD_TOUCH_I2C_PORT_NUM,
            .clk_freq = 100000,
            .i2c_addr = 0X38,
        },
        .interface_type = TOUCH_PANEL_IFACE_I2C,
        .pin_num_int = -1,
        .direction = TOUCH_DIR_TBLR,
        .width = 320,
        .height = 480,
    };
    touch_panel_find_driver(TOUCH_PANEL_CONTROLLER_FT5X06, &touch_drv);
    touch_drv.init(&touch_cfg);
    touch_drv.calibration_run(&lcd_drv, false);

    /* Initialize LittlevGL GUI */
    lvgl_init(&lcd_drv, &touch_drv);

#ifdef CONFIG_LV_DEMO_BENCHMARK
    lv_demo_benchmark();
#elif defined CONFIG_LV_DEMO_PRINTER
    lv_demo_printer();
#elif defined CONFIG_LV_DEMO_WIDGETS
    lv_demo_widgets();
#elif defined CONFIG_LV_EX_GET_STARTED
    lv_ex_get_started_1();
#elif defined CONFIG_LV_DEMO_STRESS
    lv_demo_stress();
#elif defined CONFIG_LV_EX_STYLE
    lv_ex_style_1();
#endif

    ESP_LOGI(TAG, "[APP] IDF version: %s", esp_get_idf_version());
    ESP_LOGI(TAG, "[APP] Free memory: %d bytes", esp_get_free_heap_size());
}
```
[image]

[image]

[image]

[image]

TDA-2030, 2021-08-13:
@modi12jin 
```
#include "ili9488.h"

lcd_ili9488_set_invert(1);
```

closed 2021-08-14 by modi12jin

modi12jin, 2021-08-16:
@TDA-2030  I don't display the image in 16 bit

[image]

```
/* component includes */
#include <stdio.h>

/* freertos includes */
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/timers.h"
#include "freertos/semphr.h"
#include "freertos/queue.h"
#include "freertos/event_groups.h"
#include "esp_freertos_hooks.h"
#include "board.h"
#include "lvgl_gui.h"
#include "esp_log.h"

#include "lv_examples/src/lv_demo_printer/lv_demo_printer.h"
#include "lv_examples/src/lv_demo_widgets/lv_demo_widgets.h"
#include "lv_examples/src/lv_ex_get_started/lv_ex_get_started.h"
#include "lv_examples/src/lv_demo_benchmark/lv_demo_benchmark.h"
#include "lv_examples/src/lv_demo_keypad_encoder/lv_demo_keypad_encoder.h"
#include "lv_examples/src/lv_demo_stress/lv_demo_stress.h"
#include "lv_examples/src/lv_ex_style/lv_ex_style.h"
#include "ili9488.h"

static const char *TAG = "example_lvgl";
static scr_info_t g_lcd_info;

void app_main()
{
    esp_err_t ret = ESP_OK;
    scr_driver_t lcd_drv;
    touch_panel_driver_t touch_drv;
    i2s_lcd_config_t i2s_lcd_cfg = {
        .data_width  = BOARD_LCD_I2S_BITWIDTH,
        .pin_data_num = {
            BOARD_LCD_I2S_D0_PIN,
            BOARD_LCD_I2S_D1_PIN,
            BOARD_LCD_I2S_D2_PIN,
            BOARD_LCD_I2S_D3_PIN,
            BOARD_LCD_I2S_D4_PIN,
            BOARD_LCD_I2S_D5_PIN,
            BOARD_LCD_I2S_D6_PIN,
            BOARD_LCD_I2S_D7_PIN,
            BOARD_LCD_I2S_D8_PIN,
            BOARD_LCD_I2S_D9_PIN,
            BOARD_LCD_I2S_D10_PIN,
            BOARD_LCD_I2S_D11_PIN,
            BOARD_LCD_I2S_D12_PIN,
            BOARD_LCD_I2S_D13_PIN,
            BOARD_LCD_I2S_D14_PIN,
            BOARD_LCD_I2S_D15_PIN,       
        },
        .pin_num_cs = BOARD_LCD_I2S_CS_PIN,
        .pin_num_wr = BOARD_LCD_I2S_WR_PIN,
        .pin_num_rs = BOARD_LCD_I2S_RS_PIN,

        .clk_freq = 20000000,
        .i2s_port = I2S_NUM_0,
        .buffer_size = 32000,
        .swap_data = true,
    };

    scr_interface_driver_t *iface_drv;
    scr_interface_create(SCREEN_IFACE_8080, &i2s_lcd_cfg, &iface_drv);

[... 144 lines cut ...]
[0;32mI (202) boot: Loaded app from partition at offset 0x10000[0m
[0;32mI (202) boot: Disabling RNG early entropy source...[0m
[0;32mI (217) cac���f~r dynamic allocation:[0m
[0;32mI (269) heap_init: At 3FF9E000 len 00002000 (8 KiB): RTCRAM[0m
[0;32mI (275) heap_init: At 3FFC3610 len 000389F0 (226 KiB): DRAM[0m
[0;32mI (282) heap_init: At 3FFFC000 len 00003A10 (14 KiB): DRAM[0m
[0;32mI (288) spi_flash: detected chip: generic[0m
[0;32mI (293) spi_flash: flash io: dio[0m
[0;32mI (297) cpu_start: Starting scheduler on PRO CPU.[0m
[0;32mI (303) ESP32S2_I2S_LCD: lcd_buffer_size: 32000, lcd_dma_size: 4000, lcd_dma_node_cnt: 8[0m
[0;32mI (303) ESP32S2_I2S_LCD: lcd init ok[0m
[0;32mI (753) ILI9488: MADCTL=0xe8[0m
[0;32mI (753) example_lvgl: Screen name:ILI9488 | width:480 | height:320[0m
[0;32mI (753) i2c_bus: i2c0 bus inited[0m
[0;32mI (753) FT5x06: FT5x06 frameware version [4][0m
[0;32mI (753) FT5x06: Touch panel size width: 480, height: 320[0m
[0;32mI (763) FT5x06: Initial successful | GPIO INT:-1 | ADDR:0x38 | dir:6[0m
[0;32mI (773) lvgl_gui: GUI Run at esp32s2 Pinned to Core0[0m
[0;32mI (783) lvgl adapter: Alloc memory total size: 122880 Byte[0m
[0;32mI (783) lvgl_gui: Start to run LVGL[0m
[0;32mI (873) example_lvgl: [APP] IDF version: 4.3.0[0m
[0;32mI (873) example_lvgl: [APP] Free memory: 66972 bytes[0m
```

[image]