espressif/esp-iot-solution #124: 在USB例子里面，esp32-s2-kaluga 开发板无法点亮LCD。

xuhongv, 2021-08-19:
## Environment

- Development Kit:      ``esp32-s2-kaluga-Kit V1.2``  ,  ``ESP-LyraP-LCD32 v1.1 ``
- IDF version :`` v4.4-dev-2184-g166c30e7b2``
- Build System:         ``CMake``
- Compiler version: ``xtensa-esp32-elf-gcc (crosstool-NG esp-2021r1) 8.4.0``
- Operating System:    `` Linux``
- Power Supply:         ``USB``
- project ： examples /usb /host /usb_camera_lcd_display

## Problem Description

在usb_camera_lcd_display例子里面，已经按照 README 文档里面按照好 ESP-IDF 环境，并且打好了补丁。

但是在 ```esp32-s2-kaluga``` 开发板无法正常驱动LCD屏幕，开启了使能 boot animation 开启开机动画，但依然无法点亮。

## 下面是设备运行日志：

```
I (21) boot: ESP-IDF v4.4-dev-2184-g166c30e7b2-dirty 2nd stage bootloader
I (21) boot: compile time 23:33:05
I (21) boot: chip revision: 0
I (23) qio_mode: Enabling default flash chip QIO
I (27) boot.esp32s2: SPI Speed      : 80MHz
I (31) boot.esp32s2: SPI Mode       : QIO
I (35) boot.esp32s2: SPI Flash Size : 4MB
I (39) boot: Enabling RNG early entropy source...
I (43) boot: Partition Table:
I (46) boot: ## Label            Usage          Type ST Offset   Length
I (52) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (59) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (65) boot:  2 factory          factory app      00 00 00010000 00177000
I (72) boot:  3 storage          Unknown data     01 82 00187000 001f4000
I (78) boot: End of partition table
I (82) esp_image: segment 0: paddr=00010020 vaddr=3f000020 size=0ec4ch ( 60492) map
I (99) esp_image: segment 1: paddr=0001ec74 vaddr=3ffc1d90 size=013a4h (  5028) load
I (101) esp_image: segment 2: paddr=00020020 vaddr=40080020 size=43ed4h (278228) map
I (152) esp_image: segment 3: paddr=00063efc vaddr=3ffc3134 size=00b20h (  2848) load
I (152) esp_image: segment 4: paddr=00064a24 vaddr=40024000 size=0dd88h ( 56712) load
I (168) esp_image: segment 5: paddr=000727b4 vaddr=50000000 size=00010h (    16) load
I (175) boot: Loaded app from partition at offset 0x10000
I (175) boot: Disabling RNG early entropy source...
I (186) cache: Instruction cache 	: size 8KB, 4Ways, cache line size 32Byte
I (186) cache: Data cache 		: size 8KB, 4Ways, cache line size 32Byte
I (188) spiram: Found 16MBit SPI RAM device
I (192) spiram: SPI RAM mode: sram 40m
I (195) spiram: PSRAM initialized, cache is in normal (1-core) mode.
I (201) cpu_start: Pro cpu up.
I (614) spiram:€槝0f捞槅榝f
R玍隟W仮曂褋z-5R蘒 (623) cpu_start: Pro cpu start user code
I (623) cpu_start: cpu freq: 240000000
I (623) cpu_start: Application information:
I (623) cpu_start: Project name:     uvc_mjpeg_lcd
I (628) cpu_start: App version:      b643b6a-dirty
I (632) cpu_start: Compile time:     Aug 19 2021 23:35:26
I (638) cpu_start: ELF file SHA256:  75fae21acf144609...
I (643) cpu_start: ESP-IDF:          v4.4-dev-2184-g166c30e7b2-dirty
I (649) heap_init: Initializing. RAM available for dynamic allocation:
I (655) heap_init: At 3FF9E000 len 00002000 (8 KiB): RTCRAM
I (660) heap_init: At 3FFC48A8 len 00037758 (221 KiB): DRAM
I (665) heap_init: At 3FFFC000 len 00003A10 (14 KiB): DRAM
I (671) spiram: Adding pool of 2048K of external SPI memory to heap allocator
I (678) spi_flash: detected chip: gd
I (681) spi_flash: flash io: qio
I (684) cpu_start: Starting scheduler on PRO CPU.
I (688) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (688) spi_bus: SPI2 bus created
I (698) spi_bus: SPI2 bus device added, CS=11 Mode=0 Speed=40000000
I (898) lcd st7789: MADCTL=60
I (898) uvc_demo: Screen name:ST7789 | width:320 | height:240
I (4408) UVC_STREAM: UVC Streaming Config Succeed
I (4458) UVC_STREAM: UVC Streaming Starting

```

leeebo, 2021-08-20:
ESP-LyraP-LCD32 v1.1 may have different LCD controller types (ST7789 or ILI9341), as the default configuration does not work (ST7789), please [modify your controller type ](https://github.com/espressif/esp-iot-solution/blob/usb/add_usb_solutions/examples/usb/host/usb_camera_lcd_display/main/main.c#L57-L59)to `SCREEN_CONTROLLER_ILI9341` then try again.

xuhongv, 2021-08-20:
i try it @leeebo

xuhongv, 2021-08-21:
你好，我试着更新了最新版的 esp-iot-solution 仓库，但是报错异常如下，看样子是 CMake 语法，您看看是不是你们开发同事在合并代码出问题了。  @leeebo 

```
../main/cmd_wifi.c:38:10: fatal error: tinyusb.h: No such file or directory
 #include "tinyusb.h"
          ^~~~~~~~~~~
compilation terminated.
```

leeebo, 2021-08-23:
@xuhongv From my near test, the usb-dongle example works as normal with the latest master ESP-IDF, please check if your development environment is installed correctly (include IDF submodules), and try to build other[ usb/device examples](https://github.com/espressif/esp-iot-solution/tree/usb/add_usb_solutions/examples/usb/device).

leeebo, 2021-08-23:
>   However, the component manager is not enabled.

The warning message should not cause your problem, while if you want to remove them, you can Install ESP-IDF component manager with:
1. `. ./export.sh` to add ESP-IDF environment values
2. `pip install idf-component-manager --upgrade`

xuhongv, 2021-08-24:
@leeebo  请问这个 usb/device examples 的运行分支是最新的master分支，还是指定的 commitID ？

1）我使用最新的master分支代码，报错：

```
CMake Error at /mnt/d/ESPRESSIF/Linux_Windows/esp-idf-s3/esp-idf/tools/cmake/component.cmake:306 (message):
  Include directory
  '/mnt/d/ESPRESSIF/Linux_Windows/esp-idf-s3/esp-idf/components/cbor/tinycbor/src'
  is not a directory.
Call Stack (most recent call first):
  /mnt/d/ESPRESSIF/Linux_Windows/esp-idf-s3/esp-idf/tools/cmake/component.cmake:479 (__component_add_include_dirs)
  /mnt/d/ESPRESSIF/Linux_Windows/esp-idf-s3/esp-idf/components/cbor/CMakeLists.txt:1 (idf_component_register)

-- Configuring incomplete, errors occurred!
```
-------------------------------------------

2）我使用 README 文档中指定的分支，而且尝试更新了子模块管理组件，如下：

```
xuhongv@xuhongv:/mnt/d/ESPRESSIF/Linux_Windows/usb_camera_lcd_display/esp-idf$ pip install idf-component-manager --upgrade
Collecting idf-component-manager
  Downloading idf_component_manager-0.2.100b0-py3-none-any.whl (49 kB)
     |████████████████████████████████| 49 kB 1.5 MB/s
Collecting requests
  Downloading requests-2.26.0-py2.py3-none-any.whl (62 kB)
     |████████████████████████████████| 62 kB 883 kB/s
Collecting schema
  Downloading schema-0.7.4-py2.py3-none-any.whl (16 kB)
Requirement already satisfied: future in /home/xuhongv/.espressif/python_env/idf4.4_py3.6_env/lib/python3.6/site-packages (from idf-component-manager) (0.18.2)
Collecting pyyaml
  Downloading PyYAML-5.4.1-cp36-cp36m-manylinux1_x86_64.whl (640 kB)
     |████████████████████████████████| 640 kB 167 kB/s
Collecting requests-toolbelt
  Downloading requests_toolbelt-0.9.1-py2.py3-none-any.whl (54 kB)
     |████████████████████████████████| 54 kB 254 kB/s
Collecting semantic-version>="2.8"
  Downloading semantic_version-2.8.5-py2.py3-none-any.whl (15 kB)
Requirement already satisfied: six in /home/xuhongv/.espressif/python_env/idf4.4_py3.6_env/lib/python3.6/site-packages (from idf-component-manager) (1.15.0)
Collecting tqdm
  Downloading tqdm-4.62.2-py2.py3-none-any.whl (76 kB)
     |████████████████████████████████| 76 kB 249 kB/s
Collecting charset-normalizer~=2.0.0
  Downloading charset_normalizer-2.0.4-py3-none-any.whl (36 kB)
Collecting certifi>=2017.4.17
  Downloading certifi-2021.5.30-py2.py3-none-any.whl (145 kB)
     |████████████████████████████████| 145 kB 224 kB/s
Collecting urllib3<1.27,>=1.21.1
  Downloading urllib3-1.26.6-py2.py3-none-any.whl (138 kB)
     |████████████████████████████████| 138 kB 482 kB/s
Collecting idna<4,>=2.5
  Downloading idna-3.2-py3-none-any.whl (59 kB)
     |████████████████████████████████| 59 kB 291 kB/s
Collecting contextlib2>=0.5.5
  Downloading contextlib2-21.6.0-py2.py3-none-any.whl (13 kB)
Installing collected packages: urllib3, idna, charset-normalizer, certifi, requests, contextlib2, tqdm, semantic-version, schema, requests-toolbelt, pyyaml, idf-component-manager
Successfully installed certifi-2021.5.30 charset-normalizer-2.0.4 contextlib2-21.6.0 idf-component-manager-0.2.100b0 idna-3.2 pyyaml-5.4.1 requests-2.26.0 requests-toolbelt-0.9.1 schema-0.7.4 semantic-version-2.8.5 tqdm-4.62.2 urllib3-1.26.6
WARNING: You are using pip version 21.0.1; however, version 21.2.4 is available.
You should consider upgrading via the '/home/xuhongv/.espressif/python_env/idf4.4_py3.6_env/bin/python -m pip install --upgrade pip' command.
```

再一次编译，如下错误。

```
../main/CLI_Commands.c:54:10: fatal error: tusb_cdc_acm.h: No such file or directory
 #include "tusb_cdc_acm.h"
          ^~~~~~~~~~~~~~~~
compilation terminated.
```

leeebo, 2021-08-30:
@xuhongv For convenience, we have added some modify in examples' cmakelists,  you can build with the IDF `b63ec47238fd6a` or latest master directly.

commit f8dc3f33 (2021-08-30): fix(examples/usb): cmakelist add default iot_solution path

fix #124, fix #126
closed 2021-11-16 by xuhongv