espressif/esp-iot-solution #129: can "usb_cdc_4g_module" support cat4 module?

achao1104, 2021-09-06:
Hello,

It is good to see "usb_cdc_4g_module" support more modules, but it looks all the supported modules are 4g cat1. "usb_cdc_4g_module" can support cat4 module?

Peter

leeebo, 2021-09-06:
Yes, Most vendor's Cat.4 modules using Compatible AT commands with CAT.1, which should work as normally as Cat.1 through the USB PPP interface. But as ESP32-S2/S3 only has USB Fullspeed (12Mbps), it's more suitable to use Cat.1.

Maybe we would add Cat.4 supported after ESP enters the USB 2.0 Highspeed (480Mbps) era.

achao1104, 2021-09-06:
we are using sim7600jc which is cat4. shall we have to change to cat1 module, or do you have any other way? I recalled you mentioned some usb isolater.

leeebo, 2021-09-06:
@achao1104 Yes, add a USB full-speed isolator, then attach it to the PC, find the real endpoint address of the PPP interface (Because it is a vendor specified class, the program cannot automatically identify it).

achao1104, 2021-09-06:
thanks for your reply. what is a "USB full-speed isolator", do you have a sample link?

leeebo, 2021-09-14:
@achao1104 Like this one:
[image]

closed 2023-10-16 by leeebo