espressif/esp-iot-solution #130: napt issue in "usb_cdc_4g_module" when there is other network

achao1104, 2021-09-09:
Hello,

Besides "usb_cdc_4g_module", we init the ethernet network at the same time, so there are actually two networks in the esp32s2.
how the command below to specify which network to use?
ip_napt_enable(_g_esp_netif_soft_ap_ip.ip.addr, 1);

Thanks,

Peter

achao1104, 2021-09-12:
could anybody help? thanks a lot.

leeebo, 2021-09-14:
Hi @achao1104 , until now we have no plan to support the ethernet interface.
Just add some suggestions here:
1. Please create an ethernet interface like `ap_netif`
2. Set the `DHCP` information of the ethernet interface
3. Enable `NAPT` with the IP of the ethernet interface (you can use esp_netif_get_ip_info to get IP)

achao1104, 2021-09-14:
thanks for your reply. my case is that, in almost time, we use usb 4g and enable napt, it works fine. and seldomly, we turn on w5500 and read some data from slave device via w5500 and then turn off w5500. the problem is that, once w5500 tcp/ip is enabled, the napt will be freezed, I guess napt won't know router to which network interface once another network interface is created.

leeebo, 2021-09-14:
@achao1104 You can try to disable napt of wifi interface using `ip_napt_enable(_g_esp_netif_soft_ap_ip.ip.addr, 0)` before turn on  w5500.

achao1104, 2021-09-14:
thanks for your reply. Is there a way to make napt skip w5500 network interface?

leeebo, 2021-09-14:
@achao1104 Sorry about the misunderstanding, so your question is how to prevent Wi-Fi packets router to ethernet interface, keep forward them to 4G PPP interface, right?

achao1104, 2021-09-14:
yes

achao1104, 2021-09-14:
your understanding is correct. it looks no way to prevent wifi packets router to ethernet interface.

leeebo, 2022-02-24:
@achao1104 I think because of the `ethernet interface` has higher route_prio than `PPP interface`, cause the packets are forwarded to ethernet. [Please increase PPP's route_prio from `20` to `60` to have a try](https://github.com/espressif/esp-idf/blob/release/v4.4/components/esp_netif/include/esp_netif_defaults.h#L67).

closed 2023-10-07 by leeebo