espressif/esp-iot-solution #131: Screen example with NT35510 over SPI not working according to the datasheet

DCSBL, 2021-09-16:
## Environment

- Development Kit: None
- ~Kit version (for WroverKit/PicoKit/DevKitC):~
- Module or chip used: ESP32-S2
- IDF version: `v4.3`
- Build System: CMake
- Compiler version: `xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0`
- Operating System: MacOS
- Power Supply: external 5V

## Problem Description
I've updated `examples/common_components/boards` to include my custom board and (tried to) configure the `examples/screen/` project with the NT35510 IC over SPI. The screen does not turn on. This can be te result of many factors but I've found one thing that I can't figure out, listed below:

### Expected Behavior
According to [the datasheet](https://focuslcds.com/content/NT35510.pdf) at page 43, a SPI instruction/write is packed as 3 transmits each with a 'command' byte and a data byte.

### Actual Behavior
Looking at [nt35510.c : L254-L636](https://github.com/espressif/esp-iot-solution/blob/master/components/display/screen/controller_driver/nt35510/nt35510.c#L254-L638), I see that it sends the initial configuration. When probing the data that is send, I can see that only the LSB byte is send. It looks like it happens [because of this](https://github.com/espressif/esp-iot-solution/blob/master/components/display/screen/interface_driver/scr_interface_driver.c#L212)

Also, a `dc` line is used to indicate a command, but this should be passed with the 'command' byte.

This is a screenshot of the analyser capturing these instructions:
```c
LCD_WRITE_CMD(0x1200);
LCD_WRITE_REG(0xf000, 0x0055);
LCD_WRITE_REG(0xf001, 0x00aa);
LCD_WRITE_REG(0xf002, 0x0052);
LCD_WRITE_REG(0xf003, 0x0008);
LCD_WRITE_REG(0xf004, 0x0001);
```
[image]

It only sends `0x00 0x00 0x55 0x01 0xAA 0x02 0x52 0x03 0x08 0x04 0x01`

### Steps to repropduce
*Can include this when required*

### Code to reproduce this issue

```cpp
void app_main()
    esp_err_t ret = ESP_OK;
    
    iot_board_init();
    spi_bus_handle_t spi_bus = iot_board_get_handle(BOARD_SPI2_ID);
    
    scr_interface_spi_config_t spi_lcd_cfg = 
    {
        .spi_bus = spi_bus,             /*!< Handle of spi bus */
        .pin_num_cs = BOARD_PIN_SPI_CS, /*!< SPI Chip Select Pin*/
        .pin_num_dc = BOARD_PIN_DE,     /*!< Pin to select Data or Command for LCD */
        .clk_freq = 3000000,            /*!< SPI clock frequency */
        .swap_data = true,              /*!< Whether to swap data */
    };
    
    scr_interface_driver_t *iface_drv;
    scr_interface_create(SCREEN_IFACE_SPI, &spi_lcd_cfg, &iface_drv);
    ret = scr_find_driver(SCREEN_CONTROLLER_NT35510, &g_lcd);
    if (ESP_OK != ret) {
        return;
        ESP_LOGE(TAG, "screen find failed");
    }
    
    scr_controller_config_t lcd_cfg = {
        .interface_drv = iface_drv,
        .pin_num_rst = BOARD_LCD_RESET_PIN,
        .pin_num_bckl = BOARD_LCD_BL_PIN,
        .rst_active_level = 0,
        .bckl_active_level = 1,
        .offset_hor = 0,
        .offset_ver = 0,
        .width = 480,
        .height = 800,
        .rotate = SCR_SWAP_XY | SCR_MIRROR_Y, /** equal to SCR_DIR_BTLR */
    };
    ret = g_lcd.init(&lcd_cfg);

    if (ESP_OK != ret) {
        return;
        ESP_LOGE(TAG, "screen initialize failed");
    }

    g_lcd.get_info(&g_lcd_info);
    ESP_LOGI(TAG, "Screen name:%s | width:%d | height:%d", g_lcd_info.name, g_lcd_info.width, g_lcd_info.height);
}
```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs

```
I (266) cpu_start: Starting scheduler on PRO CPU.
I (272) spi_bus: SPI2 bus created
I (275) Board: Board Info: ESP32-LCDKit
I (280) Board: Board Init Done ...
I (285) spi_bus: SPI2 bus device added, CS=41 Mode=0 Speed=3000000
I (491) lcd nt35510: MADCTL=0xa0
I (491) screen example: Screen name:NT35510 | width:800 | height:480
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

DCSBL, 2021-09-17:
i'd figured out that we better use the 8080 interface, so _my_ problem seems solved. But is the described issue above something that has to be fixed?

alic-maker, 2021-09-18:
Sorry, I don't have an nt35510 screen module with SPI interface. In fact, I didn't test through SPI, and this application is scarce.

DCSBL, 2021-09-21:
I will close this one. Got it working with the 8080 interface. Maybe we only need to add something to a doc somewhere to tell that SPI is not supported?

closed 2021-09-21 by DCSBL