espressif/esp-iot-solution #136: Add support for ESP32-S3 I2S (feature request)

DCSBL, 2021-11-04:
## Environment

- Development Kit: none
- Kit version (for WroverKit/PicoKit/DevKitC): -
- Module or chip used:  ESP32-S3
- IDF version (run ``git describe --tags`` to find it): V4.3.1
- Build System: CMake
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): `xtensa-esp32-elf-gcc (crosstool-NG esp-2021r1) 8.4.0`
- Operating System: macOS
- Power Supply: USB

## Problem Description
To allow the ESP32-S3 to control a LCD screen using these components, I2S supports needs to be added. When looking at [these files](https://github.com/espressif/esp-iot-solution/tree/master/components/bus), only the ESP32 and ESP32-S2 is supported.

leeebo, 2022-01-07:
@DCSBL ESP32-S3 LCD driver is under development, will mention you if ready

DCSBL, 2022-01-07:
Thanks!
I recently found out that IDF V4.4 natively supports LCD over I2S. So for me it is not required anymore to have the support here.

commit 0caa2de5 (2022-01-20): bus: add screen 8080 interface for esp32s3; closes #136
closed 2022-01-20 by zhanzhaocheng