espressif/esp-iot-solution #140: assert when call uvc_streaming_stop()

Dandjinh, 2021-11-25:
### error

100% happens

[image]

### esp-idf version

```
commit cef377e4a43ead59418c9abf61f27f253036d681 (HEAD -> ob_usb)
Author: Dandjinh Deng <Dandjinh_Deng@on-bright.com>
Date:   Thu Nov 11 10:05:34 2021 +0800

    patch to spi_master.c according to iot-solution/usb required

commit 6438af2ef98a45690e8f83a99b25f0797c26618b (origin/release/v4.4, release/v4.4)
Merge: 44d13c648b 1ed627d022
Author: Zim Kalinowski <zim.kalinowski@espressif.com>
Date:   Wed Nov 10 10:24:38 2021 +0000

    Merge branch 'bugfix/simplify_codeowners_v4.4' into 'release/v4.4'

    gitlab: simplify approvals for backports (v4.4)

    See merge request espressif/esp-idf!15865
```

### esp-iot-solution version:

```
commit 450e8166de8a675c16116d4934d9c2e1c8500aa0 (HEAD -> usb/add_usb_solutions, origin/usb/add_usb_solutions)
Merge: d036d0f e4148e8
Author: Li Bo <libo@espressif.com>
Date:   Mon Nov 22 07:05:22 2021 +0000

    Merge branch 'usb/add_hid_device_example' into 'usb/add_usb_solutions'

    examples(hid): add usb hid device example

    See merge request rd/esp-iot-solution!551
```

leeebo, 2021-11-25:
@Dandjinh Hi this's a known bug from ESP-IDF USB Host Controler and has been fixed in the latest commits [de6bf09f40](https://github.com/espressif/esp-idf/commit/de6bf09f4099a61ae926cc42d8d2a5f53d17c4a3) and [0a948594](https://github.com/espressif/esp-idf/commit/0a948594cb0d8e2dbc0d3f80b4b81489e6e08be1), you can checkout to latest `master ` or `release/v4.4` branch to try it.

Dandjinh, 2021-12-02:
@leeebo Thanks! It works well with the lastest idf.

closed 2021-12-02 by Dandjinh