espressif/esp-iot-solution #141: USB HOST demo运行报错

mkandy, 2021-12-01:
我正在尝试运行esp-iot-solution\examples\usb\host\usb_cdc_basic-"usb_cdc_basic" demo,Eclipse+esp-idf4.3.1 编译时提示
CMake Error at F:/ESP32/tools/cmake/build.cmake:191 (message):
  Failed to resolve component 'usb'.
Call Stack (most recent call first):
  F:/ESP32/tools/cmake/build.cmake:217 (__build_resolve_and_add_req)
  F:/ESP32/tools/cmake/build.cmake:441 (__build_expand_requirements)
  F:/ESP32/tools/cmake/project.cmake:399 (idf_build_process)
  CMakeLists.txt:19 (project)

leeebo, 2021-12-01:
Hi, USB Host not supported in IDF 4.3.1, please checkout your ESP-IDF to the latest release/v4.4 or master branch for the testing. 

Same issue #134

mkandy, 2021-12-02:
@leeebo thank you for your helping but i am tring to install the "ESP-IDF: release/v4.4 (release branch)",VScode IDE and esp-idf-tools-setup-online-2.12.exe both failed,
1.
VScode failed:
Cloning into 'C:/Users/dengweian/esp/esp-idf/components/tinyusb/tinyusb'...
fatal: unable to access 'https://github.com/espressif/tinyusb.git/': Failed to connect to github.com port 443: Timed out
fatal: clone of 'https://github.com/espressif/tinyusb.git' into submodule path 'C:/Users/dengweian/esp/esp-idf/components/tinyusb/tinyusb' failed
Failed to clone 'components/tinyusb/tinyusb'. Retry scheduled

Cloning into 'C:/Users/dengweian/esp/esp-idf/components/unity/unity'...
fatal: unable to access 'https://github.com/ThrowTheSwitch/Unity.git/': Failed to connect to github.com port 443: Timed out
fatal: clone of 'https://github.com/ThrowTheSwitch/Unity.git' into submodule path 'C:/Users/dengweian/esp/esp-idf/components/unity/unity' failed
Failed to clone 'components/unity/unity'. Retry scheduled

Cloning into 'C:/Users/dengweian/esp/esp-idf/examples/build_system/cmake/import_lib/main/lib/tinyxml2'...
fatal: unable to access 'https://github.com/leethomason/tinyxml2.git/': Failed to connect to github.com port 443: Timed out
fatal: clone of 'https://github.com/leethomason/tinyxml2.git' into submodule path 'C:/Users/dengweian/esp/esp-idf/examples/build_system/cmake/import_lib/main/lib/tinyxml2' failed
Failed to clone 'examples/build_system/cmake/import_lib/main/lib/tinyxml2'. Retry scheduled

Cloning into 'C:/Users/dengweian/esp/esp-idf/examples/peripherals/secure_element/atecc608_ecdsa/components/esp-cryptoauthlib'...
fatal: unable to access 'https://github.com/espressif/esp-cryptoauthlib.git/': OpenSSL SSL_read: Connection was reset, errno 10054
fatal: clone of 'https://github.com/espressif/esp-cryptoauthlib.git' into submodule path 'C:/Users/dengweian/esp/esp-idf/examples/peripherals/secure_element/atecc608_ecdsa/components/esp-cryptoauthlib' failed
Failed to clone 'examples/peripherals/secure_element/atecc608_ecdsa/components/esp-cryptoauthlib'. Retry scheduled

Cloning into 'C:/Users/dengweian/esp/esp-idf/components/bt/controller/lib_esp32'...
error: RPC failed; curl 56 OpenSSL SSL_read: Connection was reset, errno 10054
fatal: expected flush after ref listing
fatal: clone of 'https://github.com/espressif/esp32-bt-lib.git' into submodule path 'C:/Users/dengweian/esp/esp-idf/components/bt/controller/lib_esp32' failed
Failed to clone 'components/bt/controller/lib_esp32' a second time, aborting

2.
and the ESP IDE tools failed report:
                          WARNING: Failed to download https
                          /github.com/espressif/crosstool-NG/releases/download/esp-2021r2/xtensa-esp32s2-elf-gcc8_4_0-esp-2021r2-patch1-win64.zip to F:\esp32_tools\.espressif\dist\xtensa-esp32s2-elf-gcc8_4_0-esp-2021r2-patch1-win64.zip.tmp
                          Downloading xtensa-esp32s2-elf-gcc8_4_0-esp-2021r2-patch1-win64.zip to F:\esp32_tools\.espressif\dist\xtensa-esp32s2-elf-gcc8_4_0-esp-2021r2-patch1-win64.zip.tmp
                          Done
                          Extracting F:\esp32_tools\.espressif\dist\xtensa-esp32s2-elf-gcc8_4_0-esp-2021r2-patch1-win64.zip to F:\esp32_tools\.espressif\tools\xtensa-esp32s2-elf\esp-2021r2-8.4.0
                          Installing xtensa-esp32s3-elf@esp-2021r2-8.4.0
                          Downloading xtensa-esp32s3-elf-gcc8_4_0-esp-2021r2-patch1-win64.zip to F:\esp32_tools\.espressif\dist\xtensa-esp32s3-elf-gcc8_4_0-esp-2021r2-patch1-win64.zip.tmp
                          Done
                          Extracting F:\esp32_tools\.espressif\dist\xtensa-esp32s3-elf-gcc8_4_0-esp-2021r2-patch1-win64.zip to F:\esp32_tools\.espressif\tools\xtensa-esp32s3-elf\esp-2021r2-8.4.0
                          Installing riscv32-esp-elf@esp-2021r2-8.4.0
                          Downloading riscv32-esp-elf-gcc8_4_0-esp-2021r2-patch1-win64.zip to F:\esp32_tools\.espressif\dist\riscv32-esp-elf-gcc8_4_0-esp-2021r2-patch1-win64.zip.tmp
                          WARNING: Download failure [WinError 10060] 由于连接方在一段时间后没有正确答复或连接的主机没有反应，连接尝试失败。
                          WARNING: Failed to download https
                          /github.com/espressif/crosstool-NG/releases/download/esp-2021r2/riscv32-esp-elf-gcc8_4_0-esp-2021r2-patch1-win64.zip to F:\esp32_tools\.espressif\dist\riscv32-esp-elf-gcc8_4_0-esp-2021r2-patch1-win64.zip.tmp
                          Downloading riscv32-esp-elf-gcc8_4_0-esp-2021r2-patch1-win64.zip to F:\esp32_tools\.espressif\dist\riscv32-esp-elf-gcc8_4_0-esp-2021r2-patch1-win64.zip.tmp
                          WARNING: Download failure [WinError 10060] 由于连接方在一段时间后没有正确答复或连接的主机没有反应，连接尝试失败。
                          WARNING: Failed to download https
                          /github.com/espressif/crosstool-NG/releases/download/esp-2021r2/riscv32-esp-elf-gcc8_4_0-esp-2021r2-patch1-win64.zip to F:\esp32_tools\.espressif\dist\riscv32-esp-elf-gcc8_4_0-esp-2021r2-patch1-win64.zip.tmp
                          Downloading riscv32-esp-elf-gcc8_4_0-esp-2021r2-patch1-win64.zip to F:\esp32_tools\.espressif\dist\riscv32-esp-elf-gcc8_4_0-esp-2021r2-patch1-win64.zip.tmp
                          WARNING: Download failure <urlopen error [WinError 10060] 由于连接方在一段时间后没有正确答复或连接的主机没有反应，连接尝试失败。>
                          WARNING: Failed to download https
[... cut ...]
                              raise DownloadError()
                          __main__.DownloadError
                          
2021-12-02 09:18:57.299   --------
2021-12-02 09:18:57.299   Message box (OK):
                          安装失败，退出代码为 1
2021-12-02 09:22:54.305   User chose OK.
2021-12-02 09:22:54.312   Message box (Yes/No):
                          已创建安装日志，其中可能包含有关该问题的详细信息。
                          是否需要查看安装日志？
2021-12-02 09:22:57.056   User chose Yes.
2021-12-02 09:22:57.147   CurStepChanged raised an exception.

mkandy, 2021-12-02:
I have reinstalled the esp-idf-4.4,now it works fine ,thank you