espressif/esp-iot-solution #142: USB HOST CDC example

mkandy, 2021-12-03:
Hello,
Esp32 have not print received messages 
Board: esp32S2
Demo: ESP IoT Solution /example/usb host basic.
Device: USB device UART(not used "AT" mode)
Build IDE : VScode + esp32-idf V4.4
Running demo and Debug message:
1.Device Disconnected:
W (970528) USB_HCDC: The out buffer is too small, the data has been lost 4 I (970528) cdc_basic_demo: Send len=-1: AT

2.Device Connected
.............
 I (2808) USB_HCDC: line 263 HCD_PORT_EVENT_CONNECTION I (2808) USB_HCDC: Resetting Port I (2868) USB_HCDC: Port reset succeed I (2868) USB_HCDC: Getting Port Speed I (2868) USB_HCDC: Port speed = 1 I (2868) USB_HCDC: USB Speed: full-speed I (2868) USB_HCDC: Set Device Addr = 1 I (2878) USB_HCDC: Set Device Addr Done I (2878) USB_HCDC: Set Device Configuration = 1 I (2888) USB_HCDC: Set Device Configuration Done I (2888) USB_HCDC: Set Device Line State: dtr 1, rts 0 I (2898) USB_HCDC: Set Device Line State Done I (2898) cdc_basic_demo: Device Connected! I (2908) USB_HCDC: Creating bulk in pipe I (2908) USB_HCDC: Creating bulk out pipe I (2918) USB_HCDC: Device Connected I (2918) cdc_basic_demo: Send len=4: AT I (3918) cdc_basic_demo: Send len=4: AT
...................

Now,
It's just seems send OK,but the esp32 host have not received datas from device.
how modify the demo to let it work?

leeebo, 2021-12-03:
@mkandy Please make sure your IN endpoint address has been configured correctly, please check [usb_cdc_basic_main.c#L34](https://github.com/espressif/esp-iot-solution/blob/usb/add_usb_solutions/examples/usb/host/usb_cdc_basic/main/usb_cdc_basic_main.c#L34)

If the endpoint address is ok, USB data will be received then buffered automatically, you can check buffered data length through [usbh_cdc_get_buffered_data_len](https://github.com/espressif/esp-iot-solution/blob/usb/add_usb_solutions/components/usb/esp_usbh_cdc/include/esp_usbh_cdc.h#L97) or read it out through [usbh_cdc_read_bytes](https://github.com/espressif/esp-iot-solution/blob/usb/add_usb_solutions/components/usb/esp_usbh_cdc/include/esp_usbh_cdc.h#L107)

mkandy, 2021-12-04:
> @mkandy Please make sure your IN endpoint address has been configured correctly, please check [...]

Hi leeebo, I 'v modified the IN endpoint address in main.c file , now it is working fine , thank you

closed 2021-12-04 by mkandy