espressif/esp-iot-solution #143: how set the sensor demo supports target esp32s2

mkandy, 2021-12-05:
hello,
IDE: VScode + ESP-IDF explorer
example :esp-iot-solution\examples\sensors\sensor_hub_monitor

Issue:
1. The build was successful that tried (not modify) project.

2. If I Clicked the "ESP-IDF set Espressif device target" by VScode with set target to the esp32s2 ,then build was failed.
I have found setup 2 from readme.md file: "使用 `menuconfig` 在 `Board Options->Choose Target Board` 中选择一个目标开发板"
So,What's the difference between the two setting?

build is fail log :
esptool.py v3.2-dev
Merged 1 ELF section
Generated D:/esp32_project/esp-iot-solution/examples/sensors/sensor_hub_monitor/build/bootloader/bootloader.bin
[100/100] cmd.exe /C "cd /D D:\esp32_project\esp-iot-solution\examples\sensors\sensor_hub_monitor\build\bootloader\esp-idf\esptool_py && C:\Users\Administrator\.espressif\python_env\idf4.4_py3.8_env\Scripts\python.exe C:/Users/Administrator/esp/esp-idf/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x1000 D:/esp32_project/esp-iot-solution/examples/sensors/sensor_hub_monitor/build/bootloader/bootloader.bin"
Bootloader binary size 0x50e0 bytes. 0x1f20 bytes (28%) free.
[1028/1225] Building C object esp-idf/dac_audio/CMakeFiles/__idf_dac_audio.dir/dac_audio.c.obj
FAILED: esp-idf/dac_audio/CMakeFiles/__idf_dac_audio.dir/dac_audio.c.obj
C:\Users\Administrator\.espressif\tools\xtensa-esp32s2-elf\esp-2021r2-8.4.0\xtensa-esp32s2-elf\bin\xtensa-esp32s2-elf-gcc.exe  -Iconfig -ID:/esp32_project/esp-iot-solution/components/audio/dac_audio/include -IC:/Users/Administrator/esp/esp-idf/components/newlib/platform_include -IC:/Users/Administrator/esp/esp-idf/components/freertos/include -IC:/Users/Administrator/esp/esp-idf/components/freertos/include/esp_additions/freertos -IC:/Users/Administrator/esp/esp-idf/components/freertos/port/xtensa/include -IC:/Users/Administrator/esp/esp-idf/components/freertos/include/esp_additions -IC:/Users/Administrator/esp/esp-idf/components/esp_hw_support/include -IC:/Users/Administrator/esp/esp-idf/components/esp_hw_support/include/soc -IC:/Users/Administrator/esp/esp-idf/components/esp_hw_support/include/soc/esp32s2 -IC:/Users/Administrator/esp/esp-idf/components/esp_hw_support/port/esp32s2/. -IC:/Users/Administrator/esp/esp-idf/components/esp_hw_support/port/esp32s2/private_include -IC:/Users/Administrator/esp/esp-idf/components/heap/include -IC:/Users/Administrator/esp/esp-idf/components/log/include -IC:/Users/Administrator/esp/esp-idf/components/lwip/include/apps -IC:/Users/Administrator/esp/esp-idf/components/lwip/include/apps/sntp -IC:/Users/Administrator/esp/esp-idf/components/lwip/lwip/src/include -IC:/Users/Administrator/esp/esp-idf/components/lwip/port/esp32/include -IC:/Users/Administrator/esp/esp-idf/components/lwip/port/esp32/include/arch -IC:/Users/Administrator/esp/esp-idf/components/soc/include -IC:/Users/Administrator/esp/esp-idf/components/soc/esp32s2/. -IC:/Users/Administrator/esp/esp-idf/components/soc/esp32s2/include -IC:/Users/Administrator/esp/esp-idf/components/hal/esp32s2/include
-IC:/Users/Administrator/esp/esp-idf/components/hal/include -IC:/Users/Administrator/esp/esp-idf/components/hal/platform_port/include -IC:/Users/Administrator/esp/esp-idf/components/esp_rom/include -IC:/Users/Administrator/esp/esp-idf/components/esp_rom/include/esp32s2 -IC:/Users/Administrator/esp/esp-idf/components/esp_rom/esp32s2 -IC:/Users/Administrator/esp/esp-idf/components/esp_common/include -IC:/Users/Administrator/esp/esp-idf/components/esp_system/include -IC:/Users/Administrator/esp/esp-idf/components/esp_system/port/soc -IC:/Users/Administrator/esp/esp-idf/components/esp_system/port/public_compat -IC:/Users/Administrator/esp/esp-idf/components/xtensa/include -IC:/Users/Administrator/esp/esp-idf/components/xtensa/esp32s2/include -IC:/Users/Administrator/esp/esp-idf/components/driver/include -IC:/Users/Administrator/esp/esp-idf/components/driver/esp32s2/include -IC:/Users/Administrator/esp/esp-idf/components/esp_pm/include -IC:/Users/Administrator/esp/esp-idf/components/esp_ringbuf/include -IC:/Users/Administrator/esp/esp-idf/components/efuse/include -IC:/Users/Administrator/esp/esp-idf/components/efuse/esp32s2/include -IC:/Users/Administrator/esp/esp-idf/components/vfs/include -IC:/Users/Administrator/esp/esp-idf/components/esp_wifi/include -IC:/Users/Administrator/esp/esp-idf/components/esp_event/include -IC:/Users/Administrator/esp/esp-idf/components/esp_netif/include -IC:/Users/Administrator/esp/esp-idf/components/esp_eth/include -IC:/Users/Administrator/esp/esp-idf/components/tcpip_adapter/include -IC:/Users/Administrator/esp/esp-idf/components/esp_phy/include -IC:/Users/Administrator/esp/esp-idf/components/esp_phy/esp32s2/include -IC:/Users/Administrator/esp/esp-idf/components/esp_ipc/include -IC:/Users/Administrator/esp/esp-idf/components/app_trace/include -IC:/Users/Administrator/esp/esp-idf/components/esp_timer/include -mlongcalls -fdiagnostics-color=always -ffunction-sections -fdata-sections -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -ggdb -Og -fmacro-prefix-map=D:/esp32_project/esp-iot-solution/examples/sensors/sensor_hub_monitor=. -fmacro-prefix-map=C:/Users/Administrator/esp/esp-idf=IDF -fstrict-volatile-bitfields -Wno-error=unused-but-set-variable -fno-jump-tables
-fno-tree-switch-conversion -std=gnu99 -Wno-old-style-declaration -D_GNU_SOURCE -DIDF_VER=\"v4.4-dev-3703-gddc44956bf\" -DESP_PLATFORM -D_POSIX_READER_WRITER_LOCKS -MD -MT esp-idf/dac_audio/CMakeFiles/__idf_dac_audio.dir/dac_audio.c.obj -MF esp-idf\dac_audio\CMakeFiles\__idf_dac_audio.dir\dac_audio.c.obj.d -o esp-idf/dac_audio/CMakeFiles/__idf_dac_audio.dir/dac_audio.c.obj -c D:/esp32_project/esp-iot-solution/components/audio/dac_audio/dac_audio.c
In file included from D:/esp32_project/esp-iot-solution/components/audio/dac_audio/dac_audio.c:19:
D:/esp32_project/esp-iot-solution/components/audio/dac_audio/include/dac_audio.h:32:5: error: unknown type name 'i2s_dac_mode_t'
     i2s_dac_mode_t          dac_mode;               /*!< DAC mode configurations - see i2s_dac_mode_t*/
     ^~~~~~~~~~~~~~
[1032/1225] Building C object esp-idf/pwm_audio/CMakeFiles/__idf_pwm_audio.dir/pwm_audio.c.obj
FAILED: esp-idf/pwm_audio/CMakeFiles/__idf_pwm_audio.dir/pwm_audio.c.obj
C:\Users\Administrator\.espressif\tools\xtensa-esp32s2-elf\esp-2021r2-8.4.0\xtensa-esp32s2-elf\bin\xtensa-esp32s2-elf-gcc.exe  -Iconfig -ID:/esp32_project/esp-iot-solution/components/audio/pwm_audio/include -IC:/Users/Administrator/esp/esp-idf/components/newlib/platform_include -IC:/Users/Administrator/esp/esp-idf/components/freertos/include -IC:/Users/Administrator/esp/esp-idf/components/freertos/include/esp_additions/freertos -IC:/Users/Administrator/esp/esp-idf/components/freertos/port/xtensa/include -IC:/Users/Administrator/esp/esp-idf/components/freertos/include/esp_additions -IC:/Users/Administrator/esp/esp-idf/components/esp_hw_support/include -IC:/Users/Administrator/esp/esp-idf/components/esp_hw_support/include/soc -IC:/Users/Administrator/esp/esp-idf/components/esp_hw_support/include/soc/esp32s2 -IC:/Users/Administrator/esp/esp-idf/components/esp_hw_support/port/esp32s2/. -IC:/Users/Administrator/esp/esp-idf/components/esp_hw_support/port/esp32s2/private_include -IC:/Users/Administrator/esp/esp-idf/components/heap/include -IC:/Users/Administrator/esp/esp-idf/components/log/include -IC:/Users/Administrator/esp/esp-idf/components/lwip/include/apps -IC:/Users/Administrator/esp/esp-idf/components/lwip/include/apps/sntp -IC:/Users/Administrator/esp/esp-idf/components/lwip/lwip/src/include -IC:/Users/Administrator/esp/esp-idf/components/lwip/port/esp32/include -IC:/Users/Administrator/esp/esp-idf/components/lwip/port/esp32/include/arch -IC:/Users/Administrator/esp/esp-idf/components/soc/include -IC:/Users/Administrator/esp/esp-idf/components/soc/esp32s2/. -IC:/Users/Administrator/esp/esp-idf/components/soc/esp32s2/include -IC:/Users/Administrator/esp/esp-idf/components/hal/esp32s2/include
-IC:/Users/Administrator/esp/esp-idf/components/hal/include -IC:/Users/Administrator/esp/esp-idf/components/hal/platform_port/include -IC:/Users/Administrator/esp/esp-idf/components/esp_rom/include -IC:/Users/Administrator/esp/esp-idf/components/esp_rom/include/esp32s2 -IC:/Users/Administrator/esp/esp-idf/components/esp_rom/esp32s2 -IC:/Users/Administrator/esp/esp-idf/components/esp_common/include -IC:/Users/Administrator/esp/esp-idf/components/esp_syst [... cut ...]
D:/esp32_project/esp-iot-solution/components/audio/pwm_audio/pwm_audio.c:775:65: error: 'timg_txconfig_reg_t' {aka 'volatile union <anonymous>'} has no member named 'level_int_en'; did you mean 'tx_level_int_en'?
     handle->timg_dev->hw_timer[handle->config.timer_num].config.level_int_en = 0;
                                                                 ^~~~~~~~~~~~
                                                                 tx_level_int_en
[1033/1225] Linking C static library esp-idf\button\libbutton.a
ninja: build stopped: subcommand failed.

leeebo, 2021-12-08:
Hi, `ESP-IDF set Espressif device target` means you choose a target `Chip` for firmware building, like `esp32` `esp32s2` or others.  While `Board Options->Choose Target Board` means you choose a development board, which includes IO definitions and some initialized code, you can find details [here](https://docs.espressif.com/projects/espressif-esp-iot-solution/zh_CN/latest/basic/boards.html). 

The ideal design should be that you choose a `Choose Target Board` then the `Chip` target be chosen automatically, but we can't achieve that based on the current build system (Chip Target need manually chosen before any other configuration).

mkandy, 2021-12-09:
> Hi, `ESP-IDF set Espressif device target` means you choose a target `Chip` for firmware building, like `esp32` `esp32s2` or others. While `Board Options->Choose Target Board` means you choose a [...]

Hi leeebo,
thank you for your reply. got it~

closed 2021-12-09 by mkandy