espressif/esp-iot-solution #144: How to distinguish multiple button events

1404360768, 2021-12-05:
1, how to detect multiple button events, now the routine just gives a press detection, whether you can distinguish between different events through callback function parameters. 
2, multiple buttons can share a callback function

leeebo, 2021-12-06:
Hi@1404360768,  the callback function has a `void*` param, which is the button handle by default. You can use the handle to distinguish different buttons. https://github.com/espressif/esp-iot-solution/blob/master/components/button/test/button_test.c#L44

1404360768, 2021-12-06:
Thank you, I actually ignored this example, haha

franz-ms-muc, 2022-08-17:
https://github.com/franz-ms-muc/esp-iot-solution/tree/feature/Fix_usr_data

franz-ms-muc, 2022-08-17:
Pull Request: #202

chegewara, 2022-09-09:
Im not sure if this is by design, but i believe this is a bug. 
In this line is missing `btn->event = (uint8_t)BUTTON_PRESS_REPEAT;` 
https://github.com/espressif/esp-iot-solution/blob/master/components/button/iot_button.c#L117
or `iot_button_get_repeat` will never catch `BUTTON_PRESS_REPEAT` when we have the same callback for many events.

Thanks

leeebo, 2023-07-10:
Fixed by https://github.com/espressif/esp-iot-solution/commit/0f767f9cec3bf4090fc70dab981fa2fcfb28401b

closed 2023-07-10 by leeebo