espressif/esp-iot-solution #145: help with HID braille for open source devices for people with disabilities

discapacidad5, 2021-12-06:
help with HID braille for open source devices for people with disabilities

Hello, I am developing a device for people with disabilities. I need to use the HID braille protocol but I am not getting anything with HID for esp32, could you help me with a sample code of how I can implement the HID braille protocol?

here are references to HID braille

https://usb.org/sites/default/files/hutrr78_-_creation_of_a_braille_display_usage_page_0.pdf

https://github.com/nvaccess/nvda/blob/00cd67b737bc5b23a6f5e31cf28110b64ebf2fee/devDocs/hidBrailleTechnicalNotes.md

my repository is the following
https://github.com/brailletouch/Brailletouch

leeebo, 2021-12-08:
@discapacidad5 Hi, thanks for your introduction of the impressive project. While I can't guarantee this special HID device be officially supported. But we surely will help you with any ESP32 related questions. BTW, ESP32-S3 with native USB support may be more suitable for the project if you want USB functions in the future.

discapacidad5, 2021-12-08:
My big problem is that I do not know how the HID works, they have given me several explanations of how it could be done, there are libraries that I would like to leave you to look at. Maybe you understand what I don't understand.

I have created this repository for those who want to help me, there is the documentation that I have compiled
https://github.com/brailletouch/HID-braille-ESP32

leeebo, 2023-07-10:
@discapacidad5 USB Host HID now be supported in [ESP-IDF peripherals/usb/host/hid](https://github.com/espressif/esp-idf/tree/master/examples/peripherals/usb/host/hid). Please try with your project

closed 2023-07-10 by leeebo