espressif/esp-iot-solution #146: usb_camera_wifi_transfer 例子无法获取 USB 例子。

xuhongv, 2021-12-09:
## Environment

- Development Kit:      ESP32-S2 多功能开发板
- IDF version: v4.4-dev-3703-gddc44956bf
- esp-iot-solution: commit 49c940ca7d5faab445b8e08b99eb68feec8fcce6 (HEAD -> usb/add_usb_solutions)
- Build System:         CMake
- Compiler version : xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2) 8.4.0
- Operating System:    Linux
- Power Supply:         USB

## Problem Description

运行 分支 usb/add_usb_solutions 上的 usb_camera_wifi_transfer ，无报错，也无摄像头数据流返回。完整的获取参数见文件：
[摄像头参数.txt](https://github.com/espressif/esp-iot-solution/files/7681478/default.txt)

## Other

已确保USB摄像头是完好的，已通过 lsusb -v 获取：

```
    VideoStreaming Interface Descriptor:
        bLength                            46
        bDescriptorType                    36
        bDescriptorSubtype                  7 (FRAME_MJPEG)
        bFrameIndex                         2
        bmCapabilities                   0x00
          Still image unsupported
        wWidth                            640
        wHeight                           480
        dwMinBitRate                 49152000
        dwMaxBitRate                147456000
        dwMaxVideoFrameBufferSize      614989
        dwDefaultFrameInterval         333333
        bFrameIntervalType                  5
        dwFrameInterval( 0)            333333
        dwFrameInterval( 1)            400000
        dwFrameInterval( 2)            500000
        dwFrameInterval( 3)            666666
        dwFrameInterval( 4)           1000000
```

我的配置如下：

```
  #define DESCRIPTOR_CONFIGURATION_INDEX 1 //fix
  #define DESCRIPTOR_FORMAT_MJPEG_INDEX 1
  #define DESCRIPTOR_STREAM_INTERFACE_INDEX 1

  #define DESCRIPTOR_STREAM_ISOC_ENDPOINT_ADDR 0x83

  #define DEMO_FRAME_WIDTH 640
  #define DEMO_FRAME_HEIGHT 480
  #define DEMO_XFER_BUFFER_SIZE 614989 //Double buffer
  #define DEMO_FRAME_INDEX 2
  #define DEMO_FRAME_INTERVAL 333333

  /* max packet size of esp32-s2 is 1*512, bigger is not supported*/
  #define DEMO_ISOC_EP_MPS 512
  #define DEMO_ISOC_INTERFACE_ALT 1

    uvc_config_t uvc_config = {
        .dev_speed = USB_SPEED_FULL,
        .configuration = DESCRIPTOR_CONFIGURATION_INDEX,
        .format_index = DESCRIPTOR_FORMAT_MJPEG_INDEX,
        .frame_width = DEMO_FRAME_WIDTH,
        .frame_height = DEMO_FRAME_HEIGHT,
        .frame_index = DEMO_FRAME_INDEX,
        .frame_interval = DEMO_FRAME_INTERVAL,
        .interface = DESCRIPTOR_STREAM_INTERFACE_INDEX,
        .interface_alt = DEMO_ISOC_INTERFACE_ALT,
        .isoc_ep_addr = DESCRIPTOR_STREAM_ISOC_ENDPOINT_ADDR,
        .isoc_ep_mps = DEMO_ISOC_EP_MPS,
        .xfer_buffer_size = DEMO_XFER_BUFFER_SIZE,
        .xfer_buffer_a = xfer_buffer_a,
        .xfer_buffer_b = xfer_buffer_b,
        .frame_buffer_size = DEMO_XFER_BUFFER_SIZE,
        .frame_buffer = frame_buffer,
    };
```
串口打印是能获取到的，并且显示成功：

```
...
I (809) camera_httpd: Starting web server on port: '80'
I (809) camera_httpd: Starting stream server on port: '81'
I (819) UVC_STREAM: UVC Streaming Config Succeed
I (869) UVC_STREAM: UVC Streaming Starting
I (899) UVC_STREAM: Port=1 init succeed
I (899) UVC_STREAM: Waitting USB Connection
I (1149) UVC_STREAM: line 353 HCD_PORT_EVENT_CONNECTION
I (1149) UVC_STREAM: Resetting Port
I (1209) UVC_STREAM: Setting Port FIFO
I (1209) UVC_STREAM: Getting Port Speed
I (1209) UVC_STREAM: Port speed = 1
I (1209) UVC_STREAM: USB Speed: full-speed
I (1209) UVC_STREAM: Set Device Addr = 1
I (1219) UVC_STREAM: Set Device Addr Done
I (1219) UVC_STREAM: Set Device Configuration = 1
I (1229) UVC_STREAM: Set Device Configuration Done
I (1229) UVC_STREAM: SET_CUR Probe
I (1239) UVC_STREAM: SET_CUR Probe Done
I (1239) UVC_STREAM: GET_CUR Probe
bmHint: 0001
bFormatIndex: 1
bFrameIndex: 2
dwFrameInterval: 333333
wKeyFrameRate: 0
wPFrameRate: 0
wCompQuality: 0
wCompWindowSize: 0
wDelay: 0
dwMaxVideoFrameSize: 2147606528
dwMaxPayloadTransferSize: 3154247680
dwClockFrequency: 0
bmFramingInfo: 0
bPreferredVersion: 0
bMinVersion: 0
bMaxVersion: 0
bInterfaceNumber: 0
I (1269) UVC_STREAM: GET_CUR Probe Done, actual_num_bytes:64
I (1279) UVC_STREAM: SET_CUR COMMIT
I (1289) UVC_STREAM: SET_CUR COMMIT Done
I (1289) UVC_STREAM: Set Device Interface = 1, Alt = 1
I (1469) UVC_STREAM: Set Device Interface Done
I (1469) UVC_STREAM: Sample processing task started
I (1519) UVC_STREAM: Camera Start Streaming
```

leeebo, 2021-12-09:
Hi @xuhongv , that is a known bug from the current [IDF release/v4.4](https://github.com/espressif/esp-idf/tree/ddc44956bf718540d5451e17e1becf6c7dffe5b8), which cause Wi-Fi and USB can not coexist.

* You can use the commands below to fix:
  ```
  git fetch --all
  git cherry-pick 5918bb73d9559e
  git submodule update -f
  ```

* More directly, you can download the bugfixed version [phy lib](https://github.com/espressif/esp-phy-lib/tree/8c34eec8b8463cbd038c242f0da5e7b68cacd39b/esp32s3), then replace the two lib files in ` esp-idf/components/esp_phy/lib/esp32s3` . then change the config `ESP_PHY_ENABLE_USB` to:
  ```
      config ESP_PHY_ENABLE_USB
          bool "Enable USB when phy init"
          depends on USB_OTG_SUPPORTED || ESP_CONSOLE_USB_SERIAL_JTAG
          default y if IDF_TARGET_ESP32C3 || IDF_TARGET_ESP32S3
          default n
          help
              When using USB Serial/JTAG/OTG/CDC, PHY should enable USB, otherwise USB module
              can not work properly.
  ```

BTW, `release/v4.4` will merge this bugfix sooner or later, then everything will go back to normal

xuhongv, 2021-12-09:
thanks , But I build it by esp32s2 not esp32s3. @leeebo

leeebo, 2021-12-09:
@xuhongv Sorry, ESP32-S2 doesn't have Wi-Fi and USB coexistence bug. So the problem could be from the incorrect camera configurations. 

* Keep in mind if you insert the camera directly to PC, then using `lsusb -v`, the descriptor printed by default is USB High-speed version. 
* To get the camera's Full-speed version descriptor (ESP32-S2 currently supported), you must plug through a Full-speed isolator into the computer.

xuhongv, 2021-12-09:
@leeebo thanks ， I get the USB-Camera device by ```lsusb -v``` ,  this is about it:
[USB-Camera device info ](https://github.com/espressif/esp-iot-solution/files/7681478/default.txt)

can you help me configure it to this project <usb_camera_wifi_transfer> by esp32s2 ?

leeebo, 2021-12-09:
> @leeebo thanks ， I get the USB-Camera device by `lsusb -v` , this is about it: [USB-Camera device info ](https://github.com/espressif/esp-iot-solution/files/7681478/default.txt)
> 
> can you help me configure it to this project <usb_camera_wifi_transfer> by esp32s2 ?

@xuhongv Please plug the camera through a `Full-speed isolator` into the computer to get the `Full-speed version descriptor`, the log seems like High-speed version.

xuhongv, 2021-12-09:
@leeebo Oh，can your recommend a ```Full-speed isolator``` to me ? thank you very much.

leeebo, 2021-12-20:
> @leeebo Oh，can your recommend a `Full-speed isolator` to me ? thank you very much.

Like this one: [image]

calvarez03, 2021-12-30:
Hi @xuhongv, were you able to fix this issue? I'm having the same issue as you.

commit 0473e9b9 (2022-03-07): usb(uvc_stream/usbh_cdc): get fullspeed descriptor by default

resolve #146, resolve #152, resolve #156, resolve #157
closed 2022-12-24 by zhanzhaocheng