espressif/esp-iot-solution #150: usb_cdc_4g_module reports warning: USB_HCDC: The out buffer is too small

achao1104, 2021-12-21:
hello,

I just downloaded the example a couple of days ago, everything fine, but just occasionally reports warning as below:

W (119150) USB_HCDC: The out buffer is too small, the data has been lost 88

leeebo, 2021-12-22:
Which branch and commit are you using? There is a WiFi - USB conflict bug in the current [release/v4.4](a79dc75f0a91f4b558dcefb7ae5096c71c63e2e9), please try with following commands in ESP-IDF folder:

```
git pull
git fetch --all
git cherry-pick 65f3bf4a8a21f
git submodule update -f
```

achao1104, 2021-12-22:
yes, I am using v4.4，besides the commit below，how can I fix it in v4.4?

leeebo, 2023-07-10:
The stable version of iot_usbh_modem has been merged in [master](https://github.com/espressif/esp-iot-solution/tree/master/components/usb/iot_usbh_modem) and has been released as a [component in esp-registry](https://components.espressif.com/components/espressif/iot_usbh_modem), please give a try

closed 2023-07-10 by leeebo