espressif/esp-iot-solution #152: USB Camera + Wi-Fi Transfer Example Not Working

christiant168, 2022-01-04:
USB Camera + Wi-Fi Transfer Example Not Working

## Environment

- Development Kit:    ESP32-S2-Saola-1
- IDF version: v4.4-beta1-275-g214d62b9ad
- esp-iot-solution-usb-add_usb_solution
- Build System:  CMake
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2) 8.4.0
- Operating System:  Windows
- Power Supply:  USB

## Problem Description

I tried to use the USB Camera + Wi-Fi Transfer example from the esp-iot-solution-usb-add_usb_solution library. No errors are reported in the terminal log but the I'm not getting any video streaming. 

[image]

## Other items

### Camera Descriptors

USB Camera Descriptor obtained using lsusb -v in a Linux OS with the camera connected to USB 3.0 port: 
`Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass          239 Miscellaneous Device
  bDeviceSubClass         2 
  bDeviceProtocol         1 Interface Association
  bMaxPacketSize0        64
  idVendor           0xeb1a eMPIA Technology, Inc.
  idProduct          0x299f 
  bcdDevice            6.25
  iManufacturer           1 
  iProduct                2 
  iSerial                 3 
  bNumConfigurations      1

        wWidth                            640
        wHeight                           480
        dwMinBitRate                147456000
        dwMaxBitRate                147456000
        dwMaxVideoFrameBufferSize      614400
        dwDefaultFrameInterval         333333
        bFrameIntervalType                  1
        dwFrameInterval( 0)            333333
      VideoStreaming Interface Descriptor:
        bLength                            30
        bDescriptorType                    36
        bDescriptorSubtype                  7 (FRAME_MJPEG)
        bFrameIndex                         3
        bmCapabilities                   0x00
          Still image unsupported
`

[camera descriptors.txt](https://github.com/espressif/esp-iot-solution/files/7810622/camera.descriptors.txt)

### uvc_config_t parameters:

`  uvc_config_t uvc_config = {
        .dev_speed = USB_SPEED_FULL, 
        .configuration = 1, 
        .format_index = 1, 
        .frame_width = 640, 
        .frame_height = 480, 
        .frame_index = 3, 
        .frame_interval = 333333, 
        .interface = 1, 
        .interface_alt = 1, 
        .isoc_ep_addr = 0x81, 
        .isoc_ep_mps = 512, 
        .xfer_buffer_size = 614400, 
        .xfer_buffer_a = xfer_buffer_a, // the internal transfer buffer
        .xfer_buffer_b = xfer_buffer_b, // the internal transfer buffer
        .frame_buffer_size = 614400, // single frame image size, need to determine according to actual test
        .frame_buffer = frame_buffer, // the image frame buffer
    };`

### Terminal log: 

`ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe6100,len:0x18c0
load:0x4004c000,len:0xcd8
load:0x40050000,len:0x2fa4
entry 0x4004c22c
I (21) boot: ESP-IDF v4.4-beta1-275-g214d62b9ad 2nd stage bootloader
I (21) boot: compile time 15:13:45
I (21) boot: chip revision: 0
I (26) qio_mode: Enabling default flash chip QIO
I (31) boot.esp32s2: SPI Speed      : 80MHz
I (36) boot.esp32s2: SPI Mode       : QIO
I (40) boot.esp32s2: SPI Flash Size : 4MB
I (45) boot: Enabling RNG early entropy source...
I (50) boot: Partition Table:
I (54) boot: ## Label            Usage          Type ST Offset   Length
I (61) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (69) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (76) boot:  2 factory          factory app      00 00 00010000 00100000
I (84) boot: End of partition table
I (88) esp_image: segment 0: paddr=00010020 vaddr=3f000020 size=1881ch (100380) map
I (114) esp_image: segment 1: paddr=00028844 vaddr=3ffc7c10 size=031b4h ( 12724) load
I (116) esp_image: segment 2: paddr=0002ba00 vaddr=40024000 size=04618h ( 17944) load
I (123) esp_image: segment 3: paddr=00030020 vaddr=40080020 size=71ac4h (465604) map
I (208) esp_image: segment 4: paddr=000a1aec vaddr=40028618 size=0f5f0h ( 62960) load
I (222) esp_image: segment 5: paddr=000b10e4 vaddr=50000000 size=00010h (    16) load
I (232) boot: Loaded app from partition at offset 0x10000
I (232) boot: Disabling RNG early entropy source...
I (244) cache: Instruction cache        : size 8KB, 4Ways, cache line size 32Byte
I (244) cache: Data cache               : size 8KB, 4Ways, cache line size 32Byte
I (248) spiram: Found 16MBit SPI RAM device
I (253) spiram: SPI RAM mode: sram 80m
I (257) spiram: PSRAM initialized, cache is in normal (1-core) mode.
I (264) cpu_start: Pro cpu up.
I (487) spiram: SPI SRAM memory test OK
I (494) cpu_start: Pro cpu start user code
I (494) cpu_start: cpu freq: 240000000
I (494) cpu_start: Application information:
I (497) cpu_start: Project name:     uvc_mjpeg_wifi
I (502) cpu_start: App version:      1
I (507) cpu_start: Compile time:     Jan  4 2022 15:13:17
I (513) cpu_start: ELF file SHA256:  8afa80c9ae30cc24...
I (519) cpu_start: ESP-IDF:          v4.4-beta1-275-g214d62b9ad
I (525) heap_init: Initializing. RAM available for dynamic allocation:
I (533) heap_init: At 3FF9E000 len 00002000 (8 KiB): RTCRAM
I (539) heap_init: At 3FFCEBB8 len 0002D448 (181 KiB): DRAM
I (545) heap_init: At 3FFFC000 len 00003A10 (14 KiB): DRAM
I (551) spiram: Adding pool of 2048K of external SPI memory to heap allocator
I (559) spi_flash: detected chip: generic
I (564) spi_flash: flash io: qio
I (568) cpu_start: Starting scheduler on PRO CPU.
I (573) spiram: Reserving pool of 120K of internal memory for DMA/internal allocations
I (593) wifi:wifi driver task: 3ffd6bc4, prio:23, stack:6656, core=0
I (593) system_api: Base MAC address is not set
I (593) system_api: read default base MAC address from EFUSE
I (603) wifi:wifi firmware version: 5c3d3bf
I (603) wifi:wifi certification version: v7.0
I (603) wifi:config NVS flash: enabled
I (603) wifi:config nano formating: disabled
I (613) wifi:Init data frame dynamic rx buffer num: 32
I (613) wifi:Init management frame dynamic rx buffer num: 32
I (623) wifi:Init management short buffer num: 32
I (623) wifi:Init static tx buffer num: 16
I (633) wifi:Init tx cache buffer num: 32
I (633) wifi:Init static rx buffer size: 1600
I (643) wifi:Init static rx buffer num: 16
I (643) wifi:Init dynamic rx buffer num: 32
I (643) wifi_init: rx ba win: 16
I (653) wifi_init: tcpip mbox: 32
I (653) wifi_init: udp mbox: 6
I (653) wifi_init: tcp mbox: 6
I (663) wifi_init: tcp tx win: 5744
I (663) wifi_init: tcp rx win: 5744
I (673) wifi_init: tcp mss: 1440
I (673) wifi_init: WiFi/LWIP prefer SPIRAM
I (683) wifi_init: WiFi IRAM OP enabled
I (683) wifi_init: WiFi RX IRAM OP enabled
I (693) camera wifi: wifi_init_softap finished.SSID:ESP32S2-UVC password:
I (693) phy_init: phy_version 1800,e7ef680,Apr 13 2021,11:45:08
I (783) wifi:mode : softAP (84:f7:03:c2:bb:35)
I (783) wifi:Total power save buffer number: 8
I (783) wifi:Init max length of beacon: 752/752
I (783) wifi:Init max length of beacon: 752/752
I (793) wifi:Set ps type: 0

I (793) camera_httpd: Starting web server on port: '80'
I (803) camera_httpd: Starting stream server on port: '81'
I (803) UVC_STREAM: UVC Streaming Config Succeed
I (863) UVC_STREAM: UVC Streaming Starting
I (893) UVC_STREAM: Port=1 init succeed
I (893) UVC_STREAM: Waitting USB Connection
I (1143) UVC_STREAM: line 353 HCD_PORT_EVENT_CONNECTION
I (1143) UVC_STREAM: Resetting Port
I (1203) UVC_STREAM: Setting Port FIFO
I (1203) UVC_STREAM: Getting Port Speed
I (1203) UVC_STREAM: Port speed = 1
I (1203) UVC_STREAM: USB Speed: full-speed
I (1203) UVC_STREAM: Set Device Addr = 1
I (1213) UVC_STREAM: Set Device Addr Done
I (1213) UVC_STREAM: Set Device Configuration = 1
I (1223) UVC_STREAM: Set Device Configuration Done
I (1223) UVC_STREAM: SET_CUR Probe
I (1233) UVC_STREAM: SET_CUR Probe Done
I (1233) UVC_STREAM: GET_CUR Probe
bmHint: 0001
bFormatIndex: 1
bFrameIndex: 3
dwFrameInterval: 333333
wKeyFrameRate: 0
wPFrameRate: 0
wCompQuality: 0
wCompWindowSize: 0
wDelay: 0
dwMaxVideoFrameSize: 687970
dwMaxPayloadTransferSize: 944
dwClockFrequency: 0
bmFramingInfo: 0
bPreferredVersion: 0
bMinVersion: 0
bMaxVersion: 0
bInterfaceNumber: 0
I (1263) UVC_STREAM: GET_CUR Probe Done, actual_num_bytes:64
I (1273) UVC_STREAM: SET_CUR COMMIT
I (1283) UVC_STREAM: SET_CUR COMMIT Done
I (1283) UVC_STREAM: Set Device Interface = 1, Alt = 1
I (1283) UVC_STREAM: Set Device Interface Done
I (1293) UVC_STREAM: Sample processing task started
I (1343) UVC_STREAM: Camera Start Streaming
I (15543) wifi:new:<1,1>, old:<1,1>, ap:<1,1>, sta:<255,255>, prof:1
I (15543) wifi:station: c8:e2:65:d0:5e:f3 join, AID=1, bgn, 40U
I (15543) camera wifi: station:c8:e2:65:d0:5e:f3 join, AID=1
I (15673) esp_netif_lwip: DHCP server assigned IP to a station, IP is: 192.168.4.2
W (15683) wifi:<ba-add>idx:2 (ifx:1, c8:e2:65:d0:5e:f3), tid:0, ssn:9, winSize:64`

Thank you in advance for your support.

leeebo, 2022-01-07:
@christiant168 Please try `.interface_alt = 1,`, which is the interface of MPS=512

leeebo, 2022-01-07:
Duplicate of #146

commit 0473e9b9 (2022-03-07): usb(uvc_stream/usbh_cdc): get fullspeed descriptor by default

resolve #146, resolve #152, resolve #156, resolve #157
closed 2022-12-24 by zhanzhaocheng