espressif/esp-iot-solution #154: "deinit" usb_cdc_4g_module

achao1104, 2022-01-05:
Hello, 

How to deinit usb_cdc_4g_module? I tried  esp_modem_default_destroy(dce->dte), but it will reports as below:

I (21711) esp-netif_lwip-ppp: Connected
I (21711) esp-netif_lwip-ppp: Name Server1: 211.136.17.107
I (21711) esp-netif_lwip-ppp: Name Server2: 211.136.20.203
I (21716) usb4gutil: IP event! 6
I (21720) usb4gutil: Modem Connected to PPP Server
I (21726) usb4gutil: ppp ip: 10.203.162.141, mask: 255.255.255.255, gw: 10.64.64.64
I (21734) usb4gutil: Main DNS: 211.136.17.107
I (21739) usb4gutil: Backup DNS: 211.136.20.203
W (21745) usb4gutil: PPP netif event = 266
E (26749) esp-modem: esp_modem_default_destroy(100): Cannot destroy dce or netif_adapter
I (26752) util: Free heap all: 2202239
I (26752) util: Free heap internal: 116871
W (26752) USB_HCDC: line 267 HCD_PORT_EVENT_DISCONNECTION
I (26762) USB_HCDC: hcd port state = 7
W (26766) USB_HCDC: Resetting USB Driver
W (26771) USB_HCDC: Waitting for CDC task delete
I (26776) USB_HCDC: pipe state = 0
I (26780) USB_HCDC: pipe state = 0
I (26984) USB_HCDC: CDC task deleted
W (26986) USB_HCDC: Resetting default pipe
I (26986) USB_HCDC: pipe state = 0
I (27186) USB_HCDC: hcd recovering
I (27236) USB_HCDC: Waitting USB Connection
I (27236) USB_HCDC: Port power: ON
W (27784) usb4gutil: usb4g_init
W (27787) USB_HCDC: USB Driver has inited
E (27787) esp-modem-dte: esp_modem_dte_new(382): usb driver install failed

assert failed: modem_board_init usb4gutil.c:444 (dte != NULL)

leeebo, 2022-03-24:
@achao1104 This operation is not supported yet, our next plan is to improve the stability of the application, and finally we will consider dynamic loading and unloading

lijunru-hub, 2025-05-09:
[deinit](https://github.com/espressif/esp-iot-solution/blob/master/components/usb/iot_usbh_modem/include/usbh_modem_board.h#L71)

Already added and tested — no memory leak detected.

closed 2025-05-12 by leeebo