espressif/esp-iot-solution #155: USB_HCDC: line 267 HCD_PORT_EVENT_DISCONNECTION for usb_cdc_4g_module

achao1104, 2022-01-05:
Hello,

It can setup the connection successfully, and I also can upload some data to my cloud sever through usb_cdc_4g_module, but occasionally, it reports "HCD_PORT_EVENT_DISCONNECTION" and then it can not connect server.

W (30035) USB_HCDC: line 267 HCD_PORT_EVENT_DISCONNECTION
I (30039) USB_HCDC: hcd port state = 7
W (30043) USB_HCDC: Resetting USB Driver
W (30048) USB_HCDC: Waitting for CDC task delete
I (30053) USB_HCDC: pipe state = 0
I (30057) USB_HCDC: pipe state = 0
I (30261) USB_HCDC: CDC task deleted
W (30263) USB_HCDC: Resetting default pipe
I (30263) USB_HCDC: pipe state = 0
I (30271) bizutil: bat adc 85
I (30271) bizutil: power is ok 85
I (30277) bizutil: af10 data packed len 37
I (30277) main: sensor data len 37
I (30280) bizutil: sensor server_send sucess
I (30463) USB_HCDC: hcd recovering
I (30513) USB_HCDC: Waitting USB Connection
I (30513) USB_HCDC: Port power: ON

achao1104, 2022-01-07:
hello, could anybody reply? thanks.

leeebo, 2022-01-07:
@achao1104  Please check the power supply of the 4G module, the peak current is very high (~2A) and may shutdown unexpectedly if not reached

achao1104, 2022-01-07:
thanks, our hardware can support peak current 3A.

抄送: "Peter ***@***.******@***.***&gt;;

leeebo, 2022-01-07:
@achao1104 Could you please add your full log?

achao1104, 2022-01-07:
Thanks, here is it.

I (21) boot: ESP-IDF v4.4-beta1-284-gd83021a6e8-dirt 2nd stage bootloader
I (21) boot: compile time 10:23:01
I (22) boot: chip revision: 0
I (26) qio_mode: Enabling default flash chip QIO
I (31) boot.esp32s2: SPI Speed      : 80MHz
I (36) boot.esp32s2: SPI Mode       : QIO
I (41) boot.esp32s2: SPI Flash Size : 4MB
I (45) boot: Enabling RNG early entropy source...
I (51) boot: Partition Table:
I (54) boot: ## Label            Usage          Type ST Offset   Length
I (62) boot:  0 nvs              WiFi data        01 02 00009000 00005000
I (69) boot:  1 otadata          OTA data         01 00 0000e000 00002000
I (77) boot:  2 ota_0            OTA app          00 10 00010000 00100000
I (84) boot:  3 ota_1            OTA app          00 11 00110000 00100000
I (92) boot:  4 storage          Unknown data     01 82 00210000 001e0000
I (99) boot:  5 coredump         Unknown data     01 03 003f0000 00010000
I (107) boot: End of partition table
I (111) boot: No factory image, trying OTA 0
I (116) esp_image: segment 0: paddr=00010020 vaddr=3f000020 size=25798h (153496) map
I (151) esp_image: segment 1: paddr=000357c0 vaddr=3ffca3d0 size=037bch ( 14268) load
I (154) esp_image: segment 2: paddr=00038f84 vaddr=40024000 size=07094h ( 28820) load
I (163) esp_image: segment 3: paddr=00040020 vaddr=40080020 size=928a4h (600228) map
I (268) esp_image: segment 4: paddr=000d28cc vaddr=4002b094 size=0f330h ( 62256) load
I (282) esp_image: segment 5: paddr=000e1c04 vaddr=40070000 size=0002ch (    44) load
I (282) esp_image: segment 6: paddr=000e1c38 vaddr=50000800 size=0001ch (    28) load
I (298) boot: Loaded app from partition at offset 0x10000
I (315) boot: Set actual ota_seq=1 in otadata[0]
I (315) boot: Disabling RNG early entropy source...
I (326) cache: Instruction cache        : size 8KB, 4Ways, cache line size 32Byte
I (326) cache: Data cache               : size 8KB, 4Ways, cache line size 32Byte
I (331) spiram: Found 16MBit SPI RAM device
I (336) spiram: SPI RAM mode: sram 80m
I (341) spiram: PSRAM initialized, cache is in normal (1-core) mode.
I (348) cpu_start: Pro cpu up.
I (570) sf�ff���*��SRAM memory test OK
I (579) cpu_start: Pro cpu start user code
I (579) cpu_start: cpu freq: 240000000
I (579) cpu_start: Application information:
I (582) cpu_start: Project name:     solardevice
I (587) cpu_start: App version:      19
I (592) cpu_start: Compile time:     Jan  7 2022 11:23:30
I (598) cpu_start: ELF file SHA256:  3d2698e7f92e6c49...
I (604) cpu_start: ESP-IDF:          v4.4-beta1-284-gd83021a6e8-dirt
I (611) heap_init: Initializing. RAM available for dynamic allocation:
I (618) heap_init: At 3FF9E02C len 00001FC4 (7 KiB): RTCRAM
I (624) heap_init: At 3FFD21C8 len 00029E38 (167 KiB): D/IRAM
I (631) heap_init: At 3FFFC000 len 00003A10 (14 KiB): DRAM
I (637) spiram: Adding pool of 2048K of external SPI memory to heap allocator
I (645) spi_flash: detected chip: gd
I (649) spi_flash: flash io: qio
W (653) spi_flash: Detected size(8192k) larger than the size in the binary image header(4096k). Using the size in the binary image header.
I (669) cpu_start: Starting scheduler on PRO CPU.
I (672) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (680) gpio: GPIO[13]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (689) util: dog1_task started
I (693) gpio: GPIO[16]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (702) util: restart_task started
I (745) fatfile: Partition size: total: 1799921, used: 753
I (746) gpio: GPIO[14]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (750) i2cutil: i2c init
I (815) util: file /spiffs/config.bin write success 290
I (815) util: config write 290 bytes,include md5
I (816) util: config read success 274 bytes,not include md5
I (985) bizutil: bat adc 85
I (985) bizutil: power is ok 85
I (1349) util: led changed to 100|2000
I (1354) otautil: configured ota_0 type 00 subtype 10 address 10000 size 100000
I (1362) otautil: running ota_0 type 00 subtype 10 address 10000 size 100000
[... 293 lines cut ...]
I (25322) util: sntp update ...
I (26478) util: datetime_sync_cb:  Fri Jan  7 12:30:57 2022
I (26478) util: sntp update success
I (26479) util: datetime_sync:  Fri Jan  7 12:30:57 2022
W (26485) USB_HCDC: line 267 HCD_PORT_EVENT_DISCONNECTION
I (26489) USB_HCDC: hcd port state = 7
W (26494) USB_HCDC: Resetting USB Driver
W (26498) USB_HCDC: Waitting for CDC task delete
I (26503) USB_HCDC: pipe state = 0
I (26508) USB_HCDC: pipe state = 0
I (26712) USB_HCDC: CDC task deleted
W (26713) USB_HCDC: Resetting default pipe
I (26713) USB_HCDC: pipe state = 0
I (26722) bizutil: bat adc 27
I (26722) bizutil: power is ok 27
I (26728) bizutil: af10 data packed len 37
I (26728) main: sensor data len 37
I (26731) bizutil: sensor server_send sucess
I (26914) USB_HCDC: hcd recovering
I (26964) USB_HCDC: Waitting USB Connection
I (26964) USB_HCDC: Port power: ON
E (56735) bizutil: sensor server_send recv2: len -1 errno 11
I (56735) bizutil: sensor biz sock 54 closed

抄送: "Peter ***@***.******@***.***&gt;;

achao1104, 2022-01-07:
It looks ok with your old branch, but has problem with v4.4 release.

抄送: "Peter ***@***.******@***.***&gt;;

leeebo, 2023-02-15:
@achao1104 The new [iot_usbh_modem diver](https://github.com/espressif/esp-iot-solution/tree/master/components/usb/iot_usbh_modem) which support self-recover through daemon task can handle most of the issues, please try with the [example](https://github.com/espressif/esp-iot-solution/tree/master/examples/usb/host/usb_cdc_4g_module).

closed 2023-10-07 by leeebo