espressif/esp-iot-solution #157: uvc摄像头例子通讯报错

Gornerrrrr, 2022-01-10:
开发板：esp32s2-Saola-v1.2
开发环境：win10 + IDFv4.4 + esp iot solution
摄像头：USB2.0 uvc免驱支持MJPEG摄像头

我根据摄像头描述符配置了usb_camera_wifi_transfer例程的参数后烧写测试，发现在通讯之后报错“SWAP NO EOF ”。

运行log
```
I (816) UVC_STREAM: UVC Streaming Config Succeed
I (866) UVC_STREAM: UVC Streaming Starting
I (896) UVC_STREAM: Port=1 init succeed
I (896) UVC_STREAM: Waitting USB Connection
I (1146) UVC_STREAM: line 353 HCD_PORT_EVENT_CONNECTION
I (1146) UVC_STREAM: Resetting Port
I (1206) UVC_STREAM: Setting Port FIFO
I (1206) UVC_STREAM: Getting Port Speed
I (1206) UVC_STREAM: Port speed = 1
I (1206) UVC_STREAM: USB Speed: full-speed
I (1206) UVC_STREAM: Set Device Addr = 1
I (1226) UVC_STREAM: Set Device Addr Done
I (1226) UVC_STREAM: Set Device Configuration = 1
I (1236) UVC_STREAM: Set Device Configuration Done
I (1236) UVC_STREAM: SET_CUR Probe
I (1236) UVC_STREAM: SET_CUR Probe Done
I (1236) UVC_STREAM: GET_CUR Probe
bmHint: 0001
bFormatIndex: 2
bFrameIndex: 5
dwFrameInterval: 10000000
wKeyFrameRate: 0
wPFrameRate: 0
wCompQuality: 0
wCompWindowSize: 0
wDelay: 0
dwMaxVideoFrameSize: 614400
dwMaxPayloadTransferSize: 800
dwClockFrequency: 0
bmFramingInfo: 0
bPreferredVersion: 0
bMinVersion: 0
bMaxVersion: 0
bInterfaceNumber: 0
I (1276) UVC_STREAM: GET_CUR Probe Done, actual_num_bytes:64
I (1276) UVC_STREAM: SET_CUR COMMIT
I (1286) UVC_STREAM: SET_CUR COMMIT Done
I (1286) UVC_STREAM: Set Device Interface = 1, Alt = 4
I (1296) UVC_STREAM: Set Device Interface Done
I (1296) UVC_STREAM: Sample processing task started
I (1356) UVC_STREAM: Camera Start Streaming
W (5346) UVC_STREAM: SWAP NO EOF 9412
W (5846) UVC_STREAM: SWAP NO EOF 9536
W (6336) UVC_STREAM: SWAP NO EOF 9656
```

我的配置：
```
    uvc_config_t uvc_config = {
        .dev_speed = USB_SPEED_FULL, //固定为 USB_SPEED_FULL
        .configuration = 1, //配置描述符编号，一般为 1
        .format_index = 2, // MJPEG 对应的 bFormatIndex, bFormatIndex: 0x02 (2)
        .frame_width = 160, // MJPEG 横向像素，例如 320
        .frame_height = 120, // MJPEG 纵向像素，例如 240
        .frame_index = 5, //MJPEG 320*240 对应的 bFrameIndex, bFrameIndex: 0x05
        .frame_interval = 333333, //可选的帧率 dwFrameInterval，例如 30fps
        .interface = 1, // 可选的视频流接口 bInterfaceNumber，一般为 1
        .interface_alt = 4, // 接口选项 bAlternateSetting, bAlternateSetting: 0x04
        .isoc_ep_addr = 0x82, // 接口选项对应的 bEndpointAddress, bEndpointAddresS: 0x82 (Direction=IN EndpointID=2)
        .isoc_ep_mps = 512, // 接口选项的确定的 MPS， 例如为 512
        .xfer_buffer_size = 20*1024, //单帧图像大小，需要根据实际测试确定，320*240 一般小于 35KB
        .xfer_buffer_a = pointer_buffer_a, // 已经申请的 buffer 指针
        .xfer_buffer_b = pointer_buffer_b, // 已经申请的 buffer 指针
        .frame_buffer_size = 20*1024, //单帧图像大小，需要根据实际测试确定
        .frame_buffer = pointer_frame_buffer, // 已经申请的 buffer 指针
    };
```

摄像头描述符,我是用win10的软件获取的，比较多
[摄像头描述符.txt](https://github.com/espressif/esp-iot-solution/files/7836345/default.txt)

请问可能是什么问题导致的？uvc协议还在学习中不是很明白，是描述符配置的不对还是摄像头本身就不支持？或者说例程需要改动？

leeebo, 2022-01-12:
The root reason for this problem is that the camera has a different configuration descriptor in USB `Fullspeed mode` than in `Highspeed mode`. While the configuration descriptor obtained by the PC is `Highspeed mode`, with this configuration, the camera may not work on the ESP chip.

To get the camera's `Fullspeed descriptor`, there are two ways:
1. Use a USB Fullspeed isolator to obtain from PC
2. Run [ESP-IDF host example](https://github.com/espressif/esp-idf/tree/master/examples/peripherals/usb/host/usb_host_lib) to obtain from ESP chips

`More convenient methods will be added later`

Same problem: #146 #152 #156

kidboygmail, 2022-03-01:
> The root reason for this problem is that the camera has a different configuration descriptor in USB `Fullspeed mode` than in `Highspeed mode`. While the configuration descriptor obtained by the PC [...]

I ran the 2nd way of ESP-IDF host example, in the attached file still shows USB2.0.
Also how could we find the FrameIndex and FormatIndex information? Thank you!
[ESP32_USB_Descriptor.txt](https://github.com/espressif/esp-iot-solution/files/8164494/ESP32_USB_Descriptor.txt)

leeebo, 2022-03-07:
Hi, from the [ESP32_USB_Descriptor.txt](https://github.com/espressif/esp-iot-solution/files/8164494/ESP32_USB_Descriptor.txt), the camera could be supported because there is a `MPS 512` interface

the [latest commits](https://github.com/espressif/esp-iot-solution/commit/3505a91fe2a25054d4e5a14e2363ca953933fae2) of uvc_stream can help print the camera class's descriptors, please try to find the MJPEG `format index` and `frame index`, modify your code based on that.

IDF should also be updated to the latest release/v4.4, [these new commits](https://github.com/espressif/esp-idf/commit/2e8df440c11172175a544608e139512da7997404) must be included

Gornerrrrr, 2022-03-22:
> Hi, from the [ESP32_USB_Descriptor.txt](https://github.com/espressif/esp-iot-solution/files/8164494/ESP32_USB_Descriptor.txt), the camera could be supported because there is a `MPS 512` interface [...]

Thanks your suggest! I updated to latest release/v4.4 and run usb_camera_wifi_transfer example.
Run log:
`
I (2062) UVC_STREAM: UVC Streaming Config Succeed
D (2067) UVC_STREAM: urb alloced
D (2071) UVC_STREAM: urb alloced
D (2074) UVC_STREAM: urb alloced
D (2077) UVC_STREAM: urb alloced
D (2131) intr_alloc: Connected src 48 to int 19 (cpu 0)
I (2131) UVC_STREAM: UVC Streaming Starting
I (2131) uvc_demo: uvc streaming successed!
I (2161) UVC_STREAM: Port=1 init succeed
I (2161) UVC_STREAM: Waitting USB Connection
W (2386) wifi:<ba-add>idx:0 (ifx:0, 34:29:12:43:80:30), tid:5, ssn:0, winSize:64
I (2411) UVC_STREAM: line 358 HCD_PORT_EVENT_CONNECTION
I (2411) UVC_STREAM: Resetting Port
I (2471) UVC_STREAM: Setting Port FIFO
I (2471) UVC_STREAM: Getting Port Speed
I (2471) UVC_STREAM: Port speed = 1
I (2473) UVC_STREAM: USB Speed: full-speed
D (2478) UVC_STREAM: urb alloced
I (2481) UVC_STREAM: Set Device Addr = 1
D (2486) UVC_STREAM: Got expected pipe_handle and event
I (2491) UVC_STREAM: Set Device Addr Done
D (2495) UVC_STREAM: urb free
D (2498) UVC_STREAM: urb alloced
I (2502) UVC_STREAM: get device desc
D (2507) UVC_STREAM: Got expected pipe_handle and event
I (2511) UVC_STREAM: get device desc, actual_num_bytes:26
*** Device descriptor ***
bLength 18
bDescriptorType 1
bcdUSB 2.00
bDeviceClass 0xef
bDeviceSubClass 0x2
bDeviceProtocol 0x1
bMaxPacketSize0 64
idVendor 0x58f
idProduct 0x3841
bcdDevice 1.00
iManufacturer 1
iProduct 2
iSerialNumber 0
bNumConfigurations 1
D (2541) UVC_STREAM: urb free
D (2544) UVC_STREAM: urb alloced
I (2547) UVC_STREAM: get short config desc
D (2557) UVC_STREAM: Got expected pipe_handle and event
I (2557) UVC_STREAM: get config desc, actual_num_bytes:16
I (2563) UVC_STREAM: get full config desc
D (2605) UVC_STREAM: Got expected pipe_handle and event
I (2605) UVC_STREAM: get full config desc, actual_num_bytes:956
*** Configuration descriptor ***
bLength 9
bDescriptorType 2
wTotalLength 948
bNumInterfaces 4
bConfigurationValue 1
iConfiguration 0
bmAttributes 0x80
bMaxPower 500mA
        *** Interface descriptor ***
        bLength 9
        bDescriptorType 4
        bInterfaceNumber 0
        bAlternateSetting 0
        bNumEndpoints 1
        bInterfaceClass 0x0
[... 251 lines cut ...]
D (3270) UVC_STREAM: urb context update
`
my uvc_config:
`
uvc_config_t uvc_config = {
        .dev_speed = USB_SPEED_FULL, 
        .configuration = 1, 
        .format_index = 0x02, //bFormatIndex: 0x02
        .frame_width = 320, 
        .frame_height = 240, 
        .frame_index = 0x06, //bFrameIndex: 0x06
        .frame_interval = 1000000, 
        .interface = 1, 
        .interface_alt = 3, //0x03
        .isoc_ep_addr = 0x82, //bEndpointAddresS: 0x82
        .isoc_ep_mps = 512, 
        .xfer_buffer_size = 50*1024,
        .xfer_buffer_a = pointer_buffer_a,
        .xfer_buffer_b = pointer_buffer_b, 
        .frame_buffer_size = 50*1024,
        .frame_buffer = pointer_frame_buffer, 
    };
`
I noticed mps≠512 in log(dwMaxPayloadTransferSize: 800). When Alt=3 MPS=512 in interface descriptor.Why?
And cant get frame_callback after start.

Gornerrrrr, 2022-03-24:
> 

I'm trying to set it like this
```
#elif CONFIG_SIZE_160_120
#define DESCRIPTOR_CONFIGURATION_INDEX 0x01
#define DESCRIPTOR_FORMAT_MJPEG_INDEX  0x01
#define DEMO_FRAME_INDEX 0x03
#define DEMO_FRAME_WIDTH 160
#define DEMO_FRAME_HEIGHT 120
#define DEMO_FRAME_INTERVAL 500000
#define DEMO_XFER_BUFFER_SIZE (20 * 1024) //Double buffer
#define DESCRIPTOR_STREAM_INTERFACE_INDEX   0x01
#define DESCRIPTOR_STREAM_ISOC_ENDPOINT_ADDR  0x82
#define DEMO_ISOC_INTERFACE_ALT 0x04
#define DEMO_ISOC_EP_MPS 512
#elif CONFIG_SIZE_320_240
#define DESCRIPTOR_CONFIGURATION_INDEX 0x01
#define DESCRIPTOR_FORMAT_MJPEG_INDEX  0x01
#define DEMO_FRAME_INDEX 0x01
#define DEMO_FRAME_WIDTH 320
#define DEMO_FRAME_HEIGHT 240
#define DEMO_FRAME_INTERVAL 400000
#define DEMO_XFER_BUFFER_SIZE (30 * 1024) //Double buffer
#define DESCRIPTOR_STREAM_INTERFACE_INDEX   0x01
#define DESCRIPTOR_STREAM_ISOC_ENDPOINT_ADDR  0x82
#define DEMO_ISOC_INTERFACE_ALT 0x04
#define DEMO_ISOC_EP_MPS 512
```
It's can be working.
But GET_CUR Probe MPS=800 too.
And I tried looping Alt(1-4), MPS=800 too.
Maybe I misunderstood UVC negotiate?

leeebo, 2022-03-24:
@Gornerrrrr I think MPS=512 corresponding DEMO_ISOC_INTERFACE_ALT should be `3` from your log

UVC negotiate process will give a suggested MPS result, your camera seems prefer MPS=800 interface, but it's no problem if you can set interface Alt to 512 without a pipe stall response, after that your camera will prepare 512-Bytes packets for the host to fetch.

[image]

BTW, `And I cant get frame_callback after start stream.` , seems your camera works then.

Gornerrrrr, 2022-03-24:
> @Gornerrrrr I think MPS=512 corresponding DEMO_ISOC_INTERFACE_ALT should be `3` from your log
>
> UVC negotiate process will give a suggested MPS result, your camera seems prefer MPS=800 interface, but [...]

Thanks your help!
I've tried ALT=3 MPS=512, get HCD_PIPE_EVENT_ERROR_OVERFLOW;
ALT=4 MPS=192, get HCD_PIPE_EVENT_ERROR_OVERFLOW too;
ALT=4 MPS=512, it's woking:)
So it's no problem without a pipe stall response,

leeebo, 2022-03-24:
@Gornerrrrr You can dig deeper into the issue, if you set `ALT=4` while the camera transfer packets (please change the log level to print the received packet size) with MPS=512 not 192, It just means there's a mistake in the device's descriptor.

Gornerrrrr, 2022-03-24:
> @Gornerrrrr You can dig deeper into the issue, if you set `ALT=4` while the camera transfer packets (please change the log level to print the received packet size) with MPS=512 not 192, It just means there's a mistake in the device's descriptor.

when ALT=4 log
```
I (4010) UVC_STREAM: Camera Start Streaming
D (4010) UVC_STREAM: urb context update
D (4010) UVC_STREAM: urb context update
D (4011) UVC_STREAM: urb context update
D (4015) UVC_STREAM: urb context update
D (4107) uvc_demo: callback! frame_format = 7, seq = 1, width = 320, height = 240, length = 6716, ptr = 0
I (4107) uvc_demo: send frame = 1
I (5110) uvc_demo: send frame done = 1
D (5111) uvc_demo: callback! frame_format = 7, seq = 21, width = 320, height = 240, length = 6772, ptr = 0
I (5113) uvc_demo: send frame = 21
I (6117) uvc_demo: send frame done = 21
D (6118) uvc_demo: callback! frame_format = 7, seq = 41, width = 320, height = 240, length = 5724, ptr = 0
I (6120) uvc_demo: send frame = 41
I (7124) uvc_demo: send frame done = 41
D (7125) uvc_demo: callback! frame_format = 7, seq = 61, width = 320, height = 240, length = 6772, ptr = 0
I (7127) uvc_demo: send frame = 61
I (8131) uvc_demo: send frame done = 61
D (8133) uvc_demo: callback! frame_format = 7, seq = 81, width = 320, height = 240, length = 6812, ptr = 0
I (8134) uvc_demo: send frame = 81
I (9138) uvc_demo: send frame done = 81
D (9139) uvc_demo: callback! frame_format = 7, seq = 101, width = 320, height = 240, length = 6560, ptr = 0
```
And I tryed get descriptor in windows PC.
```
        ---------------- Interface Descriptor -----------------
bLength                  : 0x09 (9 bytes)
bDescriptorType          : 0x04 (Interface Descriptor)
bInterfaceNumber         : 0x01
bAlternateSetting        : 0x01
bNumEndpoints            : 0x01 (1 Endpoint)
bInterfaceClass          : 0x0E (Video)
bInterfaceSubClass       : 0x02 (Video Streaming)
bInterfaceProtocol       : 0x00
iInterface               : 0x04 (String Descriptor 4)
 Language 0x0409         : "USB 2.0 Camera"
Data (HexDump)           : 09 04 01 01 01 0E 02 00 04                        .........

        ----------------- Endpoint Descriptor -----------------
bLength                  : 0x07 (7 bytes)
bDescriptorType          : 0x05 (Endpoint Descriptor)
bEndpointAddress         : 0x82 (Direction=IN EndpointID=2)
bmAttributes             : 0x05 (TransferType=Isochronous  SyncType=Asynchronous  EndpointType=Data)
wMaxPacketSize           : 0x1400
 Bits 15..13             : 0x00 (reserved, must be zero)
 Bits 12..11             : 0x02 (2 additional transactions per microframe -> allows 683..1024 bytes per packet)
 Bits 10..0              : 0x400 (1024 bytes per packet)
bInterval                : 0x01 (1 ms)
Data (HexDump)           : 07 05 82 05 00 14 01                              .......

        ---------------- Interface Descriptor -----------------
bLength                  : 0x09 (9 bytes)
bDescriptorType          : 0x04 (Interface Descriptor)
bInterfaceNumber         : 0x01
bAlternateSetting        : 0x02
bNumEndpoints            : 0x01 (1 Endpoint)
bInterfaceClass          : 0x0E (Video)
bInterfaceSubClass       : 0x02 (Video Streaming)
bInterfaceProtocol       : 0x00
iInterface               : 0x04 (String Descriptor 4)
 Language 0x0409         : "USB 2.0 Camera"
Data (HexDump)           : 09 04 01 02 01 0E 02 00 04                        .........

        ----------------- Endpoint Descriptor -----------------
bLength                  : 0x07 (7 bytes)
bDescriptorType          : 0x05 (Endpoint Descriptor)
bEndpointAddress         : 0x82 (Direction=IN EndpointID=2)
[... 35 lines cut ...]
bDescriptorType          : 0x04 (Interface Descriptor)
bInterfaceNumber         : 0x01
bAlternateSetting        : 0x04
bNumEndpoints            : 0x01 (1 Endpoint)
bInterfaceClass          : 0x0E (Video)
bInterfaceSubClass       : 0x02 (Video Streaming)
bInterfaceProtocol       : 0x00
iInterface               : 0x04 (String Descriptor 4)
 Language 0x0409         : "USB 2.0 Camera"
Data (HexDump)           : 09 04 01 04 01 0E 02 00 04                        .........

        ----------------- Endpoint Descriptor -----------------
bLength                  : 0x07 (7 bytes)
bDescriptorType          : 0x05 (Endpoint Descriptor)
bEndpointAddress         : 0x82 (Direction=IN EndpointID=2)
bmAttributes             : 0x05 (TransferType=Isochronous  SyncType=Asynchronous  EndpointType=Data)
wMaxPacketSize           : 0x0200
 Bits 15..13             : 0x00 (reserved, must be zero)
 Bits 12..11             : 0x00 (0 additional transactions per microframe -> allows 1..1024 bytes per packet)
 Bits 10..0              : 0x200 (512 bytes per packet)
bInterval                : 0x01 (1 ms)
Data (HexDump)           : 07 05 82 05 00 02 01    
```
ALT=4 MPS=512
It's diffrent, because of HS in PC?

heculess, 2022-05-13:
@Gornerrrrr 你用的摄像头在哪买的？我买了几个类型，idf得到的MJPEG Frame Descriptor都是只有一个，而且只有160*120

Gornerrrrr, 2022-06-23:
> @Gornerrrrr 你用的摄像头在哪买的？我买了几个类型，idf得到的MJPEG Frame Descriptor都是只有一个，而且只有160*120

淘宝搜阿香科技买的

miaohaiwei, 2022-09-03:
> > @Gornerrrrr 你用的摄像头在哪买的？我买了几个类型，idf得到的MJPEG Frame Descriptor都是只有一个，而且只有160*120
> 
> 淘宝搜阿香科技买的

方便问下这个摄像头型号么，我买的在fullspeed模式下仅支持非常小的分辨率。

xxzling, 2022-09-25:
我使用ESP32S3尝试了4款USB摄像头，提示E (1856) UVC_STREAM: Configuration descriptor larger than control transfer max length 都不支持设置接口为 wMaxPacketSize 为 512，最小都是800，请问能给出一个可用的UVC usb摄像头的购买链接或者是型号吗？，感谢

leeebo, 2022-09-26:
Some isochronous mode cameras support multiple interface alternatives, to support different data rates. You can try to run the example to find the `bAlternateSetting` with 512 or lower MPS, If not the camera may not be supported. (Due to the ESP32S2/S3 USB hardware FIFO restriction, the `wMaxPacketSize` can not be larger than 512 Bytes)

But for camera vendors, it should be easy to add a 512 Bytes interface. 

BTW. Bulk mode camera is also supported now (no FIFO restriction, because Full speed bulk using 64 Bytes MPS), please refer https://github.com/espressif/esp-iot-solution/commit/fa1d17e562767a244e0979a7cff955065720e5dc for details.

Gornerrrrr, 2022-09-27:
> > > @Gornerrrrr 你用的摄像头在哪买的？我买了几个类型，idf得到的MJPEG Frame Descriptor都是只有一个，而且只有160*120
> > 
> > 
> > 淘宝搜阿香科技买的
> 
> 方便问下这个摄像头型号么，我买的在fullspeed模式下仅支持非常小的分辨率。

就是分辨率不高的吗,只能说能看到:(

closed 2022-12-24 by zhanzhaocheng

dzz10, 2024-10-14:
我强制改如下参数，不报错了，但是画面只有小部分有：
uvc_dev->vs_ifc->interface_alt = 4;
[image]

commit 0473e9b9 (2022-03-07): usb(uvc_stream/usbh_cdc): get fullspeed descriptor by default

resolve #146, resolve #152, resolve #156, resolve #157