espressif/esp-iot-solution #163: SPI LCD_WRITE_CMD definition calls unassigned interface driver function write_command in scr_interface_driver_t

lcebear, 2022-01-30:
When creating an interface for SPI or I2C through `esp_err_t scr_interface_create(scr_interface_type_t type, void *config, scr_interface_driver_t **out_driver)` , (scr_interface_driver.c) my problem is that interface_drv.write_cmd is assigned but .write_command is not assigned.  Only i2s / SCREEN_IFACE_8080 is assigned a write_command function.

This was a problem in my case when using the st7789.c on esp32-s3-usb-otg, as `LCD_WRITE_CMD(data)`
is defined in interface_drv_def.h like this
```
#define LCD_WRITE_COMMAND(data, length) g_lcd_handle.interface_drv->write_command(g_lcd_handle.interface_drv, (data), (length))
static inline esp_err_t LCD_WRITE_CMD(uint8_t cmd)
{
   return LCD_WRITE_COMMAND((uint8_t*)&cmd, 1);
} 
```

So LCD_WRITE_CMD in st7789.c does not work due to the definition in interface_drv_def.h / missing write_command function for the interface_driver(s)

I temporarily change the define to call `g_lcd_handle.interface_drv->write_cmd(g_lcd_handle.interface_drv, (data))`, for the screen to work in my case for now.

alic-maker, 2022-02-08:
@lcebear You are right. The interface API is updated to be compatible with some screen controllers whose command is more than 1 byte. BTW, Examples of boards are still under development.

lcebear, 2022-02-08:
Ah, I see now that you removed '#define LCD_WRITE_CMD(cmd) ...'  in 0caa2de5567a17507bc9f542d0bf212637cf3efe

Unlucky timing for me x) But I am not using screens anyway, I was just testing it out.

leeebo, 2022-02-08:
HELLO
IM LIBO
I HAVE GOT YOUR LETTER
PLEASE WAIT FOR REPLY

closed 2023-07-10 by leeebo