espressif/esp-iot-solution #164: usb_dongle on ESP32-WROOM/WROVER

johndoesenior, 2022-02-02:
## Environment

- Development Kit:      [ESP32-Wrover-Kit|ESP32-DevKitC|]
- Kit version (for WroverKit/): [v4]
- Module or chip used:  [ESP32-WROOM-32U|ESP32-WROVER|ESP32-WROVER-I|]
- IDF version (run ``git describe --tags`` to find it): 
   v4.4-rc1-109-geb3797dc3f

## Problem Description

Can the example of USB DONGLE work for ESP-WROOM/WROVER?

### Actual Behavior

Works for S2/S3

JimNickerson, 2022-02-02:
The WROVER does not have  hardware USB.
This is an interesting article https://hackaday.com/2021/03/26/usb-comes-to-the-esp32/

johndoesenior, 2022-02-03:
Both WROVER and WROOM has USB on developer kits. is it not enough?
If I follow the guide and create a USB through IO pins, what changes do I need to do to tinyusb or other libraries?

Thanks

leeebo, 2023-07-10:
The ESP32 Module does not support USB, the USB interface on the ESP32-Wrover-Kit is linked to a USB-UART bridge chip, which finally links to ESP32 through UART.

For usb_dongle or any other USB solutions, please use ESP32-S2 or ESP32-S3, which integrated with a real USB interface

closed 2023-07-10 by leeebo