espressif/esp-iot-solution #165: esp32-s3 i2s lcd driver

daniInfi7, 2022-02-07:
Development Kit: ESP32-S3-DevKitM-1
Module or chip used: ESP32-S3-MINI-1U
IDF version : V4.4
Build System: CMake
Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2-patch2) 8.4.0
Operating System: Windows
Power Supply: USB

**Problem Description**
It seems that the new [bus](https://github.com/espressif/esp-iot-solution/tree/master/components/bus)/[display](https://github.com/espressif/esp-iot-solution/tree/master/components/display) library which should also now supports the esp32-s3 over I2S doesn't work. Used is a ILI9488 driver and followed initialization is made (additionally also the RD pin is set high):
```
gpio_set_direction(38, GPIO_MODE_OUTPUT);
gpio_set_level(38, 1);
vTaskDelay(pdMS_TO_TICKS(120));
esp_err_t ret = ESP_OK;
i2s_lcd_config_t i2s_lcd_cfg = {
    .data_width  = 16,
    .pin_data_num = {16, 20, 21, 33, 4, 5,
 6 , 7, 8, 9, 10, 11, 12, 13, 14, 15, },
    .pin_num_cs = 35,
    .pin_num_wr = 37,
    .pin_num_rs = 36,

    .clk_freq = 4000000,
    .i2s_port = I2S_NUM_0,
    .buffer_size = 32000,
    .swap_data = true,
};

scr_interface_driver_t *iface_drv;
scr_interface_create(SCREEN_IFACE_8080, &i2s_lcd_cfg, &iface_drv);

ret = scr_find_driver(SCREEN_CONTROLLER_ILI9488, &g_lcd);
if (ESP_OK != ret) {
    return;
    ESP_LOGE(TAG, "screen find failed");
}
scr_controller_config_t lcd_cfg = {
    .interface_drv = iface_drv,
    .pin_num_rst = 34,
    .pin_num_bckl = -1,
    .rst_active_level = 0,
    .bckl_active_level = 1,
    .offset_hor = 0,
    .offset_ver = 0,
    .width = 320,
    .height = 480,
    .rotate = SCR_DIR_LRTB, /** equal to SCR_DIR_BTLR */
};
ret = g_lcd.init(&lcd_cfg);
```

The data was also captured with a logic analyser and all transmitted command/data is wrong. mostly the values send are just 0x02. The same pinout works with the [LCD driver](https://github.com/espressif/esp-idf/tree/master/components/esp_lcd).

alic-maker, 2022-02-08:
@daniInfi7 Can you share the timing of the logic analyzer? I am only tested it when data_width=8

daniInfi7, 2022-02-08:
@alic-maker  yes for sure. I am using only the first 8 bits to monitor the command/data. As a reference I have also added the output when the lcd driver is used and the command/data is matching to the one I set.

[lcd_logic.zip](https://github.com/espressif/esp-iot-solution/files/8021445/lcd_logic.zip)
[image]
[image]

EeeeBin, 2022-02-24:
@alic-maker Are there plans to update the data_width=16 drivers?  same error occur, 8bit work well, but 16bit not work

alic-maker, 2022-02-24:
Sorry, I have a lot of work now and can't deal with this problem quickly. You can look the [LCD driver](https://github.com/espressif/esp-idf/tree/master/components/esp_lcd) in esp-idf.

daniInfi7, 2022-03-02:
@EeeeBin which driver do you use?

I am using the ILI9488 driver and have now set the corresponding pins to work in the 8 bit mode but the library still doesn't work. Some suggestions?

brewneaux, 2023-01-17:
I believe this is still an issue. I tried to use https://github.com/espressif/esp-iot-solution/blob/master/components/bus/8080_lcd_esp32s3.c to implement the ESP32s3 in CircuitPython and was not able to get the display to work at all. I do not have a scope or logic analyzer though.

closed 2023-07-10 by leeebo