espressif/esp-iot-solution #167: usb_cdc_basic not work in normal USB CDC device

zhichunlee, 2022-02-08:
Hi
I used esp32-s3 board run usb_cdc_basic example with esp-idf-v4.4 release and esp-iot-solution at add usb solutions branch.
The esp usb cdc device(tusb_serial_device example) can work well, but the normal usb cdc device(HL-340 or CP210x) can not work.
The HL-340 error log are as follows:
----------------------------------------------------------------------------------------------
I (53194) USB_HCDC: Waitting USB Connection
W (93394) USB_HCDC: line 267 HCD_PORT_EVENT_DISCONNECTION
I (93394) USB_HCDC: hcd port state = 7
W (93394) cdc_basic_demo: Device Disconnected!
W (93394) USB_HCDC: Resetting USB Driver
I (93404) USB_HCDC: hcd recovering
I (93454) USB_HCDC: Waitting USB Connection
I (93454) USB_HCDC: Port power: ON
I (104494) USB_HCDC: line 263 HCD_PORT_EVENT_CONNECTION
I (104494) USB_HCDC: Resetting Port
I (104554) USB_HCDC: Port reset succeed
I (104554) USB_HCDC: Getting Port Speed
I (104554) USB_HCDC: Port speed = 1
I (104554) USB_HCDC: USB Speed: full-speed
I (104554) USB_HCDC: Set Device Addr = 1
I (104564) USB_HCDC: Set Device Addr Done
I (104564) USB_HCDC: Set Device Configuration = 1
I (104574) USB_HCDC: Set Device Configuration Done
I (104574) USB_HCDC: Set Device Line State: dtr 1, rts 0
W (104584) USB_HCDC: line 615 Pipe: default HCD_PIPE_EVENT_ERROR_STALL
E (104594) USB_HCDC: /home/user/esp/esp-iot-solution/components/usb/esp_usbh_cdc/esp_usbh_cdc.c:741 (_usb_set_device_line_state):urb event error
I (104604) USB_HCDC: pipe flushing: state = 1
E (104804) USB_HCDC: /home/user/esp/esp-iot-solution/components/usb/esp_usbh_cdc/esp_usbh_cdc.c:1069 (_usb_processing_task):Set device line state failed
W (104804) USB_HCDC: Resetting USB Driver
W (104804) USB_HCDC: Resetting default pipe
I (104814) USB_HCDC: pipe state = 0
I (105014) USB_HCDC: hcd recovering
I (105064) USB_HCDC: Waitting USB Connection
-------------------------------------------------------------------------------------------------------------
The lsusb massage are as follows:

Bus 001 Device 033: ID 1a86:7523 QinHeng Electronics HL-340 USB-Serial adapter
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass          255 Vendor Specific Class
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0         8
  idVendor           0x1a86 QinHeng Electronics
  idProduct          0x7523 HL-340 USB-Serial adapter
  bcdDevice            2.54
  iManufacturer           0 
  iProduct                2 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0027
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower               96mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           3
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass      1 
      bInterfaceProtocol      2 
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        **bEndpointAddress     0x82  EP 2 IN**
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        **wMaxPacketSize     0x0020  1x 32 bytes**
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        **bEndpointAddress     0x02  EP 2 OUT**
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        **wMaxPacketSize     0x0020  1x 32 bytes**
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1

I adjusted parameters following above maessage:

 /* bulk endpoint address */
#define EXAMPLE_BULK_IN_EP_ADDR 0x82
#define EXAMPLE_BULK_OUT_EP_ADDR 0x02
/* bulk endpoint max package size */
#define EXAMPLE_BULK_EP_MPS 32
/* bulk endpoint transfer interval */
#define EXAMPLE_BULK_EP_INTERVAL 0

/* choose if use user endpoint descriptors */
#define EXAMPLE_CONFIG_USER_EP_DESC

#ifdef EXAMPLE_CONFIG_USER_EP_DESC
/*
the basic demo skip the standred get descriptors process,
users need to get params from cdc device descriptors from PC side,
eg. run `lsusb -v` in linux, then hardcode the related params below
*/
static usb_ep_desc_t bulk_out_ep_desc = {
    .bLength = sizeof(usb_ep_desc_t),
    .bDescriptorType = USB_B_DESCRIPTOR_TYPE_ENDPOINT,
    .bEndpointAddress = EXAMPLE_BULK_OUT_EP_ADDR,
    .bmAttributes = 2,
    .wMaxPacketSize = EXAMPLE_BULK_EP_MPS,
    .bInterval = EXAMPLE_BULK_EP_INTERVAL,
};

static usb_ep_desc_t bulk_in_ep_desc = {
    .bLength = sizeof(usb_ep_desc_t),
    .bDescriptorType = USB_B_DESCRIPTOR_TYPE_ENDPOINT,
    .bEndpointAddress = EXAMPLE_BULK_IN_EP_ADDR,
    .bmAttributes = 2,
    .wMaxPacketSize = EXAMPLE_BULK_EP_MPS,
    .bInterval = EXAMPLE_BULK_EP_INTERVAL,
};
#endif

Please tell me how to solve it, Thank you!

leeebo, 2022-02-17:
Some USB-serial chips like `CP210x` using vendor class, which has some differences with standard CDC, silabs offer a special driver on Windows, and for other platforms like esp32s2/s3, they also offer a driver development guide for users to implement [AN571: CP210x Virtual COM Port Interface (silabs.com)](https://www.silabs.com/documents/public/application-notes/AN571.pdf)

zhichunlee, 2022-02-21:
Thank you for your help! I will close this comment.

closed 2022-02-21 by zhichunlee