espressif/esp-iot-solution #170: Examples compile but the ESP module crashes

kylix34, 2022-02-21:
Hi!
I have:
 ESP-IDF v4.3.2 on Windows 10
Nodemcu module with ESP32-WROOM-32

With esp-iot-solution master, I wanted to test the wificonfig example. The code correctly compiles and flashes on the ESP32 module but it keeps restarting it. The error I get is this:
`ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:7472
load:0x40078000,len:14912
load:0x40080400,len:3688
0x40080400: _init at ??:?

entry 0x4008067c
I (26) boot: ESP-IDF v4.3.2 2nd stage bootloader
I (26) boot: compile time 10:24:46
I (26) boot: chip revision: 1
I (29) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (36) boot.esp32: SPI Speed      : 40MHz
I (41) boot.esp32: SPI Mode       : DIO
I (45) boot.esp32: SPI Flash Size : 4MB
I (50) boot: Enabling RNG early entropy source...
I (55) boot: Partition Table:
I (59) boot: ## Label            Usage          Type ST Offset   Length
I (66) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (74) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (81) boot:  2 factory          factory app      00 00 00010000 00100000
I (89) boot: End of partition table
I (93) boot_comm: chip revision: 1, min. application chip revision: 0
I (100) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=1f74ch (128844) map
I (155) esp_image: segment 1: paddr=0002f774 vaddr=3ffb0000 size=008a4h (  2212) load
I (156) esp_image: segment 2: paddr=00030020 vaddr=400d0020 size=a2d68h (666984) map
I (402) esp_image: segment 3: paddr=000d2d90 vaddr=3ffb08a4 size=03938h ( 14648) load
I (408) esp_image: segment 4: paddr=000d66d0 vaddr=40080000 size=17d10h ( 97552) load
I (448) esp_image: segment 5: paddr=000ee3e8 vaddr=50000000 size=00010h (    16) load
I (460) boot: Loaded app from partition at offset 0x10000
I (461) boot: Disabling RNG early entropy source...
I (472) cpu_start: Pro cpu up.
I (472) cpu_start: Starting app cpu, entry point is 0x4008148c
0x4008148c: call_start_cpu1 at C:/Espressif/frameworks/esp-idf-v4.3.2/components/esp_system/port/cpu_start.c:150

I (0) cpu_start: App cpu up.
I (486) cpu_start: Pro cpu start user code
I (486) cpu_start: cpu freq: 160000000
I (486) cpu_start: Application information:
I (491) cpu_start: Project name:     lvgl_wificonfig
I (497) cpu_start: App version:      611cca2
I (501) cpu_start: Compile time:     Feb 21 2022 10:24:19
I (508) cpu_start: ELF file SHA256:  7ca63da7af693a0e...
I (514) cpu_start: ESP-IDF:          v4.3.2
I (519) heap_init: Initializing. RAM available for dynamic allocation:
I (526) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (532) heap_init: At 3FFB8560 len 00027AA0 (158 KiB): DRAM
I (538) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (544) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (551) heap_init: At 40097D10 len 000082F0 (32 KiB): IRAM
I (558) spi_flash: detected chip: gd
I (561) spi_flash: flash io: dio
I (567) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (607) spi_bus: SPI2 bus created
I (608) Board: Board Info: ESP32-LCDKit
I (608) Board: Board Init Done ...
I (611) spi_bus: SPI2 bus device added, CS=5 Mode=0 Speed=20000000
Guru Meditation Error: Core  0 panic'ed (InstrFetchProhibited). Exception was unhandled.

[... 83 lines cut ...]
I (609) Board: Board Info: ESP32-LCDKit
I (609) Board: Board Init Done ...
I (612) spi_bus: SPI2 bus device added, CS=5 Mode=0 Speed=20000000
Guru Meditation Error: Core  0 panic'ed (InstrFetchProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0xefef9861  PS      : 0x00060f30  A0      : 0x800ffef1  A1      : 0x3ffbb120
A2      : 0x3ffb4c80  A3      : 0xefef9861  A4      : 0x00000020  A5      : 0x00000001
A6      : 0x4010017c  A7      : 0x400ff7bc  A8      : 0x800ff834  A9      : 0x3ffbb110
0x4010017c: lcd_ili9341_draw_bitmap at C:/Users/Kylix/Desktop/esp-iot-solution/components/display/screen/controller_driver/ili9341/ili9341.c:244

0x400ff7bc: lcd_ili9341_get_info at C:/Users/Kylix/Desktop/esp-iot-solution/components/display/screen/controller_driver/ili9341/ili9341.c:186

A10     : 0x3ffbdaa0  A11     : 0x3ffbb120  A12     : 0x00000001  A13     : 0x0000001f
A14     : 0x00000001  A15     : 0x00000001  SAR     : 0x0000000e  EXCCAUSE: 0x00000014
EXCVADDR: 0xefef9860  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000

Backtrace:0x6fef985e:0x3ffbb120 0x400ffeee:0x3ffbb150 0x400d84d6:0x3ffbb180 0x40172a59:0x3ffbb230 0x4008e141:0x3ffbb250
0x400ffeee: lcd_ili9341_init at C:/Users/Kylix/Desktop/esp-iot-solution/components/display/screen/controller_driver/ili9341/ili9341.c:107

0x400d84d6: app_main at c:\users\kylix\desktop\esp-iot-solution\examples\hmi\lvgl_wificonfig\build/../main/app_main.c:611 (discriminator 2)

0x40172a59: main_task at C:/Espressif/frameworks/esp-idf-v4.3.2/components/freertos/port/port_common.c:133 (discriminator 2)

0x4008e141: vPortTaskWrapper at C:/Espressif/frameworks/esp-idf-v4.3.2/components/freertos/port/xtensa/port.c:168`

dvosully, 2022-02-23:
I had the same problem, the following PR may fix it:

https://github.com/espressif/esp-iot-solution/pull/171

kylix34, 2022-02-23:
It seems I get the samer error:

`rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:7472
load:0x40078000,len:14912
load:0x40080400,len:3688
0x40080400: _init at ??:?

entry 0x4008067c
I (26) boot: ESP-IDF v4.3.2 2nd stage bootloader
I (26) boot: compile time 15:10:53
I (27) boot: chip revision: 1
I (29) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (36) boot.esp32: SPI Speed      : 40MHz
I (41) boot.esp32: SPI Mode       : DIO
I (45) boot.esp32: SPI Flash Size : 4MB
I (50) boot: Enabling RNG early entropy source...
I (55) boot: Partition Table:
I (59) boot: ## Label            Usage          Type ST Offset   Length
I (66) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (74) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (81) boot:  2 factory          factory app      00 00 00010000 00100000
I (89) boot: End of partition table
I (93) boot_comm: chip revision: 1, min. application chip revision: 0
I (100) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=1d494h (119956) map
I (152) esp_image: segment 1: paddr=0002d4bc vaddr=3ffb0000 size=02b5ch ( 11100) load
I (157) esp_image: segment 2: paddr=00030020 vaddr=400d0020 size=998b0h (628912) map
I (385) esp_image: segment 3: paddr=000c98d8 vaddr=3ffb2b5c size=015c0h (  5568) load
I (388) esp_image: segment 4: paddr=000caea0 vaddr=40080000 size=17d10h ( 97552) load
I (431) esp_image: segment 5: paddr=000e2bb8 vaddr=50000000 size=00010h (    16) load
I (443) boot: Loaded app from partition at offset 0x10000
I (443) boot: Disabling RNG early entropy source...
I (455) cpu_start: Pro cpu up.
I (455) cpu_start: Starting app cpu, entry point is 0x4008148c
0x4008148c: call_start_cpu1 at C:/Espressif/frameworks/esp-idf-v4.3.2/components/esp_system/port/cpu_start.c:150

I (0) cpu_start: App cpu up.
I (469) cpu_start: Pro cpu start user code
I (469) cpu_start: cpu freq: 160000000
I (469) cpu_start: Application information:
I (474) cpu_start: Project name:     lvgl_wificonfig
I (479) cpu_start: App version:      611cca2
I (484) cpu_start: Compile time:     Feb 23 2022 15:13:37
I (490) cpu_start: ELF file SHA256:  db0d24c23de27256...
I (496) cpu_start: ESP-IDF:          v4.3.2
I (501) heap_init: Initializing. RAM available for dynamic allocation:
I (508) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (514) heap_init: At 3FFB8390 len 00027C70 (159 KiB): DRAM
I (521) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (527) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (533) heap_init: At 40097D10 len 000082F0 (32 KiB): IRAM
I (541) spi_flash: detected chip: gd
I (544) spi_flash: flash io: dio
I (549) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (636) spi_bus: SPI2 bus created
I (637) Board: Board Info: ESP32-LCDKit
I (637) Board: Board Init Done ...
I (639) spi_bus: SPI2 bus device added, CS=5 Mode=0 Speed=20000000
Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.

Core  0 register dump:
PC      : 0x5e7d9215  PS      : 0x00060f30  A0      : 0x800fbc4d  A1      : 0x3ffbaf50
A2      : 0x3ffb4b14  A3      : 0x5e7d9215  A4      : 0x00000020  A5      : 0x00000001
A6      : 0x400fbed8  A7      : 0x400fb518  A8      : 0x800fb590  A9      : 0x3ffbaf40
0x400fbed8: lcd_ili9341_draw_bitmap at C:/Users/Kylix/Desktop/esp-iot-solution/components/display/screen/controller_driver/ili9341/ili9341.c:244

[... 9 lines cut ...]

If I select the Devkitc-V4 in menuconfig, I get other errors:
`[1163/1168] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/app_main.c.obj
FAILED: esp-idf/main/CMakeFiles/__idf_main.dir/app_main.c.obj
ccache C:\Espressif\tools\xtensa-esp32-elf\esp-2021r2-8.4.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-gcc.exe -DHAVE_CONFIG_H -DLV_CONF_INCLUDE_SIMPLE=1 -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DUNITY_INCLUDE_CONFIG_H -DWITH_POSIX -Iconfig -I../main -IC:/Espressif/frameworks/esp-idf-v4.3.2/components/newlib/platform_include -IC:/Espressif/frameworks/esp-idf-v4.3.2/components/freertos/include -IC:/Espressif/frameworks/esp-idf-v4.3.2/components/freertos/port/xtensa/include -IC:/Espressif/frameworks/esp-idf-v4.3.2/components/esp_hw_support/include -IC:/Espressif/frameworks/esp-idf-v4.3.2/components/esp_hw_support/port/esp32/. -IC:/Espressif/frameworks/esp-idf-v4.3.2/components/heap/include -IC:/Espressif/frameworks/esp-idf-v4.3.2/components/log/include -IC:/Espressif/frameworks/esp-idf-v4.3.2/components/lwip/include/apps -IC:/Espressif/frameworks/esp-idf-v4.3.2/components/lwip/include/apps/sntp -IC:/Espressif/frameworks/esp-idf-v4.3.2/components/lwip/lwip/src/include -IC:/Espressif/frameworks/esp-idf-v4.3.2/components/lwip/port/esp32/include -IC:/Espressif/frameworks/esp-idf-v4.3.2/components/lwip/port/esp32/include/arch -IC:/Espressif/frameworks/esp-idf-v4.3.2/components/soc/include -IC:/Espressif/frameworks/esp-idf-v4.3.2/components/soc/esp32/. -IC:/Espressif/frameworks/esp-idf-v4.3.2/components/soc/esp32/include -IC:/Espressif/frameworks/esp-idf-v4.3.2/components/hal/esp32/include -IC:/Espressif/frameworks/esp-idf-v4.3.2/components/hal/include -IC:/Espressif/frameworks/esp-idf-v4.3.2/components/esp_rom/include -IC:/Espressif/frameworks/esp-idf-v4.3.2/components/esp_rom/esp32 -IC:/Espressif/frameworks/esp-idf-v4.3.2/components/esp_rom/include/esp3 [... cut ...]
../main/app_main.c:591:21: error: 'BOARD_LCD_SPI_CLOCK_FREQ' undeclared (first use in this function); did you mean 'BOARD_IO_SPI2_SCK'?
         .clk_freq = BOARD_LCD_SPI_CLOCK_FREQ,
                     ^~~~~~~~~~~~~~~~~~~~~~~~
                     BOARD_IO_SPI2_SCK
../main/app_main.c:616:27: error: 'BOARD_TOUCH_SPI_CS_PIN' undeclared (first use in this function); did you mean 'BOARD_IO_SPI2_MOSI'?
             .pin_num_cs = BOARD_TOUCH_SPI_CS_PIN,
                           ^~~~~~~~~~~~~~~~~~~~~~
                           BOARD_IO_SPI2_MOSI
ninja: build stopped: subcommand failed.
ninja failed with exit code 1`

kylix34, 2022-02-23:
Thank you! It is working now. I have a display without the touch matrix, but I see the calibration screen now.

closed 2022-02-24 by kylix34