espressif/esp-iot-solution #172: led_indicator 运行错误

flair0626, 2022-03-01:
我在使用led_indicator 实现快速闪烁（1ms）时，系统运行错误。把时间改长则能够正常运行。

const blink_step_t blink_data_transmission[] = {
    {LED_BLINK_HOLD, LED_STATE_ON, 20}, 
    {LED_BLINK_HOLD, LED_STATE_OFF, 1}, 
    {LED_BLINK_STOP, 0, 0}, 
};

    while (1)
    {
        led_indicator_start(led_handle, BLINK_DATA_TRANSMISSION); 
        vTaskDelay(pdMS_TO_TICKS(100));
    }

assert failed: prvProcessReceivedCommands timers.c:883 (( pxTimer->xTimerPeriodInTicks > 0 ))

Backtrace:0x40023542:0x3ffc1fa00x40026831:0x3ffc1fc0 0x4002c0e1:0x3ffc1fe0 0x400290cb:0x3ffc2100 0x4002911f:0x3ffc2140 0x40029235:0x3ffc2170 
0x40023542: panic_abort at C:/Users/flair/esp/esp-idf/components/esp_system/panic.c:402
0x40026831: esp_system_abort at C:/Users/flair/esp/esp-idf/components/esp_system/esp_system.c:121
0x4002c0e1: __assert_func at C:/Users/flair/esp/esp-idf/components/newlib/assert.c:85
0x400290cb: prvProcessReceivedCommands at C:/Users/flair/esp/esp-idf/components/freertos/timers.c:883 (discriminator 1)
0x4002911f: prvTimerTask at C:/Users/flair/esp/esp-idf/components/freertos/timers.c:600 (discriminator 1)
0x40029235: vPortTaskWrapper at C:/Users/flair/esp/esp-idf/components/freertos/port/xtensa/port.c:131

leeebo, 2022-03-04:
请把 FreeRTOS 的 `Tick Rate` 提高到 1000 Hz `Component config → FreeRTOS → Tick Rate` 修改为 1000，这个值默认为 100，所以小于 10ms 的定时都会变成 0

larryli, 2022-03-04:
@leeebo 使用最新版本的 led_indicator 是不是需要复制组件目录到项目文件，然后修改 `components/led_indicator/include/led_indicator.h` 中的 `led_indicator_blink_type_t` 定义和 `components/led_indicator/led_indicator.c` 中的 `led_indicator_blink_lists` 定义？

我看了一下 42a5d75c 以前版本是可以不用修改 led_indicator 组件代码，直接在 `main/main.c` 定义数据就能调用自定义的 blink step。

leeebo, 2022-03-07:
目前的设计是将指示灯闪烁定义在头文件中，这样做的初衷是简化配置，同类/同生态产品一般会定义固定的指示灯闪烁逻辑。

closed 2023-02-08 by zhanzhaocheng