espressif/esp-iot-solution #178: WeMos lcd initial filed

ARC-MX, 2022-04-09:
## Environment

- Development Kit:      [WeMos ]
- 
[image]

- Kit version (for WroverKit/PicoKit/DevKitC): [v1|v2|v3|v4]
- Module or chip used:  [ESP32-WROOM-32]
- IDF version (run ``git describe --tags`` to find it): 
    // v4.3.2
## Problem Description
My LCD monitor should show the relevant image, but it didn't show it successfully, I use a custom board WeMos and want to use the iot driver to display something on the lcd, so I wrote my code with reference to the test of ssd1306 in "lcd_mono_test.c", but found that the initialization of the lcd could not pass, and could not locate, can you Provide relevant interface usage routines
//Detailed problem description goes here.

### Expected Behavior
My LCD monitor should show the relevant image, but it didn't show it successfully
### Actual Behavior
"lcd.init(&lcd_cfg)" failed, 
[image]
The initialization of the LCD display did not succeed
### Steps to repropduce

1. step1
2. ...

// It helps if you attach a picture of your setup/wiring here.

### Code to reproduce this issue

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
#include <stdio.h>

#include "board.h"
#include "esp_spi_flash.h"
#include "esp_system.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "sdkconfig.h" // generated by "make menuconfig"
#include "img_array.h"
#include "screen_driver.h"

static const char *TAG = BOARD_NAME;
#define CHECK(a, str) if (!(a)) {                                     \
    ESP_LOGE(TAG, "%s:%d (%s):%s", __FILE__, __LINE__, __FUNCTION__, str); \
    exit(1);                                                               \
}

void app_main(void) {
    printf("Hello world!\n");
    iot_board_init();

    i2c_bus_handle_t i2c_bus;
    i2c_bus = (i2c_bus_handle_t)iot_board_get_handle(BOARD_I2C0_ID);
    CHECK(i2c_bus != NULL, "i2c_bus0 creat failed");
    scr_driver_t lcd;
    scr_info_t lcd_info;

    scr_interface_i2c_config_t iface_cfg = {
        .i2c_bus = i2c_bus,
        .clk_speed = 100000,
        .slave_addr = 0x3C,
    };

    scr_interface_driver_t *iface_drv;
    CHECK(ESP_OK == scr_interface_create(SCREEN_IFACE_I2C, &iface_cfg, &iface_drv), "scr iface_drv creat failed");

    scr_controller_config_t lcd_cfg = {0};
    lcd_cfg.interface_drv = iface_drv;
    lcd_cfg.pin_num_rst = -1;
    lcd_cfg.pin_num_bckl = -1;
    lcd_cfg.rst_active_level = 0;
    lcd_cfg.bckl_active_level = 1;
    lcd_cfg.width = img_width;
    lcd_cfg.height = img_height;
    lcd_cfg.rotate = SCR_DIR_LRTB;
    CHECK(ESP_OK == scr_find_driver(SCREEN_CONTROLLER_SSD1306, &lcd), "scr_find_driver() failed");

    CHECK(ESP_OK == lcd.init(&lcd_cfg), "lcd.init() failed");

    CHECK(ESP_OK == lcd.get_info(&lcd_info), "lcd.get_info() failed");
    ESP_LOGI(TAG, "Screen name:%s | width:%d | height:%d", lcd_info.name, lcd_info.width, lcd_info.height);
    CHECK(ESP_OK == lcd.draw_bitmap(0, 0, img_width, img_height, (uint16_t *)bmp_image_128_64), "lcd.draw_bitmap() failed");
}

## Debug Logs

ARC-MX, 2022-04-09:
I found this bug because there is no "interface_i2s->interface_drv.write_command" function in "scr_interface_driver.c" for SPI or I2C interfaces.When using LCD_WRITE_CMD, the macro definition of g_lcd_handle.interface_drv->write_command causes an error. Please fix this bug as soon as possible.

TDA-2030, 2022-04-14:
You are right. there is no initialization for pointer of write_command. 
Only the 8080 interface initialized this pointer:
https://github.com/espressif/esp-iot-solution/blob/ff6da114532ef11768fa2516149cf921c538722a/components/display/screen/interface_driver/scr_interface_driver.c#L299