espressif/esp-iot-solution #187: About getting 4G modem signal quality

hqs1, 2022-05-14:
Hello, how to get the signal quality of the 4G modem? I noticed that the AT command is used to obtain the signal quality in the example, but it seems that it has not been used yet.

leeebo, 2022-05-27:
To achive this, the driver need to open another AT interface (most 4G modem has mutilpul interfaces, which can be enabled at the same time), this feature will be merged in the near future.

hqs1, 2022-06-05:
感谢回复，这个得用一个串口复用
我看合宙那边有一个虚拟串口是ppp+AT的。

leeebo, 2023-02-15:
@hqs1  Open dual interface one for PPP and the other for AT, has been supported in the new [iot_usbh_modem diver](https://github.com/espressif/esp-iot-solution/tree/master/components/usb/iot_usbh_modem), please try with the [example](https://github.com/espressif/esp-iot-solution/tree/master/examples/usb/host/usb_cdc_4g_module).