espressif/esp-iot-solution #188: Hard code in usb_cdc_4g_module for gpio reset

njbinbin, 2022-05-22:
In modem_board.c, some MACRO define as:
#define MODEM_POWER_GPIO_INACTIVE_LEVEL     1
#define MODEM_POWER_GPIO_ACTIVE_MS          500
#define MODEM_POWER_GPIO_INACTIVE_MS        8000
#define MODEM_RESET_GPIO_INACTIVE_LEVEL     1
#define MODEM_RESET_GPIO_ACTIVE_MS          200
#define MODEM_RESET_GPIO_INACTIVE_MS        5000

on my board, reset gpio inactive level is 0, so I changed it to 0, but in function "modem_board_force_reset(void)" below, there are hard codes:

......
    ESP_LOGE(TAG, "Force reset system");
    gpio_config_t io_config = {
            .pin_bit_mask = BIT64(MODEM_RESET_GPIO),
            .mode = GPIO_MODE_OUTPUT
    };
    gpio_config(&io_config);

    gpio_set_level(MODEM_RESET_GPIO, 0);
                                                                  ^^^^^^

......

Same codes ocurred in app_main.c, so I must find them all and revert 0 and 1.
I suggest that move the MACRO define code into modem_board.h, and change hard code to:

gpio_set_level(MODEM_RESET_GPIO, MODEM_RESET_GPIO_INACTIVE_LEVEL);

njbinbin, 2022-05-22:
and
gpio_set_level(MODEM_RESET_GPIO, 1 - MODEM_RESET_GPIO_INACTIVE_LEVEL);

leeebo, 2023-02-15:
@njbinbin  Thanks for your report, the bug has been fixed. and new example has been released in master branch https://github.com/espressif/esp-iot-solution/blob/master/components/usb/iot_usbh_modem/src/usbh_modem_board.c#L149

closed 2023-07-10 by leeebo