espressif/esp-iot-solution #192: Trying to enable physical buttons with lvgl and the task watchdog got triggered

armwdev, 2022-06-11:
## Environment

- Development Kit:      ESP32-S2-DevKitM-1
- Kit version: v1
- Module or chip used:  [ESP32-S2FH4]
- IDF version ESP-IDF v4.4.1
- Build System:         [CMake]
- Compiler versionxtensa-esp32-elf-gcc (crosstool-NG esp-2021r2-patch3) 8.4.0
- Operating System:     [Windows]
- Power Supply:         [USB]

## Problem Description

Fighting with this for couple of days. I've compiled the lvgl_wificonfig, disabled the touch screen and used the lv_port_indev_.c from lvgl to enable physical buttons. The screen displays but the grouping and buttons does not work as expected. In the logs it shows:
```
I (745) lvgl_gui: Start to run LVGL
E (5736) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (5736) task_wdt:  - IDLE (CPU 0)
E (5736) task_wdt: Tasks currently running:
E (5736) task_wdt: CPU 0: lv gui
E (5736) task_wdt: Print CPU 0 (current core) backtrace
```
### Expected Behavior

The watchdog should not trigger because I've checked the gui_task() that has the vTaskDelay(pdMS_TO_TICKS(10)); in it's while(1) loop. Grouped lvgl objects should be highlighted in the display so I can browse through components using physical buttons. In this example I've tried to browse through a list of buttons. If I remove the input code, the grouping highlights the list and watchdog does not get triggered but then it's unusable without the input code.

### Code to reproduce this issue

from gui.c (lv_list list2 omitted):
```cpp
        lv_port_indev_init();
	cur_group = lv_group_create();
	lv_indev_set_group(indev_keypad, cur_group);
	lv_group_add_obj(cur_group, list2);
```

from  lv_port_indev.c:
```cpp
void lv_port_indev_init(void)
{
    static lv_indev_drv_t indev_drv;
    /*------------------ * Keypad * -----------------*/
    /*Initialize your keypad or keyboard if you have*/
    //keypad_init();
    /*Register a keypad input device*/
     lv_indev_drv_init(&indev_drv);
     indev_drv.type = LV_INDEV_TYPE_KEYPAD;
     indev_drv.read_cb = keypad_read;
     indev_keypad = lv_indev_drv_register(&indev_drv);//problem
}
```
from lv_hal_indev.c:
```cpp
lv_indev_t * lv_indev_drv_register(lv_indev_drv_t * driver)
{

    if(driver->disp == NULL) driver->disp = lv_disp_get_default();

    if(driver->disp == NULL) {
        LV_LOG_WARN("lv_indev_drv_register: no display registered hence can't attach the indev to "
                    "a display");
        return NULL;
    }

    lv_indev_t * indev = _lv_ll_ins_head(&LV_GC_ROOT(_lv_indev_ll));
    if(!indev) {
        LV_ASSERT_MEM(indev);
        return NULL;
    }

[... 162 lines cut ...]
I (755) lvgl_gui: Start to run LVGL
E (5836) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (5836) task_wdt:  - IDLE (CPU 0)
E (5836) task_wdt: Tasks currently running:
E (5836) task_wdt: CPU 0: lv gui
E (5836) task_wdt: Print CPU 0 (current core) backtrace

Backtrace:0x400B1F92:0x3FFC0F800x4002485E:0x3FFC0FA0 0x4008D283:0x3FFC97F0 0x4008E7CE:0x3FFC9810 0x4009C70D:0x3FFC9840 0x4009C7F8:0x3FFC9860 0x400A71A2:0x3FFC9880 0x4002B775:0x3FFC98C0
0x400b1f92: task_wdt_isr at C:/esp/esp-idf-v4.4.1/components/esp_system/task_wdt.c:183 (discriminator 3)

0x4002485e: _xt_lowint1 at C:/esp/esp-idf-v4.4.1/components/freertos/port/xtensa/xtensa_vectors.S:1111

0x4008d283: indev_proc_reset_query_handler at C:/esp/esp-iot-solution/components/gui/lvgl_gui/lvgl/src/lv_core/lv_indev.c:1096

0x4008e7ce: _lv_indev_read_task at C:/esp/esp-iot-solution/components/gui/lvgl_gui/lvgl/src/lv_core/lv_indev.c:122

0x4009c70d: lv_task_exec at C:/esp/esp-iot-solution/components/gui/lvgl_gui/lvgl/src/lv_misc/lv_task.c:409 (discriminator 1)

0x4009c7f8: lv_task_handler at C:/esp/esp-iot-solution/components/gui/lvgl_gui/lvgl/src/lv_misc/lv_task.c:142

0x400a71a2: gui_task at C:/esp/esp-iot-solution/components/gui/lvgl_gui/lvgl_gui.c:92

0x4002b775: vPortTaskWrapper at C:/esp/esp-idf-v4.4.1/components/freertos/port/xtensa/port.c:131   

```

TDA-2030, 2022-06-13:
Is there an endless loop in the `keypad_read` function? Also, you should pay attention to the return value of this function and the thread safety of LVGL.

armwdev, 2022-06-13:
Thanks for the reply. No, there is no loop in this function. Should I put an endless loop there? How can I check about the thread safety? Again I got this from lvgl porting examples: https://github.com/lvgl/lvgl/blob/master/examples/porting/lv_port_indev_template.c

This is the keypad_read function and I haven't modified this function:
```cpp
/*Will be called by the library to read the mouse*/
static void keypad_read(lv_indev_drv_t * indev_drv, lv_indev_data_t * data)
{
    static uint32_t last_key = 0;
    /*Get the current x and y coordinates*/
    mouse_get_xy(&data->point.x, &data->point.y);
    /*Get whether the a key is pressed and save the pressed key*/
    uint32_t act_key = keypad_get_key();
    if(act_key != 0) {

    data->state = LV_INDEV_STATE_PR;
    /*Translate the keys to LVGL control characters according to your key definitions*/
    switch(act_key) {

    case 1:
    act_key = LV_KEY_NEXT;
    break;
    case 2:
    act_key = LV_KEY_PREV;
    break;
    case 3:
    act_key = LV_KEY_LEFT;
    break;
    case 4:
    act_key = LV_KEY_RIGHT;
    break;
    case 5:
    act_key = LV_KEY_ENTER;
    break;
    }
    last_key = act_key;
    } else {

    data->state = LV_INDEV_STATE_REL;
    }
    data->key = last_key;
}
```

TDA-2030, 2022-06-14:
You should not use the master version of the porting example. The LVGL of esp-iot-solution is older.

armwdev, 2022-06-14:
Yes, you were right the v7 finally worked without an issue. Thanks a lot for your help, I really appreciate this.

armwdev, 2022-06-29:
Hi again, I can now browse through components using physical buttons and grouping also works so thanks for that. 

Now I am fighting on a different issue but receiving a similar "Task watchdog got triggered" error message. 

The issue is that when I press a button which calls "static lv_res_t list_release_action(lv_obj_t * list_btn)" like 20 times which is why I thought the watchdog got triggered. When the watchdog got tiggered it pointed to lv_task_exec(lv_task_t * task).

I've tried a workaround to call this 1 time but then the lv_obj_align() causes the crash. It points to the LV_ASSERT_OBJ(base, LV_OBJX_NAME) in the error logs.

I am not sure if those are LVGL or ESP32 bugs but this a bit discouraging as I am using the stock code. These things used to be simpler in the past for ESP32. But now for a simple menu with physical buttons and switching screens became such a nightmare and crashes with the stock lvgl_wificonfig example almost makes me avoid using ESP32 altogether but I'll do my best so I hope to use ESP32 in my future projects.

This is the log:
```
E (12829) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in 
time:
E (12829) task_wdt:  - IDLE (CPU 0)
E (12829) task_wdt: Tasks currently running:
E (12829) task_wdt: CPU 0: lv gui
E (12829) task_wdt: Print CPU 0 (current core) backtrace

Backtrace:0x400BCAE2:0x3FFC92000x40024AC2:0x3FFC9220 0x400908A6:0x3FFE2910 0x40087D62:0x3FFE2940 0x40088128:0x3FFE2960 0x400873EB:0x3FFE2990 0x4008FF29:0x3FFE29B0 0x4008FFAA:0x3FFE29E0 0x400A7D3F:0x3FFE2A00 0x4008EDDB:0x3FFE2A20 0x4008FA22:0x3FFE2A40 0x4009E625:0x3FFE2A70 0x4009E710:0x3FFE2A90 0x400ABAF4:0x3FFE2AB0 0x4002EEC1:0x3FFE2AD0
0x400bcae2: task_wdt_isr at C:/esp/esp-idf-v4.4.1/components/esp_system/task_wdt.c:183 (discriminator 3)

0x40024ac2: _xt_lowint1 at C:/esp/esp-idf-v4.4.1/components/freertos/port/xtensa/xtensa_vectors.S:1111

0x400908a6: lv_obj_align at C:/esp/esp-iot-solution/components/gui/lvgl_gui/lvgl/src/lv_core/lv_obj.c:940 (discriminator 1)

0x40087d62: littlevgl_wificonfig_init at C:\esp\esp-iot-solution\examples\hmi\lvgl_wificonfig\build/../main/wifi.c:360

0x40088128: wificonfig at C:\esp\esp-iot-solution\examples\hmi\lvgl_wificonfig\build/../main/wifi.c:552

0x400873eb: list_release_action at C:\esp\esp-iot-solution\examples\hmi\lvgl_wificonfig\build/../main/gui.c:126

0x4008ff29: lv_event_send_func at C:/esp/esp-iot-solution/components/gui/lvgl_gui/lvgl/src/lv_core/lv_obj.c:1898 (discriminator 1)

0x4008ffaa: lv_event_send at C:/esp/esp-iot-solution/components/gui/lvgl_gui/lvgl/src/lv_core/lv_obj.c:1809 (discriminator 2)

0x400a7d3f: lv_list_signal at C:/esp/esp-iot-solution/components/gui/lvgl_gui/lvgl/src/lv_widgets/lv_list.c:671

0x4008eddb: indev_keypad_proc at C:/esp/esp-iot-solution/components/gui/lvgl_gui/lvgl/src/lv_core/lv_indev.c:469

0x4008fa22: _lv_indev_read_task at C:/esp/esp-iot-solution/components/gui/lvgl_gui/lvgl/src/lv_core/lv_indev.c:113

0x4009e625: lv_task_exec at C:/esp/esp-iot-solution/components/gui/lvgl_gui/lvgl/src/lv_misc/lv_task.c:409 (discriminator 1)

0x4009e710: lv_task_handler at C:/esp/esp-iot-solution/components/gui/lvgl_gui/lvgl/src/lv_misc/lv_task.c:142

0x400abaf4: gui_task at C:/esp/esp-iot-solution/components/gui/lvgl_gui/lvgl_gui.c:171

0x4002eec1: vPortTaskWrapper at C:/esp/esp-idf-v4.4.1/components/freertos/port/xtensa/port.c:131

```

lv gui (gui_task):
``` cpp
static void gui_task(void *args)
{
    ESP_LOGI(TAG, "Start to run LVGL");
    while (1) {
        vTaskDelay(pdMS_TO_TICKS(10));

        /* Try to take the semaphore, call lvgl related function on success */
        if (pdTRUE == xSemaphoreTake(xGuiSemaphore, portMAX_DELAY)) {
            lv_task_handler();
            lv_tick_inc(5);//added
            xSemaphoreGive(xGuiSemaphore);
        }
    }
}
```

lb_obj_align():
```cpp
void lv_obj_align(lv_obj_t * obj, const lv_obj_t * base, lv_align_t align, lv_coord_t x_ofs, lv_coord_t y_ofs)
{
    LV_ASSERT_OBJ(obj, LV_OBJX_NAME);

    if(base == NULL) base = lv_obj_get_parent(obj);

    LV_ASSERT_OBJ(base, LV_OBJX_NAME);

    obj_align_core(obj, base, align, true, true, x_ofs, y_ofs);

#if LV_USE_OBJ_REALIGN
    /*Save the last align parameters to use them in `lv_obj_realign`*/
    obj->realign.align       = align;
    obj->realign.xofs        = x_ofs;
    obj->realign.yofs        = y_ofs;
    obj->realign.base        = base;
    obj->realign.mid_align = 0;
#endif
}
```

keypad_read():
```cpp
static bool keypad_read(lv_indev_drv_t * indev_drv, lv_indev_data_t * data)
{
    static uint32_t last_key = 0;

    /*Get the current x and y coordinates*/
    mouse_get_xy(&data->point.x, &data->point.y);
    
    /*Get whether the a key is pressed and save the pressed key*/
    uint32_t act_key = KEY_Scan(0);

    //data->state = LV_INDEV_STATE_REL;

    if(act_key != 0) {
        data->state = LV_INDEV_STATE_PR;

        /*Translate the keys to LVGL control characters according to your key definitions*/
        switch(act_key) {
        case 1:
            act_key = LV_KEY_UP;
            break;
        case 2:
            act_key = LV_KEY_DOWN;
            break;
        case 3:
            act_key = LV_KEY_LEFT;
            break;
        case 4:
            act_key = LV_KEY_RIGHT;
            break;
        case 5:
            act_key = LV_KEY_ENTER;
            break;
        }

        last_key = act_key;
    } else {
        data->state = LV_INDEV_STATE_REL;
    }

    data->key = last_key;

    vTaskDelay(pdMS_TO_TICKS(20));
    lv_task_handler();
    lv_tick_inc(5);

    /*Return `false` because we are not buffering and no more data to read*/
    return false;
}

```

closed 2023-03-22 by armwdev

Tianbawen, 2024-03-14:
Could you tell me how to solve it?