espressif/esp-iot-solution #196: how usb_cdc_4g_module supports ethernet?

achao1104, 2022-06-29:
Hello,

I am using esp32s2+w5500+4g module, and it now works well as a wifi router, that means, you can connect your mobile to it by wifi and your mobile can browse internet since there is a 4g module.
and now, we want it work as an ethernet router, for example, we can connect my computer to it by ethernet cable and then computer can browse internet through it since there still is a 4g module.

Do you have any sample or clue?

Regards,
Peter

achao1104, 2022-07-01:
Could anyone help? please.

lijunru-hub, 2022-09-02:
not support

leeebo, 2023-02-15:
@achao1104 I think it can be supported through this line [usbh_modem_wifi.c#L93](https://github.com/espressif/esp-iot-solution/blob/master/components/usb/iot_usbh_modem/src/usbh_modem_wifi.c#L93), please modify `_g_esp_netif_soft_ap_ip.ip.addr,` to the ethernet interface's IP address then give it a try

closed 2023-10-07 by leeebo