espressif/esp-iot-solution #219: 请给出一个UVC usb

xxzling, 2022-09-25:


leeebo, 2022-09-26:
@xxzling Due to the ESP32S2/S3 USB hardware FIFO restriction, the `wMaxPacketSize` can not be larger than 512 Bytes. Please contact the camera vendor for details (from the camera side, adding an interface with MPS 512 is easy).

Bulk mode camera is also supported now (no FIFO restriction, because Full speed bulk using 64 Bytes MPS), please refer fa1d17e562767a244e0979a7cff955065720e5dc

xxzling, 2022-09-27:
OK, thank you for your reply
&nbsp;

lixin
***@***.***

&nbsp;

------------------&nbsp;原始邮件&nbsp;------------------
发件人: ***@***.***&gt;; 
发送时间: 2022年9月26日(星期一) 晚上10:05
收件人: ***@***.***&gt;; 
抄送: ***@***.***&gt;; ***@***.***&gt;; 
主题: Re: [espressif/esp-iot-solution] 请给出一个UVC usb摄像头的购买链接或者是型号，感谢 (Issue #219)

 
@xxzling由于 ESP32S2/S3 USB 硬件 FIFO 限制，wMaxPacketSize不能大于 512 字节。详情请联系摄像头供应商（从摄像头方面，添加与 MPS 512 的接口很容易）。
 
现在也支持批量模式相机（没有 FIFO 限制，因为全速批量使用 64 字节 MPS），请参考fa1d17e
 
—
直接回复此邮件，在 GitHub 上查看，或取消订阅。
你收到这个是因为你被提到了。消息 ID：<espressif/esp-iot-solution/issues/219/1258094240 @ github 。com&gt;

xxzling, 2022-09-27:
OK, thank you for your reply

closed 2022-12-24 by zhanzhaocheng