espressif/esp-iot-solution #222: Integration of USB PPPoS + USB AT command integration

abhi-ratnman, 2022-10-06:
Hello, I have tested both the PPPoS over USB and AT commands over USB using the examples provided. Now i wanted to ask that, if i want to do both using the same USB port is it possible? What are the thing i will need to take care while integration of both? Do i need to turn off the data mode of PPPoS before sending AT commands? 

Also are both the examples thread safe?

@leeebo your support will be highly appreciated

ankit-thealchemist, 2022-10-06:
yes mostly is it natively thread safe?

leeebo, 2022-10-06:
@abhi-ratnman  Yes using same USB port with two modes is possible. 

For some modems with only one modem interface(support AT+PPP), you need to using command like `+++` to change to AT mode, then using `ATO` back to PPP mode. 

While some modems support both modem interface and a secondary AT interface, you can using AT and PPP at the same time.

leeebo, 2022-10-06:
Please refer: https://github.com/espressif/esp-iot-solution/tree/usb/add_usb_solutions/examples/usb/host/usb_cdc_4g_module#hardware-requirement

abhi-ratnman, 2022-10-07:
@leeebo we're using sim7672x module. So it is supported by the example

ankit-thealchemist, 2022-10-07:
Is usb cdc host library thread safe? We are most probably sending the commands from different threads.

leeebo, 2022-10-08:
> Is usb cdc host library thread safe? We are most probably sending the commands from different threads.

USB CDC host library is thread-safe, but `dce_common_commands` API is not thread-safe, you need to add an external mutex to `sending->waiting->handling` from different threads

ankit-thealchemist, 2022-10-08:
@leeebo Thanks for quick reply!
we are mostly sending the AT commands over the cdc, which we tested using the example usb cdc host. 
https://github.com/espressif/esp-iot-solution/blob/usb/add_usb_solutions/examples/usb/host/usb_cdc_basic/main/usb_cdc_basic_main.c

ankit-thealchemist, 2022-10-10:
> @abhi-ratnman Yes using same USB port with two modes is possible.
>
> For some modems with only one modem interface(support AT+PPP), you need to using command like `+++` to change to AT mode, then [...]

@leeebo Thanks for the help!!

Could you give the example, how to route the at commands to the secondary AT interface, say endpoints are IN 0x82 and out 0x0C for secondary at interface

abhi-ratnman, 2022-10-13:
hi @leeebo any update?

leeebo, 2023-02-15:
@abhi-ratnman @ankit-thealchemist For the new iot_usbh_modem driver, you can config it through menuconfig https://github.com/espressif/esp-iot-solution/blob/master/components/usb/iot_usbh_modem/Kconfig#L77

closed 2023-11-02 by leeebo