espressif/esp-iot-solution #226: EC600N能连上，无法上网，好像USB驱动自动掉了

Leadrive, 2022-10-25:
I (89685) phy_init: phy_version 503,13653eb,Jun  1 2022,17:47:08
I (89724) wifi:mode : softAP (68:b6:b3:2b:c2:fd)
I (89727) wifi:Total power save buffer number: 8
I (89727) wifi:Init max length of beacon: 752/752
I (89727) wifi:Init max length of beacon: 752/752
I (89732) modem_wifi: Wi-Fi AP started
I (89740) wifi:Total power save buffer number: 8
I (89741) modem_wifi: softap ssid: esp_4g_router password: 12345678
I (89747) modem_wifi: NAT is enabled
W (89751) 4g_main: changed: ap dns addr (auto): 221.179.38.7
I (89758) 4g_main: Task dump

I (89761) 4g_main: Load	Stack left	Name	PRI

I (89766) 4g_main: 1.16 	1824 	cdc-data 	6

I (89771) 4g_main: 2.37 	908 	usbh_proc 	5

I (89776) 4g_main: 0.16 	1480 	main 	1

I (89781) 4g_main: 95.62 	816 	IDLE 	0

I (89785) 4g_main: 98.64 	812 	IDLE 	0

I (89790) 4g_main: 0.02 	1364 	Tmr Svc 	1

I (89795) 4g_main: 0.02 	2284 	tiT 	18

I (89799) 4g_main: 0.29 	864 	usb_data_recv 	8

I (89805) 4g_main: 0.15 	548 	modem_daemon 	8

I (89810) 4g_main: 0.00 	3348 	esp_timer 	22

I (89815) 4g_main: 0.72 	704 	ipc1 	24

I (89820) 4g_main: 0.03 	624 	sys_evt 	20

I (89824) 4g_main: 0.13 	4184 	wifi 	23

I (89829) 4g_main: 0.68 	812 	ipc0 	24

I (89834) 4g_main: Free heap=2207839 Free mini=2031616 bigst=2207651, internal=166791 bigst=163840
W (94498) ESP_USBH: line 148 PORT_EVENT_DISCONNECTION
I (94498) ESP_USBH: hcd port state = 7
I (94499) USB_HCDC: CDC task Reset
W (94501) ESP_USBH: pipe=0x3fcea888 clear failed
W (94507) USB_HCDC: in pipe flush failed
I (94511) ESP_USBH: esp_usbh_pipe_deinit: pipe state = 1
I (94517) ESP_USBH: Pipe deinit succeed, addr: 81
W (94623) ESP_USBH: pipe=0x3fcea93c clear failed
W (94623) USB_HCDC: out pipe flush failed
I (94623) ESP_USBH: esp_usbh_pipe_deinit: pipe state = 1
I (94628) ESP_USBH: Pipe deinit succeed, addr: 0A
I (94633) USB_HCDC: rx0 flush -0 = 0
I (94638) USB_HCDC: tx0 flush -0 = 0
W (94642) ESP_USBH: Resetting USB Driver
I (94642) modem_board: Modem state STAGE_DTE_LOSS, Start
W (94653) esp-modem: Waiting exit the PPP mode gracefully
W (94654) modem_board: Modem event! 3
W (94663) ESP_USBH: Resetting default pipe
W (94768) ESP_USBH: pipe=0x3fcea7d4 clear failed
I (94768) ESP_USBH: esp_usbh_pipe_deinit: pipe state = 1
I (94768) ESP_USBH: Pipe deinit succeed, addr: 00
I (94774) ESP_USBH: hcd recovering
I (94778) ESP_USBH: Port power: ON Succeed
I (94843) 4g_main: Task dump

I (94843) 4g_main: Load	Stack left	Name	PRI

I (94843) 4g_main: 2.74 	1824 	cdc-data 	6

I (94846) 4g_main: 1.67 	808 	main 	1

I (94850) 4g_main: 95.15 	816 	IDLE 	0

I (94855) 4g_main: 96.81 	812 	IDLE 	0

I (94859) 4g_main: 0.01 	1364 	Tmr Svc 	1

I (94864) 4g_main: 1.78 	908 	usbh_proc 	5

I (94869) 4g_main: 0.02 	2284 	tiT 	18

I (94874) 4g_main: 0.24 	548 	modem_daemon 	8

I (94879) 4g_main: 1.43 	848 	usb_data_recv 	8

I (94883) ESP_USBH: Waiting USB Connection
I (94884) 4g_main: 0.00 	624 	sys_evt 	20

I (94894) 4g_main: 0.05 	3140 	esp_timer 	22

I (94899) 4g_main: 0.09 	4184 	wifi 	23

I (94904) 4g_main: 0.00 	812 	ipc0 	24

I (94908) 4g_main: 0.00 	704 	ipc1 	24

I (94913) 4g_main: Free heap=2208495 Free mini=2031616 bigst=2204683, internal=167447 bigst=163840
[0m
e mini=2031616 bigst=2201959, internal=166783 bigst=163840ain: Load	Stack left	Name	PRI

Leadrive, 2022-10-25:
https://item.taobao.com/item.htm?id=649942716265
menuconfig中的EC600N两个型号都试过了，同样的问题。
手机可以连上4g-router，显示无互联网。

Leadrive, 2022-10-25:
中国移动，中国电信都试了，APN也设了。

leeebo, 2022-10-26:
@Leadrive 检查 4G 模块电源灯是否正常，大概率是供电问题，4G 模块峰值电流为 2A，wifi 峰值电流 500mA，请确保适配器满足供电需求。

Leadrive, 2022-10-26:
@leeebo 谢谢回复。
接错12V，把esp32-S3板子烧掉了。得买新的才能试了。过几天再来汇报下结果。

Leadrive, 2022-10-30:
````
I (0) cpu_start: App cpu up.
I (502) spiram: SPI SRAM memory test OK
I (510) cpu_start: Pro cpu start user code
I (510) cpu_start: cpu freq: 240000000
I (510) cpu_start: Application information:
I (513) cpu_start: Project name:     cdc_4g
I (517) cpu_start: App version:      1cd3f3d
I (522) cpu_start: Compile time:     Oct 30 2022 16:39:18
I (528) cpu_start: ELF file SHA256:  cda44b50d6b93c1a...
I (534) cpu_start: ESP-IDF:          v4.4.2-dirty
I (540) heap_init: Initializing. RAM available for dynamic allocation:
I (547) heap_init: At 3FCA0A00 len 0003F600 (253 KiB): D/IRAM
I (553) heap_init: At 3FCE0000 len 0000EE34 (59 KiB): STACK/DRAM
I (560) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (566) heap_init: At 600FE000 len 00002000 (8 KiB): RTCRAM
I (573) spiram: Adding pool of 2048K of external SPI memory to heap allocator
I (581) spi_flash: detected chip: generic
I (585) spi_flash: flash io: qio
W (589) spi_flash: Detected size(8192k) larger than the size in the binary image header(4096k). Using the size in the binary image header.
I (602) sleep: Configure to isolate all GPIO pins in sleep state
I (609) sleep: Enable automatic switching of GPIO sleep configuration
I (616) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (632) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (3654) 4g_main: ====================================
I (3654) 4g_main:      ESP 4G Cat.1 Wi-Fi Router
I (3654) 4g_main: ====================================
I (3660) gpio: GPIO[15]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (3669) gpio: GPIO[17]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (3679) gpio: GPIO[16]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
W (3688) modem_board: Force reset modem
I (3692) gpio: GPIO[13]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
W (3702) modem_board: Resetting modem using io=13, level=0
W (3908) modem_board: Waiting for modem initialize ready
I (8938) ESP_USBH: USB Port=1 init succeed, fifo strategy=0
I (8938) ESP_USBH: usb port start succeed
I (8938) USB_HCDC: CDC task start
I (8938) USB_HCDC: usb driver install succeed
I (8947) USB_HCDC: Waiting CDC Device Connection
I (8938) ESP_USBH: USB Processing Start
I (8956) ESP_USBH: Waiting USB Connection
I (8961) ESP_USBH: Port power: ON Succeed
````
停留在这，不动了。怎么办？

closed 2022-10-31 by Leadrive