espressif/esp-iot-solution #227: USB 4G, 停留在这，不动了 ESP_USBH: Port power: ON Succeed

Leadrive, 2022-10-30:
````
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x2b (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x17b0
load:0x403c9700,len:0xe64
load:0x403cc700,len:0x30e8
entry 0x403c9980
I (25) boot: ESP-IDF v4.4.2-dirty 2nd stage bootloader
I (25) boot: compile time 17:49:18
I (25) boot: chip revision: 0
I (27) qio_mode: Enabling default flash chip QIO
I (32) boot.esp32s3: Boot SPI Speed : 80MHz
I (37) boot.esp32s3: SPI Mode       : QIO
I (42) boot.esp32s3: SPI Flash Size : 4MB
I (46) boot: Enabling RNG early entropy source...
I (52) boot: Partition Table:
I (55) boot: ## Label            Usage          Type ST Offset   Length
I (63) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (70) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (78) boot:  2 factory          factory app      00 00 00010000 00100000
I (85) boot:  3 storage          Unknown data     01 82 00110000 00100000
I (93) boot: End of partition table
I (97) esp_image: segment 0: paddr=00010020 vaddr=3c080020 size=19b2ch (105260) map
I (121) esp_image: segment 1: paddr=00029b54 vaddr=3fc98560 size=041d0h ( 16848) load
I (124) esp_image: segment 2: paddr=0002dd2c vaddr=40374000 size=022ech (  8940) load
I (129) esp_image: segment 3: paddr=00030020 vaddr=42000020 size=78970h (493936) map
I (210) esp_image: segment 4: paddr=000a8998 vaddr=403762ec size=12270h ( 74352) load
I (225) esp_image: segment 5: paddr=000bac10 vaddr=50000000 size=00010h (    16) load
I (233) boot: Loaded app from partition at offset 0x10000
I (233) boot: Disabling RNG early entropy source...
I (245) cpu_start: Pro cpu up.
I (245) cpu_start: Starting app cpu, entry point is 0x403755b0
0x403755b0: call_start_cpu1 at /home/lubuntu/esp/esp-idf/components/esp_system/port/cpu_start.c:160

I (0) cpu_start: App cpu up.
I (259) cpu_start: Pro cpu start user code
I (259) cpu_start: cpu freq: 240000000
I (259) cpu_start: Application information:
I (262) cpu_start: Project name:     cdc_4g
I (267) cpu_start: App version:      1cd3f3d
I (272) cpu_start: Compile time:     Oct 30 2022 17:49:08
I (278) cpu_start: ELF file SHA256:  d0ff3e383b210f06...
I (284) cpu_start: ESP-IDF:          v4.4.2-dirty
I (289) heap_init: Initializing. RAM available for dynamic allocation:
I (296) heap_init: At 3FC9FEF0 len 00040110 (256 KiB): D/IRAM
I (303) heap_init: At 3FCE0000 len 0000EE34 (59 KiB): STACK/DRAM
I (309) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (316) heap_init: At 600FE000 len 00002000 (8 KiB): RTCRAM
I (322) spi_flash: detected chip: generic
I (327) spi_flash: flash io: qio
W (331) spi_flash: Detected size(8192k) larger than the size in the binary image header(4096k). Using the size in the binary image header.
I (344) sleep: Configure to isolate all GPIO pins in sleep state
I (350) sleep: Enable automatic switching of GPIO sleep configuration
I (358) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (3415) 4g_main: ====================================
I (3415) 4g_main:      ESP 4G Cat.1 Wi-Fi Router
I (3415) 4g_main: ====================================
I (3421) gpio: GPIO[15]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (3430) gpio: GPIO[17]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (3440) gpio: GPIO[16]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
W (3449) modem_board: Force reset modem
I (3453) gpio: GPIO[13]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
W (3463) modem_board: Resetting modem using io=13, level=0
W (3669) modem_board: Waiting for modem initialize ready
I (8699) ESP_USBH: USB Port=1 init succeed, fifo strategy=0
I (8699) ESP_USBH: usb port start succeed
I (8699) USB_HCDC: CDC task start
I (8699) USB_HCDC: usb driver install succeed
I (8708) USB_HCDC: Waiting CDC Device Connection
I (8699) ESP_USBH: USB Processing Start
I (8717) ESP_USBH: Waiting USB Connection
I (8722) ESP_USBH: Port power: ON Succeed
````
ESP32S3, EC600N， Air724UG，试了，是同样停留在这。
是哪里设置有问题呢？

Leadrive, 2022-10-30:
运行usb_cdc_basic程序，日志也是类似
````
I (0) cpu_start: App cpu up.
I (186) cpu_start: Pro cpu start user code
I (186) cpu_start: cpu freq: 160000000
I (186) cpu_start: Application information:
I (188) cpu_start: Project name:     usb_cdc_basic
I (194) cpu_start: App version:      1cd3f3d
I (199) cpu_start: Compile time:     Oct 30 2022 18:55:52
I (205) cpu_start: ELF file SHA256:  1976aba53fa3de10...
I (211) cpu_start: ESP-IDF:          v4.4.2-dirty
I (216) heap_init: Initializing. RAM available for dynamic allocation:
I (224) heap_init: At 3FC95640 len 0004A9C0 (298 KiB): D/IRAM
I (230) heap_init: At 3FCE0000 len 0000EE34 (59 KiB): STACK/DRAM
I (237) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (243) heap_init: At 600FE000 len 00002000 (8 KiB): RTCRAM
I (250) spi_flash: detected chip: generic
I (254) spi_flash: flash io: dio
W (258) spi_flash: Detected size(8192k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (271) sleep: Configure to isolate all GPIO pins in sleep state
I (278) sleep: Enable automatic switching of GPIO sleep configuration
I (285) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (305) cdc_basic_demo: using user bulk endpoint descriptor
I (305) cdc_basic_demo: itf 1 using default bulk endpoint descriptor
I (345) ESP_USBH: USB Port=1 init succeed, fifo strategy=0
I (345) USB_HCDC: CDC task start
I (345) ESP_USBH: USB Processing Start
I (345) ESP_USBH: Waiting USB Connection
I (345) ESP_USBH: Port power: ON Succeed
I (355) ESP_USBH: usb port start succeed
I (355) USB_HCDC: usb driver install succeed
I (365) USB_HCDC: Waiting CDC Device Connection
````

Leadrive, 2022-10-30:
拔掉DP线，日志是
````
I (98425) ESP_USBH: line 144 PORT_EVENT_CONNECTION
I (98525) ESP_USBH: Resetting Port
I (98585) ESP_USBH: Port reset succeed
I (98585) ESP_USBH: Getting Port Speed
I (98585) ESP_USBH: USB Speed: low-speed
I (98585) ESP_USBH: Pipe init succeed, addr: 00
I (98585) ESP_USBH: Set Device Addr = 1
W (98595) ESP_USBH: line 179 Pipe: default PIPE_EVENT_ERROR_XFER
E (98595) ESP_USBH: /home/lubuntu/esp/esp-iot-solution/components/usb/esp_usbh/esp_usbh.c:894 (_usbh_set_device_addr):urb event error
E (98715) ESP_USBH: /home/lubuntu/esp/esp-iot-solution/components/usb/esp_usbh/esp_usbh.c:1014 (_usbh_processing_task):Set device address failed
W (98715) ESP_USBH: Resetting USB Driver
W (98715) ESP_USBH: Resetting default pipe
I (98825) ESP_USBH: esp_usbh_pipe_deinit: pipe state = 0
I (98825) ESP_USBH: Pipe deinit succeed, addr: 00
I (98825) ESP_USBH: hcd recovering
I (98825) ESP_USBH: Port power: ON Failed
I (98925) ESP_USBH: Waiting USB Connection
````

Leadrive, 2022-10-30:
拔掉DM线，日志是
````
W (153185) ESP_USBH: line 152 PORT_EVENT_ERROR
I (153185) ESP_USBH: hcd port state = 7
W (153185) cdc_basic_demo: Device Disconnected!
W (153185) ESP_USBH: Resetting USB Driver
I (153185) ESP_USBH: hcd recovering
I (153195) ESP_USBH: Port power: ON Succeed
I (153295) ESP_USBH: Waiting USB Connection
I (153545) ESP_USBH: line 144 PORT_EVENT_CONNECTION
I (153645) ESP_USBH: Resetting Port
I (153705) ESP_USBH: Port reset succeed
I (153705) ESP_USBH: Getting Port Speed
I (153705) ESP_USBH: USB Speed: full-speed
I (153705) ESP_USBH: Pipe init succeed, addr: 00
I (153705) ESP_USBH: Set Device Addr = 1
W (153715) ESP_USBH: line 179 Pipe: default PIPE_EVENT_ERROR_XFER
E (153715) ESP_USBH: /home/lubuntu/esp/esp-iot-solution/components/usb/esp_usbh/esp_usbh.c:894 (_usbh_set_device_addr):urb event error
E (153835) ESP_USBH: /home/lubuntu/esp/esp-iot-solution/components/usb/esp_usbh/esp_usbh.c:1014 (_usbh_processing_task):Set device address failed
W (153835) ESP_USBH: Resetting USB Driver
W (153835) ESP_USBH: Resetting default pipe
I (153945) ESP_USBH: esp_usbh_pipe_deinit: pipe state = 0
I (153945) ESP_USBH: Pipe deinit succeed, addr: 00
I (153945) ESP_USBH: hcd recovering
I (153945) ESP_USBH: Port power: ON Failed
I (154045) ESP_USBH: Waiting USB Connection
````

Leadrive, 2022-10-30:
拔掉GND，不断报下面错误
````
W (50696) ESP_USBH: line 148 PORT_EVENT_DISCONNECTION
I (50696) ESP_USBH: hcd port state = 7
W (50696) cdc_basic_demo: Device Disconnected!
W (50696) ESP_USBH: Resetting USB Driver
I (50696) ESP_USBH: hcd recovering
I (50706) ESP_USBH: Port power: ON Succeed
I (50806) ESP_USBH: Waiting USB Connection
I (51056) ESP_USBH: line 144 PORT_EVENT_CONNECTION
I (51156) ESP_USBH: Resetting Port
E (51156) ESP_USBH: /home/lubuntu/esp/esp-iot-solution/components/usb/esp_usbh/esp_usbh.c:272 (_usbh_port_reset):Port reset failed
E (51156) ESP_USBH: /home/lubuntu/esp/esp-iot-solution/components/usb/esp_usbh/esp_usbh.c:998 (_usbh_processing_task):Port Reset failed
W (51166) ESP_USBH: Resetting USB Driver
I (51176) ESP_USBH: hcd recovering
I (51176) ESP_USBH: Port power: ON Failed
I (51286) ESP_USBH: Waiting USB Connection
I (51756) ESP_USBH: line 144 PORT_EVENT_CONNECTION
I (51856) ESP_USBH: Resetting Port
````

Leadrive, 2022-10-30:
再从电脑上接根USB线，到4G模块上，S3可以打印出4G模块的信息了，
一直在输出配置信息，上不了网，同时UBUNTU显示一个上网卡，状态是Disconnected。
感觉是PC机驱动了4G模块

lijunru-hub, 2022-10-31:
需要通过menuconfig配置一下4g模组的端点号

Leadrive, 2022-10-31:
@lijunru-hub 
谢谢，对USB的协议，不是很清楚。我现在手上有EC600N， AIR724UG，ML302，有没有什么参考设置？
按README接了引脚，还是不行。不知道如何下手了。

Leadrive, 2022-10-31:
用的移远EC600N
https://detail.tmall.com/item.htm?id=649942716265&skuId=4865795463711
测试usb_cdc_basic程序
````
/* bulk endpoint address */
#define EXAMPLE_BULK_IN0_EP_ADDR 0x86
#define EXAMPLE_BULK_OUT0_EP_ADDR 0x0f
#define EXAMPLE_BULK_IN1_EP_ADDR 0x86
#define EXAMPLE_BULK_OUT1_EP_ADDR 0x0f
/* bulk endpoint max package size */
#define EXAMPLE_BULK_EP_MPS 64
/* bulk endpoint transfer interval */
#define EXAMPLE_BULK_EP_INTERVAL 0
````
程序运行后，如果EC600N，USB口不是接到PC机上，程序一直停留在
I (403) USB_HCDC: Waiting CDC Device Connection

此时，如果把EC600N的USB线，接到PC主板后面的USB口，将会显示如下信息
````
I (0) cpu_start: App cpu up.
I (182) cpu_start: Pro cpu start user code
I (182) cpu_start: cpu freq: 160000000
I (183) cpu_start: Application information:
I (185) cpu_start: Project name:     usb_cdc_basic
I (191) cpu_start: App version:      1cd3f3d-dirty
I (196) cpu_start: Compile time:     Oct 31 2022 23:08:34
I (202) cpu_start: ELF file SHA256:  01fab7378450e33f...
I (208) cpu_start: ESP-IDF:          v4.4.2-dirty
I (214) heap_init: Initializing. RAM available for dynamic allocation:
I (221) heap_init: At 3FC95640 len 0004A9C0 (298 KiB): D/IRAM
I (227) heap_init: At 3FCE0000 len 0000EE34 (59 KiB): STACK/DRAM
I (234) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (240) heap_init: At 600FE000 len 00002000 (8 KiB): RTCRAM
I (247) spi_flash: detected chip: generic
I (251) spi_flash: flash io: dio
W (255) spi_flash: Detected size(8192k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (269) sleep: Configure to isolate all GPIO pins in sleep state
I (275) sleep: Enable automatic switching of GPIO sleep configuration
I (282) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (303) cdc_basic_demo: using user bulk endpoint descriptor
I (303) cdc_basic_demo: itf 1 using default bulk endpoint descriptor
I (343) ESP_USBH: USB Port=1 init succeed, fifo strategy=0
I (393) USB_HCDC: CDC task start
I (393) ESP_USBH: USB Processing Start
I (393) ESP_USBH: Waiting USB Connection
I (393) ESP_USBH: Port power: ON Succeed
I (393) ESP_USBH: usb port start succeed
I (403) USB_HCDC: usb driver install succeed
I (403) USB_HCDC: Waiting CDC Device Connection
I (16403) ESP_USBH: line 144 PORT_EVENT_CONNECTION
I (16503) ESP_USBH: Resetting Port
E (16503) ESP_USBH: /home/lubuntu/esp/esp-iot-solution/components/usb/esp_usbh/esp_usbh.c:272 (_usbh_port_reset):Port reset failed
E (16503) ESP_USBH: /home/lubuntu/esp/esp-iot-solution/components/usb/esp_usbh/esp_usbh.c:998 (_usbh_processing_task):Port Reset failed
W (16513) ESP_USBH: Resetting USB Driver
I (16523) ESP_USBH: hcd recovering
I (16523) ESP_USBH: Port power: ON Failed
I (16633) ESP_USBH: Waiting USB Connection
I (25963) ESP_USBH: line 144 PORT_EVENT_CONNECTION
I (26063) ESP_USBH: Resetting Port
E (26063) ESP_USBH: /home/lubuntu/esp/esp-iot-solution/components/usb/esp_usbh/esp_usbh.c:272 (_usbh_port_reset):Port reset failed
E (26063) ESP_USBH: /home/lubuntu/esp/esp-iot-solution/components/usb/esp_usbh/esp_usbh.c:998 (_usbh_processing_task):Port Reset failed
W (26073) ESP_USBH: Resetting USB Driver
I (26083) ESP_USBH: hcd recovering
I (26083) ESP_USBH: Port power: ON Failed
I (26193) ESP_USBH: Waiting USB Connection
I (32463) ESP_USBH: line 144 PORT_EVENT_CONNECTION
I (32563) ESP_USBH: Resetting Port
I (32623) ESP_USBH: Port reset succeed
I (32623) ESP_USBH: Getting Port Speed
[... 206 lines cut ...]
I (38003) cdc_basic_demo: Send itf0 len=4: AT

I (38003) cdc_basic_demo: Send itf1 len=4: AT

I (38033) cdc_basic_demo: Itf 1, Receive len=6: 
OK

I (39003) cdc_basic_demo: Send itf0 len=4: AT

I (39003) cdc_basic_demo: Send itf1 len=4: AT

I (40003) cdc_basic_demo: Send itf0 len=4: AT

I (40003) cdc_basic_demo: Send itf1 len=4: AT

I (40033) cdc_basic_demo: Itf 0, Receive len=6: 
OK

I (41003) cdc_basic_demo: Send itf0 len=4: AT

I (41003) cdc_basic_demo: Send itf1 len=4: AT

I (41033) cdc_basic_demo: Itf 1, Receive len=6: 
OK
````

Leadrive, 2022-10-31:
感觉这些输出，是GPIO20, GPIO19和电脑的USB信息，叠加在一起的效果。
单独接DP，DM，不好使，好像缺少一个激活的信号。

lijunru-hub, 2022-11-01:
看起来像是 移远的模组供电不足，建议usb连接到esp32s3的同时再给移远模组个5v供电。模组供电不足，就会一直等待usb连接。下面有usb描述符的的正常的。

lijunru-hub, 2022-11-01:
idf.py menuconfig 配置一些模组型号，你按  /  ,然后搜 EC600N 选中对应的型号，目前支持了 ML302 和 EC600N

Leadrive, 2022-11-01:
换了块S2板子，可以往下运行了，虽然还不能连网。先关闭了

closed 2022-11-01 by Leadrive

Leadrive, 2022-11-04:
又买了块S3的板子，不用外接5V也能带动。之前的S3板子，应该有问题。害我折腾那么久