espressif/esp-iot-solution #228: esp32s2 4g

zghnx, 2022-11-01:
E (40183) esp-netif_lwip-ppp: pppos_input_tcpip failed with -1
E (40184) esp-netif_lwip-ppp: pppos_input_tcpip failed with -1
E (40189) esp-netif_lwip-ppp: pppos_input_tcpip failed with -1
E (40208) esp-netif_lwip-ppp: pppos_input_tcpip failed with -1
E (40302) esp-netif_lwip-ppp: pppos_input_tcpip failed with -1
E (40328) esp-netif_lwip-ppp: pppos_input_tcpip failed with -1
E (40333) esp-netif_lwip-ppp: pppos_input_tcpip failed with -1
E (41147) esp-netif_lwip-ppp: pppos_input_tcpip failed with -1
E (41488) esp-netif_lwip-ppp: pppos_input_tcpip failed with -1

lijunru-hub, 2022-11-18:
Can you describe it in more detail?

leeebo, 2023-02-15:
@zghnx The issue happens because stations upload too much data, more than lwip can consume. Please refer the Performance optimization from https://github.com/espressif/esp-iot-solution/tree/master/examples/usb/host/usb_cdc_4g_module#debugging-method

closed 2023-10-07 by leeebo