espressif/esp-iot-solution #229: Failed to resolve component 'esp_adc_cal'

jonshouse1, 2022-11-03:
Using esp-iot-solution main and esp-idf-main
Trying to build this:  https://github.com/jonshouse1/ESPVNCC

Not sure how I fix this?

jon@jon-VirtualBoxESP:~/test/esp32_vncc$ idf.py build
Executing action: all (aliases: build)
Running cmake in directory /home/jon/test/esp32_vncc/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DPYTHON=/home/jon/.espressif/python_env/idf5.1_py3.8_env/bin/python -DESP_PLATFORM=1 -DCCACHE_ENABLE=0 /home/jon/test/esp32_vncc"...
-- IDF_TARGET not set, using default target: esp32
-- Found Git: /usr/bin/git (found version "2.25.1") 
-- Component directory /home/jon/esp/esp-iot-solution/components/features does not contain a CMakeLists.txt file. No component will be added
-- Component directory /home/jon/esp/esp-iot-solution/components/framework does not contain a CMakeLists.txt file. No component will be added
-- Component directory /home/jon/esp/esp-iot-solution/components/general does not contain a CMakeLists.txt file. No component will be added
-- Component directory /home/jon/esp/esp-iot-solution/components/hmi does not contain a CMakeLists.txt file. No component will be added
-- Component directory /home/jon/esp/esp-iot-solution/components/i2c_devices does not contain a CMakeLists.txt file. No component will be added
-- Component directory /home/jon/esp/esp-iot-solution/components/i2s_devices does not contain a CMakeLists.txt file. No component will be added
-- Component directory /home/jon/esp/esp-iot-solution/components/motor does not contain a CMakeLists.txt file. No component will be added
-- Component directory /home/jon/esp/esp-iot-solution/components/network does not contain a CMakeLists.txt file. No component will be added
-- Component directory /home/jon/esp/esp-iot-solution/components/platforms does not contain a CMakeLists.txt file. No component will be added
-- Component directory /home/jon/esp/esp-iot-solution/components/spi_devices does not contain a CMakeLists.txt file. No component will be added
-- Component directory /home/jon/esp/esp-iot-solution/components/wifi does not contain a CMakeLists.txt file. No component will be added
-- Component directory /home/jon/esp/esp-iot-solution/components/i2c_devices/others does not contain a CMakeLists.txt file. No component will be added
-- Component directory /home/jon/esp/esp-iot-solution/components/i2c_devices/sensor does not contain a CMakeLists.txt file. No component will be added
-- The C compiler identification is GNU 11.2.0
-- The CXX compiler identification is GNU 11.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/jon/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /home/jon/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /home/jon/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /home/jon/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++
-- Check for working CXX compiler: /home/jon/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32
-- Building empty lvgl component due to configuration
-- Building empty ugfx component due to configuration
CMake Error at /home/jon/esp/esp-idf/tools/cmake/build.cmake:263 (message):
  Failed to resolve component 'esp_adc_cal'.
Call Stack (most recent call first):
  /home/jon/esp/esp-idf/tools/cmake/build.cmake:305 (__build_resolve_and_add_req)
  /home/jon/esp/esp-idf/tools/cmake/build.cmake:592 (__build_expand_requirements)
  /home/jon/esp/esp-idf/tools/cmake/project.cmake:440 (idf_build_process)
  CMakeLists.txt:10 (project)

-- Configuring incomplete, errors occurred!
See also "/home/jon/test/esp32_vncc/build/CMakeFiles/CMakeOutput.log".
HINT: The component esp_adc_cal has been moved to the IDF component manager or has been removed and refactored into some other component.
Please look out for component in 'https://components.espressif.com' and add using 'idf.py add-dependency' command.
Refer to the migration guide for more details.
cmake failed with exit code 1, output of the command is in the /home/jon/test/esp32_vncc/build/log/idf_py_stderr_output_10576 and /home/jon/test/esp32_vncc/build/log/idf_py_stdout_output_10576

lijunru-hub, 2022-11-18:
Please make sure that you have installed idf correctly, and that you have made the export directory and the path to esp-iot-soloution

jonshouse1, 2022-11-18:
>Please make sure that you have installed idf correctly, and that you have made the export directory and the path to esp-iot-soloution

I need this patch to make my project reliable on my hardware (I hope)
https://github.com/espressif/esp-idf/commit/3c8f6dd1cfebaee3860fbc66ff985528ba56b447

Which versions of ESP IDF and ESP-IOT-SOLUTION do I install to make sure it includes this patch?

Thanks.

leeebo, 2023-02-08:
@jonshouse1 @lijunru-hub Hi, this is because the `esp_adc_cal` component is ramed to `esp_adc` from IDF 5.0
* If you want such examples build under IDF 5.0, please modify the required component name to `esp_adc`
* If you want IDF v4.4 & 5 compatibility, please refer https://esp32.com/viewtopic.php?t=29480#p102327