espressif/esp-iot-solution #230: Is there any example to capture a image from USB camera

minux1010, 2022-11-09:
There is example to get the video frame from usb camera on ESP32-S2.
Is there any example to get the image only from usb camera?
For example, I can trigger the read image from wifi and it can send back one image only.

leeebo, 2022-11-10:
@minux1010 Please checkout to usb/add_usb_solutions branch for usb examples, for usb camera please refer https://github.com/espressif/esp-iot-solution/tree/usb/add_usb_solutions/components/usb/uvc_stream

dmitrij9992905, 2022-11-19:
@minux1010 do you mean still images?
There are no examples regarding to the still images, there is stream initiation only.
Due to USB1.1 limitations, we can't get the stream better than 640*480@15fps.
But in some cases, we don't need the stream image but we need high-quality still image with the resolution as high as possible from the camera and what fits into PSRAM buffer. 
I'm struggling with that for weeks, and I'm modifying the uvc_stream library provided by this repo.
According to the UVC specs, I can use method 2: suspend the stream, trigger still, process the image, resume the stream.
But you need to probe and commit both stream and still by format and frame indexes.
For now, I can't get still image better than 640*480 px (for now I'm judging by maxbuffersize got from the camera), but camera is capable to give still images 1280*1024 at frame index corresponding to MJPEG and 1600*1200 at frame index corresponding to uncompressed.
Is it possible to remove the limitation on buffer size?

dmitrij9992905, 2022-11-23:
Hi @leeebo 
have you tried the [usb_host_uvc](https://github.com/espressif/idf-extra-components/tree/master/usb/usb_host_uvc), which is based on libuvc. - yes, I tried, but still triggering makes no sense: stream suspends on some time, than I get striped image with the SAME size as the stream, not the image size agreed no still.
According to the specs, the payload has the bitmap which has the bit that points if the image is a part of still or not, but not considered in this payload in the libuvc. With libuvc I couldn't get the image better than 160*120, if more, it will be striped. As well it got striped after the moment of still triggering.
As well, I tried to modify lib from Espressif. I've incorporated still image commiting, then selected frame size which respects the imagesizepattern is needed for me, for example, 1280*1024, but I couldn't get the image better than 640*480.

As for buffer, as I need the still image with high resolution, I chose the ESP32S3R8, with 8MB PSRAM, which is sufficient for high resolution uncompressed image.
Maxbuffersize to be got from the camera during probe.

j1981, 2022-12-04:
don't waste your time usb_camera_wifi_transfer example thath used uvc_stream doesn't work.After a few transmitted frames it returns W (80191) UVC_STREAM: line 378 HCD_PORT_EVENT_ERROR and the connection to the camera is terminated. After disconnect camera from usb and reconnect it again the result is a same as described above.

dmitrij9992905, 2022-12-04:
With some of cameras yes, it was, it needs to test with different cameras.
Now it's needed to improve the cameras compatibility.
If you try to use libuvc which uses libusb lib, there are some errors with camera handling. With uvc_stream I managed to connect to embedded USB camera with ultra wide lens with MJPEG support and I could get images
--
Отправлено из Mail.ru для Android воскресенье,

leeebo, 2023-01-06:
Feature request: support `Still Image Trigger Control`